Ysurac/openmptcprouter #443: Request Mikrotik Metarouter

umekzg, 2019-04-23:
Is it possible to make a build for Mikrotik Metarouter virtualisation?
MIPS image of openWRT exists and it works.

https://wiki.mikrotik.com/wiki/Manual:Metarouter
https://openwrt.org/docs/guide-user/virtualization/metarouter

Im using a Mikrotik RB with a RPI and VLANS as its easy to expand to real interfaces.
It would be possible to remove RPI and use all internal virtual network to work it all on Mikrotik.

Cheers and good work! Big fan! :)

Ysurac, 2019-04-23:
First answer: not possible.
Long answer: For now OpenMPTCProuter can't compile on MIPS processor, (can't compile ndpi and some others packages). Also I can't find any Mikrotik using MIPS with enough CPU power and enough flash space (I need 256Mo if I don't want to remove some packages).

umekzg, 2019-04-23:
to bad, it would be a killer combo :)

umekzg, 2019-04-23:
I think this one can handle it.

https://mikrotik.com/product/rb4011igs_rm

Ysurac, 2019-04-23:
But... no OpenWrt support for now.

umekzg, 2019-04-23:
ARM has no support for Metarouter, well that a bummer

juliansw, 2019-05-07:
Hi
What about the mikrotik RBM33G ? 
Dual LTE standard.

closed 2019-07-04 by Ysurac

dnwk, 2023-08-08:
It looks like RBM33G has OpenWRT support now (https://openwrt.org/toh/mikrotik/rbm33g)