Ysurac/openmptcprouter #444: Is it possible to OMR-Bypass to a secondary Shadowsocks server? [Feature Request]

dnwk, 2019-04-24:
Is it possible to OMR-Bypass to a secondary Shadowsocks server?  Route traffic to a second Shadowsocks server based on OMR-Bypass rules.

dnwk, 2019-04-24:
Or is there a way to set different ss-redir rule for different socks server?

Ysurac, 2019-04-24:
Not possible for now. You can define needed iptables rules in custom firewall rules.

dnwk, 2019-04-24:
So, I could set d

dnwk, 2019-04-24:
So, I could set IPtables rules for traffic goes to bypass or traffic goes to secondary SS server?  Could you help me with an example?

closed 2019-04-24 by dnwk

Ysurac, 2019-04-24:
You can use current rules as example.
You can see them using ```iptables-save```, the default rule for shadowsocks use port 1100. You can clone it, omr-bypass use ipset in dnsmasq configuration so you can use also a match-set to redirect some domains like omr-bypass.

dnwk, 2019-04-25:
Is there a way to batch load IP/CIDR into omr-bypass?

Ysurac, 2019-04-25:
No. But all configuration use ```uci```, so you write a script and put entries in omr-bypass

dnwk, 2019-04-25:
how do I use uci to add entries to omr-bypass? It's not clear in the commandline help

Ysurac, 2019-04-25:
https://oldwiki.archive.openwrt.org/doc/uci

dnwk, 2019-04-25:
So, I have to run these commands:

	set omr-bypass.ips[-1]=ips
	set omr-bypass.ips[-1].ip=$ip
	set omr-bypass.ips[-1].interface=eth1  
Is it?

Ysurac, 2019-04-25:
```uci add omr-bypass ips``` as first command

dnwk, 2019-04-25:
Got it figured out. Thanks

dnwk, 2019-04-25:
for ip in $(cat '/etc/route.txt'); do
uci add omr-bypass ips
uci set omr-bypass.@ips[-1]=ips
uci set omr-bypass.@ips[-1].ip=$ip
uci set omr-bypass.@ips[-1].interface="eth0.2"
uci commit omr-bypass
done

I have not quite figured out how to use uci batch. But getting close.

dnwk, 2019-04-25:
It doesn't work at all. There are 8k networks. Seems overload omr-bypass

dnwk, 2019-04-25:
It worked after reboot after running the command.  There are 8k networks. Seems overload omr-bypass webpage. whenever I try to load omr-bypass web page, it crashed the entire system

bournehlm, 2019-11-12:
How to take effect immediately after run "uci commit omr-bypass"?
And how to bypass all data traffic? I have try add "0.0.0.0/0" into ips, but not working.