Ysurac/openmptcprouter #446: Which wan to use as master?

DuredhelFinceleb, 2019-04-24:
It's actually a question an not a bug 
Sorry if this is not the right place to post this!

I have one VDSL2 connection and on 1 4G connection
Which one should be used as master and why? Are there recommendations for this?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.48 r0+9745-b0395cfc56 / LuCI Master (git-19.095.24565-19d15f2) 
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.992 >
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

Ysurac, 2019-04-24:
Connection with lowest RTT is used as master by default. In system->OpenMPTCProuter->Advanced settings you can choose to fix the master wan or to make it dynamic.
Master connection should be a stable connection.

DuredhelFinceleb, 2019-04-24:
Well I've used the 4g as master because according to speedtest it gave me a lot more bandwidth especially upload bandwidth

VDSL2 as master
Ping: 31 ms
D: 78.86
U: 19.22

4G as master
Ping: 30 ms
D: 102.05
U: 46.20

Around 40 Mb/s upload is what speedtest is giving me on the VPS (a FirstHeberg 200 Mb/s), so this is very good 

But it's true that it seems to be less stable, I don't have figures to prove it, but sometimes there are long delays before transfers start and sometime some items won't be downloaded (I have to retry once or twice to get them)

I'll revert back to use my VDSL2 as master. 
Thanks for the help

diogosilvan, 2019-04-28:
If the interface selected as master is offline, will the others work alone?
I have two connections, one master and another enable, if the master connection goes offline the second one will work alone? I understand that in this condition it will only be a "VPN" in a scenario where only one connection works.

Ysurac, 2019-04-28:
Yes, and when master is back online it's then used back as master.

closed 2019-05-25 by Ysurac