Ysurac/openmptcprouter #447: Nperf: waiting to close

DuredhelFinceleb, 2019-04-25:
## Expected Behavior

After reading that nPerf was better than Speedtest, I gave it a try.
I've tested it with and without OMR
nPerf should chain the DL test, the UL test and the ping test flawlessly

## Actual Behavior
When using OMR, nPerf stops for 2m30s after the DL test with the messeage 'Wating to close'
After that it completes both the UL and ping tests without hiccups.
[image]

## Steps to Reproduce the Problem

  1. ensure OMR is active
  2. go to nPerf website
  3. perform the speedtest

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.48 r0+9745-b0395cfc56 / LuCI Master (git-19.095.24565-19d15f2) 
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.992 > FirstHeberg VPS 200Mb/s
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

Ysurac, 2019-04-25:
I'm not able to reproduce the issue. You have the problem at each try ?

DuredhelFinceleb, 2019-04-25:
Yes, every single time

Lemax9, 2019-05-05:
Me too, the same problem
(with the same configuration VPS + Tpi3 b+)
but not with speedtest.telstra.com or fast.com or meter.net

DuredhelFinceleb, 2019-05-06:
I also have the issue with GOG Galaxay client update process.
It will freeze its download for about 2 to 3 mn at 2.5% then at 31.5% then at 52% and at 87.7% 
And finally it ends up with a download failed process... I tried reinstalling it using a fresh client from the website but at the next client update, it behaved exactly the same way
Weird thing is the issue is not present when Galaxy downloads games. 

I'm quite new to all this... What data can we provided you that might help you figuring out what's happening? And how to get those data?

closed 2019-07-04 by Ysurac