Ysurac/openmptcprouter #448: Waiting for root device PARTUUID=

Nadeistos, 2019-04-25:
## Expected Behavior

Full boot

## Actual Behavior

Stuck at the begining of boot : "Waiting for root device PARTUUID="
The boot goes on if i insert an usb key with omr 

## Steps to Reproduce the Problem

Copied omr on the internal drive on a minipc, dd or etcher same problem

## Specifications

  - OpenMPTCProuter version: 0.48
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-04-27:
I really need more info. Like log, platform config,...

Nadeistos, 2019-04-28:
Beelink Z85 Mini PC 2G Ram+ 64G Rom Intel Atom Z8350
removed win10 and installed Linux mint works fine
lsblk : 
mmcblk0 58,2Go disk
---mmcblk0p1 512Mo part boot
---mmcblk0p1 57,8 part /
mmcblk0boot0 4Mo disk
mmcblk0boot1 4Mo disk

i will try to find the log later ^^

Ysurac, 2019-04-28:
mmc ? x86_64 release was never tested for that.

Nadeistos, 2019-04-28:
ok :( i'll post the log tomorrow if you want...

Ysurac, 2019-04-28:
Yes. To check how I can make it work on that.

Nadeistos, 2019-04-29:
No logs...
here the screens of the boot

[image]
[image]

Missing mmc controler drivers ?

bosseb, 2019-05-18:
i have the problem with Beelink BT3 Pro Mini PC 4 Go de RAM DDR3 + 64 Go de ROM eMMC Système Windows Processeur: Intel x5-Z8350 GPU: Gen-9 

what can i do to find a solution ??

Ysurac, 2019-05-18:
You can try to compile it yourself and find what module or changes are needed to make it work ;)

bosseb, 2019-05-18:
on other forum they writing that is it because there is ROM eMMC

Ysurac, 2019-05-18:
Yes it's possible that the module needed is not available. But not idea what is needed.

bosseb, 2019-05-18:
i have find the solution, i have erase the emmc partition and i put an usb disk with OMC image on it, and it's works !!!

closed 2019-07-04 by Ysurac