Ysurac/openmptcprouter #449: Update to kernel 4.19

Ysurac, 2019-04-27:
OpenMPTCProuter will be updated to kernel 4.19 and MPTCP 0.95

What is working:
- x86_64 ext4
- RPI3 64bits ext4 (wifi not yet tested)

What is not working:
- x86_64 squashfs
- RPI3 64bits squashfs

What is not tested yet:
- x86
- RPI2
- WRT3200ACM/WRT32X

What will not be ported to 4.19 for now:
- BPI-R2

suyuan168, 2019-04-27:
This is really exciting. you're great

user747, 2019-04-28:
Have been testing out some of the generated builds with bcm2710

v0.48-25 squashfs - F2FS issue 
v0.48-37 ext4 - LUCI issues (pages were timing out) 

Can easily revert back to stable 0.48 4.14 for now. 

For VPS  4.19 kernel, have been testing a minimal .config using linux-image-4.19.0-0.bpo.4-cloud-amd64
that is built directly on a Openstack VPS. Not sure what the implications of this might be. I don't really have any side by side comparisons or benchmark.

```
cp /boot/config-4.19.0-0.bpo.4-cloud-amd64 ~/mptcp-f446ba359c3a36ac19a853982af731c368073683/.config
make oldconfig
sudo sed -ri '/CONFIG_SYSTEM_TRUSTED_KEYS/s/=.+/=""/g' .config
fakeroot make-kpkg --initrd --append-to-version=-mptcp kernel-image kernel-headers -j2
```

Ysurac, 2019-04-28:
Since v0.48 41 No more F2FS issues (it's disabled).
I don't have LUCI issues (yet)

VPS test script already have a  working 4.19 kernel with MPTCP.

suyuan168, 2019-04-28:
The configuration could not be saved just after testing with rpi2. It seems that I can't use it normally.openmptcprouter-v0.48-41-ge147d26-r0+9908-17cb490ac4-brcm2708-bcm2709-rpi-2-squashfs-sysupgrade .img

suyuan168, 2019-04-28:
Supported devices: rpi-2-b rpi-3-b rpi-3-b-plus rpi-cm raspberrypi,2-model-b raspberrypi,3-model-b raspberrypi,3-model-b-plus raspberrypi,3-compute-module raspberrypi,compute-module-3
Image check 'fwtool_check_image' failed.

user747, 2019-04-29:
bcm2710/openmptcprouter-v0.48-43-g5db886e-r0%2B9908-17cb490ac4-brcm2708-bcm2710-rpi-3-squashfs-factory.img.gz

Add WAN 1
Set VPS Server IP & ShadowSocks key --> Save settings
Page hangs and redirects to http://192.168.100.1/cgi-bin/luci
Log back in and all settings are reverted back to default

Ysurac, 2019-04-29:
I'm not able to reproduce the issue.

user747, 2019-04-29:
I will try to reproduce again with more logs. 
Should note that I'm also testing with custom VPS kernel, and using a different port for SS.

When I reverted back to 0.48 4.14, I didn't have this issue, and everything worked as expected.

suyuan168, 2019-04-30:
The same is true for me, RPI2 cannot save any settings. The default setting still needs to reset the password after rebooting. HDCP cannot get any IP.

jayanta525, 2019-04-30:
RPI2 unable to save any settings is an issue with OpenWrt-master.

Ysurac, 2019-04-30:
There is no RPI support upstream with 4.19 kernel. Do you have same problem with ext4 image ?

suyuan168, 2019-04-30:
Ext4 image I have tested the settings in RPI2 and can't save it.

user747, 2019-04-30:
Maybe my SD card is corrupt some how, but I have no issues with 0.48 4.14 on same hardware. 
Using Win32 Disk Imager to burn them to the card. Probably should be using dd at this point. 

openmptcprouter-v0.48-43-g5db886e-r0+9908-17cb490ac4-brcm2708-bcm2710-rpi-3-ext4-sysupgrade.img

```
root@OpenMPTCProuter:~# logread -f
Tue Apr 30 05:37:10 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.0.0.1)
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.103848] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.114652] Mem abort info:
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.118366]   ESR = 0x96000046
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.122316]   Exception class = DABT (current EL), IL = 32 bits
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.129234]   SET = 0, FnV = 0
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.133192]   EA = 0, S1PTW = 0
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.137201] Data abort info:
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.140918]   ISV = 0, ISS = 0x00000046
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.145581]   CM = 0, WnR = 1
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.149341] user pgtable: 4k pages, 39-bit VAs, pgdp = 00000000d4ab6b43
Tue Apr 30 05:37:11 2019 kern.alert kernel: [  168.156844] [0000000000000008] pgd=0000000036397003, pud=0000000036397003, pmd=0000000000000000
Tue Apr 30 05:37:11 2019 kern.emerg kernel: [  168.167246] Internal error: Oops: 96000046 [#1] SMP
```
SSH disconnects at this point. LUCI still works but pages time out and there's TCP timeouts to 192.168.100.1:80

I think this is the corresponding entry in LUCI System Log
```
Tue Apr 30 05:37:15 2019 kern.crit kernel: [   13.276201] EXT4-fs error (device mmcblk0p2): __ext4_new_inode:1114: comm sed: failed to insert inode 107: doubly allocated?
Tue Apr 30 05:37:15 2019 daemon.notice procd: /etc/rc.d/S5openmptcprouter: sed: can't create temp file '/etc/sysctl.confXXXXXX': I/O error
Tue Apr 30 05:37:15 2019 daemon.notice procd: /etc/rc.d/S5openmptcprouter: sed: can't create temp file '/etc/sysctl.d/10-default.confXXXXXX': Read-only file system
Tue Apr 30 05:37:15 2019 kern.crit kernel: [   13.294536] EXT4-fs (mmcblk0p2): Remounting filesystem read-only
```

Same with openmptcprouter-v0.48-43-g5db886e-r0+9908-17cb490ac4-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade .img 

No log entries though. It just resets back to default, which isn't really a bad thing.

Ysurac, 2019-04-30:
I have a running RPI3 ext4 image and a RPI2 squashfs image. Both are running well. I even can restore a backup to the RPI2 image, backup doesn't work on the x86_64 image. This is strange...

@user747 Do you have something on USB port ?

suyuan168, 2019-04-30:
squashfs-sysupgrade .img。Can it be used normally?Why can't I do it. strange

user747, 2019-04-30:
> I have a running RPI3 ext4 image and a RPI2 squashfs image. Both are running well. I even can restore a backup to the RPI2 image, backup doesn't work on the x86_64 image. This is strange...
> 
> @user747 Do you have something on USB port ?

No USB devices attached. BCM2710 using USB is a no in general since it shares resources with LAN. 

```
# remove sysctl already defined in /etc/sysctl.d/
	sed -i -e '/tcp_fin_timeout/d' -e '/tcp_keepalive_time/d' -e '/nf_conntrack_max/d' -e '/tcp_syn_retries/d' -e '/tcp_fastopen/d' /etc/sysctl.conf
	sed -i -e '/tcp_fin_timeout/d' -e '/tcp_keepalive_time/d' -e '/nf_conntrack_max/d' -e '/tcp_syn_retries/d' -e '/tcp_fastopen/d' /etc/sysctl.d/10-default.conf
```

```
root@OpenMPTCProuter:/etc/rc.d# ls -l /etc/sysctl.conf
-rw-r--r--    1 root     root            80 Apr 26 13:31 /etc/sysctl.conf

```
```
-rw-r--r--    1 root     root            80 Apr 26 13:31 /etc/sysctl.conf
-rw-r--r--    1 root     root           642 Apr 26 13:31 /etc/sysctl.d/10-default.conf
```
permissions ok. This happens when changing Server Port on http://192.168.100.1/cgi-bin/luci/admin/services/shadowsocks-libev/servers

Maybe since it's not 65101 that breaks things somewhere ?

Ysurac, 2019-04-30:
Same write problem with squashfs on RPI3 64bits.
No, it's a write problem. I need to find why, but I think I have a similar problem also on x86_64, so this may not be a rpi patch problem.

Ysurac, 2019-04-30:
For now 4.19 is really unstable even on x86_64 (loop reboot).
I will make a 4.14 release before trying to fix 4.19.

Ysurac, 2019-05-01:
I have some of the problems with kernel 4.14 on x86_64, so at least some problem are not from 4.19 kernel.
Package and LuCi feeds are now from master, this can be the source of some bugs.
I will make some test with stable feeds, and check what bugs I have.

Ysurac, 2019-05-01:
Crash x86_64 problem seems to come from master openwrt branch not from package and LuCi. I need to find which commit put this problem.

Ysurac, 2019-05-06:
A crash is solved, but now I have problems with 4.19 MPTCP kernel on both VPS and OpenMPTCProuter.
Next release will be 4.14 based, then I will try to debug 4.19 MPTCP kernel.

user747, 2019-05-08:
4.14.113 OMR v0.48-53 and a small 10MB IPv4 only 4.19.36 on Debian Openstack VPS. Only ss-server for now on VPS. I will be testing glorytun-tcp next for other traffic. OMR forwards all UDP to master WAN macvlan. 

My VPS spec are minimal (1GB memory, 1vcpu) Memory used at boot 40MB and never over 100MB. 
I import this image into Openstack, https://cdimage.debian.org/cdimage/openstack/current/debian-9.9.0-openstack-amd64.raw

Install mptcp kernel, shadowsocks, shorewall ect ect, reboot, and only add loaded modules on top of a known good .config, then rebuild again. 

Only 1 VPS crash so far (I think it was due to running ss-server with -v) Haven't really done much looking though for the issue.  Inbound ICMP is also turned off via Shorewall. This might break omr-tracker ? 

`Wed May  8 02:02:28 2019 user.notice omr-tracker: Can't create route to 1.0.0.1 via 192.168.12.1 dev wan6 src 192.168.12.2. waiting...`

I'll try building some 4.19 BCM2710 images and test more. Also still using a local Pi-Hole server for DNS without issues. Next I need to get this working https://docs.pi-hole.net/guides/dns-over-https/ 
I'm hesitant to run Pi-Hole on VPS for now (resources / open resolver security)

Ysurac, 2019-05-08:
Pi-Hole on VPS is supported in latest test build and test VPS script (both are back on 4.14 now). I made a script to install Pi-Hole and configure it for OMR : http://www.openmptcprouter.com/server/omr-pihole.sh (this make it listen only on VPN interfaces and need more tests).

I have errors on VPS side with 4.19 kernel and reboot loop on OMR side. I will try to solve this after 0.49 release.

suyuan168, 2019-05-08:
I used the latest vps script with version 0.993 and the latest rpi2 is working fine. Rpi2 is the 4.14 kernel.

user747, 2019-05-08:
For VPS side , I have been testing a fast way to enable 2FA and add some extra ssh security 

```
#Generate ssh keypair and copy pub to server
ssh-keygen -t rsa -b 4096
cat ~/.ssh/id_rsa.pub | ssh <user>@<hostname> 'cat >> .ssh/authorized_keys && echo "Key copied"'

#disable root ssh login
sed -i 's/PermitRootLogin yes/PermitRootLogin no/g' /etc/ssh/sshd_config
#disable password authentication
sed -i 's/#PasswordAuthentication.*/PasswordAuthentication no/' /etc/ssh/sshd_config
#Enable 2FA
sed -i 's/ChallengeResponseAuthentication no/ChallengeResponseAuthentication yes/g' /etc/ssh/sshd_config
echo "AuthenticationMethods publickey,password publickey,keyboard-interactive" | sudo tee -a /etc/ssh/sshd_config

#Install and configure TOTP
sudo apt-get install -y libpam-google-authenticator
google-authenticator -t -d -W -r 1 -R 15 -Q utf8 -f

#Restart sshd.service
systemctl restart sshd.service

```

suyuan168, 2019-05-08:
The download speed is still very slow and can't break through 100M even if I have plugged in a wired network. Opening a web page also does not display properly. I thought it was solved.

user747, 2019-05-09:
Getting error when using OMR_TARGET="rpi3" OMR_KERNEL="4.19" ./build.sh

```
Applying /home/debian/openmptcprouter/rpi3/source/target/linux/brcm2708/patches-4.19/0435-Revert-cgroup-Disable-cgroup-memory-by-default.patch using plaintext:
patching file kernel/cgroup/cgroup.c
Hunk #1 FAILED at 5290.
Hunk #2 FAILED at 5332.
Hunk #3 FAILED at 5721.
3 out of 3 hunks FAILED -- saving rejects to file kernel/cgroup/cgroup.c.rej
Patch failed!  Please fix /home/debian/openmptcprouter/rpi3/source/target/linux/brcm2708/patches-4.19/0435-Revert-cgroup-Disable-cgroup-memory-by-default.patch!
Makefile:34: recipe for target '/home/debian/openmptcprouter/rpi3/source/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.19.38/.prepared_6f6927b27881531891eb883690563f92' failed
make[4]: *** [/home/debian/openmptcprouter/rpi3/source/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.19.38/.prepared_6f6927b27881531891eb883690563f92] Error 1
make[4]: Leaving directory '/home/debian/openmptcprouter/rpi3/source/target/linux/brcm2708'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/debian/openmptcprouter/rpi3/source/target/linux'
time: target/linux/compile#8.88#7.47#13.49
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/debian/openmptcprouter/rpi3/source'
target/Makefile:18: recipe for target '/home/debian/openmptcprouter/rpi3/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile' failed
make[1]: *** [/home/debian/openmptcprouter/rpi3/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/debian/openmptcprouter/rpi3/source'
Build failed - please re-run with -j1 to see the real error message
/home/debian/openmptcprouter/rpi3/source/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 1

```

Ysurac, 2019-05-10:
Fixed in master for both errors.

user747, 2019-05-10:
Noticing more kernel freezes with 4.19 on VPS.  What is your debugging process look like? 
I might try https://multipath-tcp.org/pmwiki.php/Users/HandleCrashdumps

Ysurac, 2019-05-11:
True, I have same problems.
I will release OpenMPTCProuter 0.49 still based on 4.14 today or tomorrow, then I will try to debug 4.19.

user747, 2019-05-12:
`echo "kernel.printk = 8 8 1 7" | sudo tee -a /etc/sysctl.conf `
Doesn't seem to help for this type of bug. Looking at the Openstack console log though does give me more information

```
[  190.399668] BUG: unable to handle kernel NULL pointer dereference at 0000000000000000
[  190.401691] PGD 0 P4D 0 
[  190.402424] Oops: 0000 [#1] SMP PTI
[  190.403352] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.36-mptcp #1
[  190.405006] Hardware name: OpenStack Foundation OpenStack Nova, BIOS Ubuntu-1.8.2-1ubuntu1~cloud0 04/01/2014
[  190.407456] RIP: 0010:__list_del_entry_valid+0x25/0x90
[  190.409124] Code: 00 00 00 00 00 48 8b 07 48 b9 00 01 00 00 00 00 ad de 48 8b 57 08 48 39 c8 74 26 48 b9 00 02 00 00 00 00 ad de 48 39 ca 74 2b <48> 8b 12 48 39 d7 75 34 48 8b 50 08 48 39 d7 75 3c b8 01 00 00 00
[  190.414139] RSP: 0018:ffff973c7ea03a98 EFLAGS: 00010207
[  190.415422] RAX: 0000000000000000 RBX: ffff973c7c641b00 RCX: dead000000000200
[  190.417221] RDX: 0000000000000000 RSI: 0000000078ddd407 RDI: ffff973c7c641b78
[  190.419046] RBP: ffff973c7c7a71c0 R08: 0000000000000000 R09: 0000000000000000
[  190.420808] R10: ffff973c7ea03908 R11: 00000000000d2c4f R12: ffff973c7c63b880
[  190.422727] R13: 0000000000000000 R14: ffff973c7c7a0900 R15: ffff973c7c7a08c0
[  190.424696] FS:  0000000000000000(0000) GS:ffff973c7ea00000(0000) knlGS:0000000000000000
[  190.426891] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  190.428758] CR2: 0000000000000000 CR3: 000000002f60a004 CR4: 00000000000606b0
[  190.430538] Call Trace:
[  190.431260]  
[  190.431950]  mptcp_data_ack+0x35c/0xa60
[  190.432957]  mptcp_handle_options+0x155/0x510
[  190.434058]  tcp_validate_incoming+0x397/0x490
[  190.435202]  tcp_rcv_established+0x1b1/0x630
[  190.436312]  tcp_v4_do_rcv+0x103/0x220
[  190.437344]  tcp_v4_rcv+0xaf6/0xc20
[  190.438485]  ip_local_deliver_finish+0xac/0x1e0
[  190.439679]  ip_local_deliver+0x6b/0xe0
[  190.440881]  ? ip_rcv_core.isra.17+0x2a0/0x2a0
[  190.442034]  ip_rcv+0x52/0xd0
[  190.443105]  ? ip_rcv_finish_core.isra.16+0x370/0x370
[  190.444587]  __netif_receive_skb_one_core+0x52/0x70
[  190.445767]  netif_receive_skb_internal+0x34/0xe0
[  190.446920]  napi_gro_receive+0xb8/0xe0
[  190.448008]  receive_buf+0x3be/0x1410 [virtio_net]
[  190.449178]  ? vring_unmap_one+0x16/0x70 [virtio_ring]
[  190.450491]  ? detach_buf+0x68/0x110 [virtio_ring]
[  190.451781]  virtnet_poll+0xdf/0x333 [virtio_net]
[  190.452973]  net_rx_action+0x297/0x400
[  190.454238]  __do_softirq+0x10d/0x2c3
[  190.455488]  irq_exit+0xc2/0xd0
[  190.456464]  do_IRQ+0x52/0xe0
[  190.457553]  common_interrupt+0xf/0xf
[  190.458713]  
[  190.459333] RIP: 0010:native_safe_halt+0x2/0x10
[  190.460458] Code: fe ff ff fb 66 66 90 66 66 90 e9 ae fe ff ff 0f 0b e9 6e ff ff ff 0f 0b e9 e6 fe ff ff 90 90 90 90 90 90 90 90 90 90 90 fb f4  0f 1f 00 66 2e 0f 1f 84 00 00 00 00 00 f4 c3 90 90 90 90 90 90
[  190.464965] RSP: 0018:ffffffffb3603ea0 EFLAGS: 00000246 ORIG_RAX: ffffffffffffffd3
[  190.466815] RAX: ffffffffb2c19dd0 RBX: 0000000000000000 RCX: ffff973c7ea1a400
[  190.468597] RDX: ffffffffb3648df8 RSI: ffff973c7ea1a400 RDI: 0000000000000082
[  190.470456] RBP: 0000000000000000 R08: 000000000001c980 R09: 0000002c5aeddf31
[  190.472126] R10: 00000000000256ee R11: 0000000000000000 R12: ffffffffb3613740
[  190.474164] R13: ffffffffb37089e0 R14: 0000000000000000 R15: 0000000000000000
[  190.475888]  ? __sched_text_end+0x1/0x1
[  190.476870]  default_idle+0x1c/0x140
[  190.477834]  do_idle+0x9d/0x150
[  190.478912]  cpu_startup_entry+0x6f/0x80
[  190.480097]  start_kernel+0x516/0x536
[  190.481053]  secondary_startup_64+0xa4/0xb0
[  190.482199] Modules linked in: xt_recent xt_comment ipt_REJECT nf_reject_ipv4 xt_addrtype xt_mark iptable_mangle xt_tcpudp xt_CT iptable_raw xt_multiport xt_conntrack nfnetlink_log xt_NFLOG xt_LOG nf_log_ipv4 nf_log_common nf_nat_tftp nf_nat_snmp_basic nf_conntrack_snmp nf_nat_sip nf_nat_pptp nf_nat_proto_gre nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda ts_kmp nf_conntrack_amanda nf_conntrack_sane nf_conntrack_tftp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_proto_gre nf_conntrack_netlink nfnetlink nf_conntrack_netbios_ns nf_conntrack_broadcast nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp iptable_filter ipt_MASQUERADE iptable_nat nf_nat_ipv4 nf_nat kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd cryptd glue_helper evdev
[  190.499445]  serio_raw virtio_balloon button sch_fq ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 fscrypto ata_generic virtio_net net_failover failover virtio_blk ata_piix libata crc32c_intel virtio_pci virtio_ring virtio scsi_mod
[  190.505004] CR2: 0000000000000000
[  190.506027] ---[ end trace cb29c51a68d37293 ]---
[  190.507233] RIP: 0010:__list_del_entry_valid+0x25/0x90
[  190.508696] Code: 00 00 00 00 00 48 8b 07 48 b9 00 01 00 00 00 00 ad de 48 8b 57 08 48 39 c8 74 26 48 b9 00 02 00 00 00 00 ad de 48 39 ca 74 2b <48> 8b 12 48 39 d7 75 34 48 8b 50 08 48 39 d7 75 3c b8 01 00 00 00
[  190.513227] RSP: 0018:ffff973c7ea03a98 EFLAGS: 00010207
[  190.514553] RAX: 0000000000000000 RBX: ffff973c7c641b00 RCX: dead000000000200
[  190.516531] RDX: 0000000000000000 RSI: 0000000078ddd407 RDI: ffff973c7c641b78
[  190.518497] RBP: ffff973c7c7a71c0 R08: 0000000000000000 R09: 0000000000000000
[  190.520325] R10: ffff973c7ea03908 R11: 00000000000d2c4f R12: ffff973c7c63b880
[... cut ...]
[  190.523743] FS:  0000000000000000(0000) GS:ffff973c7ea00000(0000) knlGS:0000000000000000
[  190.526160] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  190.527624] CR2: 0000000000000000 CR3: 000000002f60a004 CR4: 00000000000606b0
[  190.529372] Kernel panic - not syncing: Fatal exception in interrupt
[  190.531736] Kernel Offset: 0x31600000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[  190.534991] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]--
```

uryupinsk, 2019-05-13:
Feedback: OpenMPTCProuter 0.49 (x86_64) with VPS 0.993 released is all working correctly for me on kernel 4.14. I haven't tested kernel 4.19.

Ysurac, 2019-05-25:
Latest release on latest MPTCP 0.95, with some patches, on kernel 4.19.44 seems more stable on x86_64 (and VPS side). Seems also better on RPI.

Ysurac, 2019-05-26:
Still some crash on RPI, I will update RPI 4.19 patches.

uryupinsk, 2019-05-26:
OMR Status Page tells me there is a 0.994 VPS, but I don't see any related commit on the repo. Also, are OMR 0.49.7 and VPS 0.994 on MPTCP 0.95 and kernel 4.19 by default?

Ysurac, 2019-05-26:
Yes there is a 0.994 VPS script.
And OMR 0.49.7 and VPS 0.994 are not on 4.19 kernel. This will be a major release and not while it's not stable.

Ysurac, 2019-06-01:
4.19 is now stable using ext4 on RPI and x86_64.
F2FS/SquashFS still not working. It's a know issue by OpenWrt.

suyuan168, 2019-06-02:
F2FS/SquashFS hopes to be solved. 4.19 Is there a BLEST?

suyuan168, 2019-06-16:
Can RPI2's 4.19 save settings normally? I have tried F2FS root EXT4 or there is no way to save any settings. What is the problem

Ysurac, 2019-06-16:
This work with ext4.

suyuan168, 2019-06-16:
EXT4 is problematic. Unable to save settings. prompt
[ 17.693295] EXT4-fs error (device mmcblk0p2): ext4_mb_generate_buddy:747: group 1, block bitmap and bg descriptor inconsistent: 24880 vs 24882 free clusters
[ 17.710356] EXT4-fs (mmcblk0p2): Remounting filesystem read-only
[ 17.718104] EXT4-fs (mmcblk0p2): ext4_writepages: jbd2_start: 2147483646 pages, ino 924; err -30
I failed again.

suyuan168, 2019-06-16:
I can experience 4.19 without restarting or unplugging the network cable. The speed is very fast. My RPI can download 110M/S to upload 300M/S. But at the moment it seems that ext4 and F2FS still have problems. It is a kernel panic. Just unplug the cable and reinsert it and the system settings will not be saved. But at least I see a lot of hope.Come on, thank you very much.

Ysurac, 2019-06-16:
I have no problem with ext4 on RPI3 64bits, never tested on RPI2.
The problem with F2FS seems to be related with loop device problem.

suyuan168, 2019-06-17:
F2FS
Is there a repair plan? Or is it the official release of 4.19 to fix the F2FS solution?
I use ext4 to prompt EXT4 is problematic. Unable to save settings. prompt.
The most stable at present is 4.14-123

Ysurac, 2019-06-21:
I now have a working BPI-R2 with ext4 and 4.19 kernel. Will be available in repo soon.

user747, 2019-07-05:
No crashes yet with ext4 OMR v0.50.2-54 with 4.19 and VPS on latest commit of MPTCP v0.95 branch. 

On rpi3 with squashfs i still lose all settings when rebooting, but will use ext4 from now on.

Feels more stable with a small speed boost as well.

tentious, 2019-07-10:
@Ysurac 

I'm running OMR v0.50.2-53-gac8eeb3 r0+10358-ef5b7aa6c3 with latest test VPS script and facing ext4 read-only file system error after upgrading. I can not save changes.

**System Log file is filled with this log:**
Wed Jul 10 11:17:09 2019 daemon.err omr-tracker[6344]: /bin/omr-tracker: line 221: can't create 1: Read-only file system
Wed Jul 10 11:17:09 2019 daemon.err omr-tracker[6344]: /bin/omr-tracker: line 217: can't create 1: Read-only file system

**Kernel Log:**
[   10.613723] usbserial: USB Serial support registered for Qualcomm USB modem
[   10.639781] usbcore: registered new interface driver rt2800usb
[   10.662877] usbcore: registered new interface driver rtl8192cu
[   10.669124] kmodloader: done loading kernel modules from /etc/modules.d/*
[   12.892949] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   13.294915] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   14.502976] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[   14.713287] EXT4-fs error (device mmcblk0p2): mb_free_blocks:1457: group 1, inode 118: block 44321:freeing already freed block (bit 11553); block bitmap corrupt.
[   14.731882] EXT4-fs (mmcblk0p2): Remounting filesystem read-only
[   14.738071] EXT4-fs error (device mmcblk0p2): ext4_mb_generate_buddy:747: group 1, block bitmap and bg descriptor inconsistent: 27499 vs 27500 free clusters
[  305.115068] EXT4-fs (mmcblk0p2): error count since last fsck: 4
[  305.121109] EXT4-fs (mmcblk0p2): initial error at time 1562292766: __ext4_new_inode:1114
[  305.129345] EXT4-fs (mmcblk0p2): last error at time 1562292773: ext4_mb_generate_buddy:747

Ysurac, 2019-07-13:
Current images at https://www.openmptcprouter.com/release/4.19/ are ready for tests.
VPS script at http://www.openmptcprouter.com/server/debian9-x86_64-test.sh

tentious, 2019-07-15:
Updated to latest test image, this resolved the read-only ext4 fs, however I'm getting a ton of  Shadowsocks restarts now.

`
Mon Jul 15 14:23:21 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:23:21 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:23:24 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:23:24 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:23:45 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:23:45 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:23:50 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:23:50 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:24:05 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:24:05 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:24:08 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:24:08 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:24:27 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:24:27 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:24:31 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:24:31 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:24:51 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Mon Jul 15 14:24:58 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:24:58 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:25:17 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:25:17 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:25:30 2019 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Mon Jul 15 14:25:33 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:25:33 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:25:43 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:25:43 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:25:58 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:25:58 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:26:09 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:26:09 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:26:20 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:26:20 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:26:28 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:26:28 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:26:41 2019 daemon.err miniupnpd[3864]: ioctl(s, SIOCGIFADDR, ...): Address not available
Mon Jul 15 14:26:41 2019 daemon.err miniupnpd[3864]: Failed to get IP for interface 6in4-omr6in4
Mon Jul 15 14:26:41 2019 daemon.warn miniupnpd[3864]: HTTP Connection from 192.168.1.18 closed unexpectedly
Mon Jul 15 14:26:41 2019 daemon.warn miniupnpd[3864]: HTTP Connection from 192.168.1.18 closed unexpectedly
Mon Jul 15 14:26:41 2019 daemon.warn miniupnpd[3864]: HTTP Connection from 192.168.1.18 closed unexpectedly
Mon Jul 15 14:26:41 2019 daemon.warn miniupnpd[3864]: HTTP Connection from 192.168.1.18 closed unexpectedly
Mon Jul 15 14:26:45 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:26:45 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:26:49 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:26:49 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:26:51 2019 daemon.err miniupnpd[3864]: ioctl(s, SIOCGIFADDR, ...): Address not available
[... cut ...]
Mon Jul 15 14:27:09 2019 daemon.err omr-tracker[19685]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
Mon Jul 15 14:27:14 2019 daemon.err miniupnpd[3864]: ioctl(s, SIOCGIFADDR, ...): Address not available
Mon Jul 15 14:27:14 2019 daemon.err miniupnpd[3864]: Failed to get IP for interface 6in4-omr6in4
Mon Jul 15 14:27:14 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Mon Jul 15 14:27:14 2019 daemon.err omr-tracker[19687]: /etc/rc.common: /etc/init.d/shadowsocks-libev: line 267: syntax error: unexpected "}"
`

Ysurac, 2019-07-15:
It's fixed in current release.

tentious, 2019-07-16:
Sorry, yes, I just updated again(to 7/14 release) and this is now fixed. So far, looks really great! I'm hoping the read-only FS doesn't pop up again.

closed 2019-12-20 by Ysurac