Ysurac/openmptcprouter #450: Carte réseau instable

seb-65, 2019-04-28:
Bonjour Ysurac,

Je viens d'installer la dernière version x86-64 UEFI (64-bit) sur le boitier suivant :
- Intel Celeron processor J1900
- 4 x 10/100/1000 Mbps Ethernet ports (Intel i211)

Au bout de quelques heures je perds complétement l'accès au routeur sur le port LAN. Je ne peux pas pinguer le router ni lancer l'interface web ni lancer SSH.

Même chose sur les autres interfaces WAN, si je me branche directement dessus je ne peux pas pinguer l'IP enregistrée sur l'interface... Je n'ai pas encore monté/configué la partie VPS !

Si je redémarre électriquement le routeur tout refonctionne pendant quelques heures... Avant d'installer OMPTCP j'avais Debian et aucun problème.

Est-ce un problème de firmware à rajouter ?

Merci d'avance du retour.

seb-65, 2019-04-28:
Re,

Je pense plutôt que le système est instable... J'ai lancé une console sur ttyS0 afin d'avoir le contenu de logread -f mais au bout de quelques heures je n'avais plus la main sur le système. Impossible de faire répondre le routeur via la console !

Obligé de faire redémarrage électrique  :(

Ysurac, 2019-04-28:
Déjà il me faut la version utilisée... Il manque aussi le log.
Sinon ça ressemble beaucoup à un soucis d'IP déjà utilisé sur le réseau.

seb-65, 2019-04-29:
La dernière version présente sur le site => squashfs image: openmptcprouter-v0.48-r09745-b0395cfc56-x86-64-combined-squashfs.img.gz (2019/04/03 - 38.36Mo - sha256sum)

>Il manque aussi le log.
Dernières lignes dans la console avant plantage complet du système :
`Sun Apr 28 15:40:11 2019 daemon.err omr-tracker[6761]: ping: unknown iface yes
Sun Apr 28 15:40:17 2019 daemon.err omr-tracker[6761]: ping: unknown iface yes
Sun Apr 28 15:40:25 2019 daemon.err omr-tracker[6760]: ping: unknown iface yes
Sun Apr 28 15:40:30 2019 daemon.err omr-tracker[6760]: ping: unknown iface yes
Sun Apr 28 15:40:37 2019 daemon.err omr-tracker[6761]: ping: unknown iface yes
Sun Apr 28 15:40:37 2019 daemon.err omr-tracker[6760]: ping: unknown iface yes
Sun Apr 28 15:40:44 2019 daemon.err omr-tracker[6760]: ping: unknown iface yes
Sun Apr 28 15:40:44 2019 daemon.err omr-tracker[6761]: ping: unknown iface yes`

> Sinon ça ressemble beaucoup à un soucis d'IP déjà utilisé sur le réseau.
Absolument pas, le routeur est connecté uniquement à un PC (avec ou sans DHCP même problème).

Ysurac, 2019-04-29:
"iface yes" ? Il va me falloir la configuration la (dans Système->OpenMPTCProuter puis Show all settings en enlevant les clefs/ip du serveur).

seb-65, 2019-04-29:
`dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[0].network='lan'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].input='REJECT'
firewall.@zone[1].output='ACCEPT'
firewall.@zone[1].forward='REJECT'
firewall.@zone[1].masq='1'
firewall.@zone[1].mtu_fix='1'
firewall.@zone[1].network='wan wan6 wan1 wan2'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
[... 846 lines cut ...]
upnpd.config.enabled='0'
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3'`

seb-65, 2019-05-11:
Bonjour,

Une idée du problème ? Merci !

TRSATOC, 2019-05-14:
Bonjour,

Je ne sais pas si notre problème est lié, mais de notre côté également nous rencontrons des instabilités et nos logs remontent également le fameux: 'ping: unknown iface yes', exemple:
`Tue May 14 17:18:06 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:18:08 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:18:13 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:18:21 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:18:22 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:18:26 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:18:32 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:18:33 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:18:40 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:18:42 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:18:48 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:18:52 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:00 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:02 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:19:05 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:10 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:16 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:19:19 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:28 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:29 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:19:36 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:19:36 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:42 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:50 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:19:52 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:19:59 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:20:07 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:20:10 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:20:20 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:20:23 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:20:25 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:20:36 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:20:38 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:20:43 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:20:50 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:20:51 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:20:57 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:21:01 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:21:03 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes
Tue May 14 17:21:11 2019 daemon.err omr-tracker[5724]: ping: unknown iface yes
Tue May 14 17:21:15 2019 daemon.err omr-tracker[5725]: ping: unknown iface yes`

Ysurac, 2019-05-14:
I'm able to reproduce the "unknown iface yes" this happen when no VPS ip is defined. I remove this log error in next release.

Ysurac, 2019-05-15:
I removed the "ping unknow iface yes error" in OpenMPTCProuter 0.49.2

closed 2019-12-20 by Ysurac