Ysurac/openmptcprouter #451: LuCi applications in make menuconfig

jayanta525, 2019-04-29:
I successfully build OpenMPTCProuter for a sun8i platform(OrangePi R1; 14$ SBC with dual 100M ethernet ports based on Allwinner H2+ SoC).
The build is working perfectly with default build-parameters.

How do i compile additional luci-app-* which weren't there on make menuconfig. I tried adding the luci feeds to feeds.conf followed by scripts/feeds update and install, but that reset all the default package selection.

I need luci-app-statistics and collectd modules for system health tracking and also davfs2 for webdav mounting.

Thanks. Keep up the great work.

Ysurac, 2019-04-29:
```scripts/feeds install -a luci``` if openmptcprouter related packages are not selected you can select all in OpenMPTCProuter menu with openmptcprouter-full package (if the package is not present, then there is a conflict somewhere).

jayanta525, 2019-04-29:
will selecting openmptcprouter-full package include all the other required packages, like mptcp, luci-app-mptcp?

jayanta525, 2019-04-29:
what are the minimum hardware requirements for OpenMPTCProuter to be installed on?

jayanta525, 2019-04-29:
I came across this error while compiling:

`make -j17 V=sc
`
```
Collected errors:
 * check_data_file_clashes: Package luci-lib-iptparser wants to install file /workspace/openmptcprouter/opi-r1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/usr/lib/lua/luci/sys/iptparser.lua
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package luci-app-statistics.
 * check_data_file_clashes: Package luci-lib-iptparser wants to install file /workspace/openmptcprouter/opi-r1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/usr/lib/lua/luci/sys/iptparser.lua
        But that file is already provided by package  * luci-base
 * opkg_install_cmd: Cannot install package luci-lib-iptparser.
 * resolve_conffiles: Existing conffile /workspace/openmptcprouter/opi-r1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/etc/config/glorytun is different from theconffile in the new package. The new conffile will be placed at /workspace/openmptcprouter/opi-r1/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/etc/config/glorytun-opkg.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/workspace/openmptcprouter/opi-r1/source'
```

Package installation conflict. Any suggestions?

Ysurac, 2019-04-29:
Not using luci-app-statistics ;)
I need to separate luci-base.

Ysurac, 2019-04-29:
Should be better in latest commit. But master is not really stable.

jayanta525, 2019-04-30:
I installed the collectd modules and enabled listener and server interfaces. I can now check system health and performance graphs on the VPS itself.

luci-app-statistics uses RRDTool which makes graphs with the data from collectd-modules. Its better to send the data to the VPS and let the VPS display the graphs on a web-server.

Thanks.

closed 2019-04-30 by jayanta525