Ysurac/openmptcprouter #452: How to access OMR from wan ?

Scaff31, 2019-04-30:
Hello,

I try many rules and port redirect but can't access OMR from VPS wan IP

Thanks for your work.

suyuan168, 2019-05-01:
You can refer to the wiki here to write clearly. very simple
https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

Scaff31, 2019-05-01:
Yeah thx suyuan but i want to access to omr from port 65400 redirect to 80 on OMR i try many config but no one work

suyuan168, 2019-05-01:
You only need to enable all traffic sent from openmptcprouter in the system openmptcprouter settings - advanced settings. And enable the forwarding rule to save the configuration in the openmptcprouter firewall. But you must ensure that your forwarding settings are forwarded by the VPN to the LAN. And your local computer sets the correct LAN IP address. Pay attention to the firewall settings of the local computer.I tested it through a tutorial on the wiki. The VPS was successfully opened and forwarded to the local port. And it is successful.

Ysurac, 2019-05-01:
Port 65000-65535 are reserved to OpenMPTCProuter, so not possible to redirect them. And port 65400 is used by iperf3 on OpenMPTCProuter VPS if I remember well.

Scaff31, 2019-05-02:
Ok so i try with port 59888 but can't access OMR from wan.
i can acces everything behind ( esxi, pfsense, web server, .... ) but not on omr.

From: any host in vpn via any router at port 59888 forward IP:192.168.100.1 port 80 in lan

What is wrong ?

Scaff31, 2019-05-02:
my bad

my rules was Ok

Juste another rules who block this one

Thx for help ^^

closed 2019-05-02 by Scaff31