Ysurac/openmptcprouter #453: Error compiling 0.47.1 release for x86_64

lukerutherford, 2019-04-30:
## Expected Behavior

Expected to compile the 0.47.1 release for x86_64 successfully

## Actual Behavior

Configuring luci-proto-ncm.
Configuring kmod-net-rtl8192su.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 *      luci-lib-iptparser
 * opkg_install_cmd: Cannot install package luci.
 * resolve_conffiles: Existing conffile /home/lrutherford/openmptcprouter-0.47.1/x86_64/source/build_dir/target-x86_64_musl/root-x86/etc/config/glorytun is different from the conffile in the new package. The new conffile will be placed at /home/lrutherford/openmptcprouter-0.47.1/x86_64/source/build_dir/target-x86_64_musl/root-x86/etc/config/glorytun-opkg.
make[2]: *** [package/install] Error 255
make[2]: Leaving directory `/home/lrutherford/openmptcprouter-0.47.1/x86_64/source'
make[1]: *** [/home/lrutherford/openmptcprouter-0.47.1/x86_64/source/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory `/home/lrutherford/openmptcprouter-0.47.1/x86_64/source'
make: *** [world] Error 2

## Steps to Reproduce the Problem

  1.  Untar the openmptcprouter-0.47.1.tar.gz file
  2.  Run the build.
  3.

## Specifications

  - OpenMPTCProuter version: (0.47,1)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)

Ysurac, 2019-05-01:
You need to use the OpenMPTCProuter feeds from 0.47.1 also. So you can't just run the build.

lukerutherford, 2019-05-01:
Oh I see, thanks!  I downloaded the 0.47.1 feeds and untarred them.  I replaced the files in here (from my previous build attempt that failed):
/home/lrutherford/openmptcprouter-0.47.1/feeds/openmptcprouter
with the 0.47.1 files I downloaded.  Then from I did:
lrutherford@vcaeng14:~/openmptcprouter-0.47.1/x86_64/source$ make clean
Then 
lrutherford@vcaeng14:~/openmptcprouter-0.47.1/x86_64/source$ make -j

I get this warning:
time: target/linux/prereq#0.07#0.09#0.16
WARNING: your configuration is out of sync. Please run make menuconfig, oldconfig or defconfig!
 make[1] world
 make[2] tools/compile
 ..

Is there a better way to do this so it doesn't think my configuration is out of sync?

I

Ysurac, 2019-05-01:
You can set build in build.sh

closed 2019-05-02 by Ysurac