Ysurac/openmptcprouter #454: Can VPN tunnel down be linked to deletion of omr6in4 interface?

xabix, 2019-05-01:
## Expected Behavior
Hello Experts,

Not sure if it's linked but I removed all ipv6 in sysctl.conf and deleted omr6in4 interface which was not launched (at boot). In status I see 'VPN TUNNEL DOWN'.

BTW in the config, they are still 2 references to 'omr6in4' which are left over and not automatically deleted.
[config.txt](https://github.com/Ysurac/openmptcprouter/files/3135958/config.txt)

merci for your inputs on why this tunnel down message.

## Actual Behavior/Logs
Wed May  1 22:09:59 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Wed May  1 22:09:59 2019 daemon.info glorytun[10734]: INITIALIZED tun0
Wed May  1 22:10:00 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Wed May  1 22:10:00 2019 user.notice post-tracking-post-tracking: Set server vps (VPS_IP) route via 192.168.9.1 metric 7
Wed May  1 22:10:00 2019 daemon.err glorytun[10734]: VPS_IP.65001: connected
Wed May  1 22:10:00 2019 daemon.info glorytun[10734]: STARTED tun0
Wed May  1 22:10:01 2019 user.notice post-tracking-post-tracking: Set server vps (VPS_IP) route via 192.168.8.1 metric 6
Wed May  1 22:10:06 2019 user.notice post-tracking-post-tracking: Set server vps (VPS_IP) route via 192.168.9.1 metric 7
Wed May  1 22:10:07 2019 daemon.err glorytun[10734]: tun write: I/O error
Wed May  1 22:10:07 2019 daemon.info glorytun[10734]: STOPPED tun0
Wed May  1 22:10:37 2019 daemon.info glorytun[13265]: INITIALIZED tun0
Wed May  1 22:10:38 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Wed May  1 22:10:38 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Wed May  1 22:10:38 2019 daemon.notice netifd: Network device 'tun0' link is up
Wed May  1 22:10:38 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Wed May  1 22:10:38 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Wed May  1 22:10:38 2019 daemon.notice netifd: omrvpn (14157): udhcpc: started, v1.30.1
Wed May  1 22:10:38 2019 daemon.notice netifd: omrvpn (14157): udhcpc: sending discover
Wed May  1 22:10:38 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Wed May  1 22:10:38 2019 daemon.err glorytun[13265]: VPS_IP.65001: connected
Wed May  1 22:10:38 2019 daemon.info glorytun[13265]: STARTED tun0
Wed May  1 22:10:42 2019 daemon.notice netifd: omrvpn (14157): udhcpc: sending discover
Wed May  1 22:10:45 2019 daemon.notice netifd: omrvpn (14157): udhcpc: sending discover
Wed May  1 22:10:47 2019 user.notice post-tracking-post-tracking: tun0 switched off
Wed May  1 22:10:47 2019 user.notice post-tracking-post-tracking: Delete vps (VPS_IP) route via 10.255.255.1 dev tun0
Wed May  1 22:10:47 2019 user.notice post-tracking-post-tracking: omrvpn down
Wed May  1 22:10:47 2019 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.2.1 dev wan1
Wed May  1 22:10:47 2019 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Wed May  1 22:10:47 2019 user.notice Shadowsocks: Rules DOWN

## Specifications
  - OpenMPTCProuter version: 0.48
  - OpenMPTCProuter VPS version: 0.922
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-05-01:
True there is some problems with glorytun when IPv6 is disabled via sysctl, you can also have some problems with shadowsocks.

xabix, 2019-05-02:
Ok I have reinstalled everything and it s back to normal (make sense :) ).

The best way to keep the VPS up-to-date is to re run the install script each time (instead of doing apt-get update to avoid getting a non mtcp kernel) ?
If the SSH port is changed would running again the install script will put it back to 65222? I was thinking to maybe try to change 'default' ports but if each time I get the linux packets updated through the install script everything goes back to defaults then it's an issue.

merci for the great work and support! Kuddos

Ysurac, 2019-05-02:
You can remove default debian/ubuntu kernel and update as you want, but VPS apps are for now only updated via VPS script. A script with debian repo is also in test in another issue.
Only if port is set to 22 this is modified to 65222.

xabix, 2019-05-02:
Thank you Ysurac. Very clear, I will follow your recommendations

closed 2019-05-02 by xabix