Ysurac/openmptcprouter #455: Good way to forward some traffic from VPS server to specific rule

dur3x, 2019-05-02:
What I'm doing currently is working but needed to be done every X by a cronjob because of not persistent of this rule. I don't know exactly why (probably because the vps reload sometimes the rules and then loose the old ones).
So what I'm doing is this:

in a cronjob:
iptables -t nat -I OUTPUT -p tcp -d <ip> -j DNAT --to-destination 127.0.0.1:12345
which in my example 127.0.0.1:12345 is a redsocks which is used to proxify some request blocked from my VPS.

Do you see a better way to do without the cronjob implementation ?
I seen that you are using shorewall instead of iptables which I'm not familiar to add my rule as persistent.

dur3x, 2019-05-02:
One proper way looks be to use extension scripts for shorewall (http://shorewall.net/shorewall_extension_scripts.htm)

By putting all my iptables rules inside this file `/etc/shorewall/start` it works like wantend :-)
Currently I don't think we can do that directly from the openmptcprouter interface, I mean that all the redirect/forward/bypass rules are directly for the openmptcprouter (client part) and not applied for the VPS right ?

dur3x, 2019-05-02:
In fact my request could be linked to an already existing issue about custom rules: https://github.com/Ysurac/openmptcprouter-vps/issues/16

Ysurac, 2019-05-02:
A custom redirect like you did will not be supported via OpenMPTCProuter. It's a corner use case and I you find it's quite easy to use a custom rule on shorewall

closed 2019-05-02 by Ysurac