Ysurac/openmptcprouter #456: MPTCP is not enabled on the server ??

Lemax9, 2019-05-02:
## Expected Behavior
Hello,
Anomaly noted:
- very long loading time on some pages of OpenMPTCProuter (with 192.168.100.1)
- error displayed on status page "Cant contact Server admin script (No Token)" - MPTCP is not enabled on the server

Internet is OK (but agregation, I dont know)
Thanks

## Actual Behavior

## Steps to Reproduce the Problem

  1. reboot VPS 
  2 - new installation script + reboot vps
... no change
on VPS --> uname -a ===> root@vps-xxx:~# uname -a
Linux vps-44678.fhnet.fr 4.14.110-mptcp #1 SMP Fri Apr 5 20:35:39 UTC 2019 x86_64 GNU/Linux

## Specifications

  - OpenMPTCProuter version: (Last is not a version)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: RPI3 B+
[image]

Ysurac, 2019-05-02:
I would need the configuration (remove server ip and keys) in System->OpenMPTCProuter->show all settings

Lemax9, 2019-05-02:
OK 

> dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].ipset='/iptv-……/omr_dst_bypass_all,omr6_dst_bypass_all'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_default='1'
dhcp.lan.ra_management='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='Z8'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='00:1E:B8:AA:C3:72'
dhcp.@host[0].ip='192.168.100.15'
dhcp.@host[0].gw='192.168.100.1'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='0'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
firewall.@zone[1].network='wan' 'wan6' 'wan1' 'wan2'
firewall.@zone[1].input='REJECT'
[... 942 lines cut ...]
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'

Ysurac, 2019-05-02:
All seems to be ok on this side. You can check OpenMPTCProuter logs and VPS logs.

Lemax9, 2019-05-03:
After a new install VPS and OpenMPTCProuter on Rpi 3 ... it's OK (?!  :)

Thanks

[image]

closed 2019-05-03 by Lemax9