Ysurac/openmptcprouter #458: Server not accessible in LAN

Peppe2201, 2019-05-03:
I've forwarded ports 1-64000 with NAT rule but the server is only reachable with the public IP IP and not longer working in the LAN.
Is it an iptables problem?

## Specifications

  - OpenMPTCProuter version: 0.48
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-05-03:
The server is not accessible on SSH port 65222 ?

Peppe2201, 2019-05-03:
No, the VPS in LAN is not working but with another public IP is working.

Ysurac, 2019-05-03:
I don't understand what is the problem. You can access it over SSH from LAN or not ?

Peppe2201, 2019-05-03:
The problem is that: I've configured a port forwarding of a range of ports (from 1 to 64000) from VPN zone to LAN zone with NAT Loopback enabled.
The problem is that open ports are forwarded correctly, they are reachable from other ISP, but are not accessible from LAN typing the public IP and I don't know why

Ysurac, 2019-05-03:
Routeback is not enabled in shorewall on the VPS, you can try to enable it by adding routeback to options in "/etc/shorewall/interfaces" on the VPS, and ```/etc/init.d/shorewall restart```

Peppe2201, 2019-05-03:
On which interface?

Ysurac, 2019-05-03:
You can try on all interfaces.

Peppe2201, 2019-05-03:
Adding `routeback` on all interface don't make any difference, have I to reboot the router?

Peppe2201, 2019-05-04:
Can maybe be sourceback=0 the problem?

Peppe2201, 2019-05-05:
Not working anyway, i've resolved temporarily with DNS redirect

Peppe2201, 2019-05-07:
I'm trying this https://www.the-art-of-web.com/system/iptables-nat/ but seems not working, have I to do it via shadowsocks rules?

closed 2019-05-12 by Peppe2201