Ysurac/openmptcprouter #462: OMR-Bypass Subdomains

jwthecolorist, 2019-05-06:
I hope this is in the right place as this is more of a question/suggestion.

How does OMR-Bypass handle subdomains for domains added to be bypassed?

For instance, I would like example.com and all subdomains to go out wan2 without going over the vpn or shadowsocks. I'm finding that if I place example.com in OMR-Bypass, it seems hit or miss whether the subdomains are also routed over wan2 vs the vpn/shadowsocks.

Sometimes they are, sometimes they aren't.

## Actual Behavior

When it appears to be working and I update OMR-Bypass to redirect the domain over a different wan, it breaks the subdomains completely and they go over the VPN. I have not been able to track what the solution to this is. Sometimes deleting and recreating the rule followed by a reboot fixes it. Other times I simply cannot get the subdomains to work at all (always going over the vpn/shadowsocks). 

Other times it appears to work or stop working at random. (this I'm uncertain of, still tracking behavior)

Thanks!

## Steps to Reproduce the Problem

  1. Add example.com to OMR Bypass and select a wan.
  2. Run tracert (or traceroute on Linux) to subdomain.example.com
  3. See if the result is routed over the VPN or directly over selected wan.

## Specifications

  - OpenMPTCProuter version: v0.48 r0+9745-b0395cfc56
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-05-06:
You have to use OpenMPTCProuter as DNS to make it working. IP are added to ipset for bypass when they are resolved via OpenMPTCProuter DNS, so if you have domain in client cache it's not added.

jwthecolorist, 2019-05-06:
Thanks! I should have thought of that. I had been having trouble with unbound not resolving at times so I added 8.8.8.8 as a second upstream server to dsnmasq (besides unbound's 127.0.0.1#5353) plus added 8.8.8.8 to be sent as the secondary dns server via dhcp for one of my lans. That definitely could have been messing it up. 

Still haven't determined why Unbound has trouble resolving sometimes but last night/today I noticed that if the vpn drops for some reason and tries to reconnect, sometimes it's having trouble getting a dhcp response from my vps. If I reboot the vps it fixes it, not sure what's going on there. Will have to check the logs on the vps. For now, I've set omrvpn (tun0) to have a static ip instead. Do you know offhand if doing that will break anything in your scripts?

jwthecolorist, 2019-05-06:
Quick Update: I just did another quick test and after removing 8.8.8.8 from both places and rebooting the vps and the router, it appeared to not be routing properly (going over the vpn). Then I flushed the local dns cache and that took care of it! Perfect!

Ysurac, 2019-05-06:
omrvpn static IP will be the default in next release (that will be out soon but still on 4.14 kernel).

jwthecolorist, 2019-05-07:
Perfect!

closed 2019-05-25 by Ysurac

jwthecolorist, 2019-05-27:
Just a quick update to this (sorry for re-opening):

I thought it was working and then the next day it broke. It has been hit and miss since then. Sometimes fine, sometimes not.

So last night I started over with a fresh install (moved to a different motherboard) and added my WAN's (3) and connected to my VPS. Made almost no changes outside of that. Currently running openmptcprouter v0.49.7 & 0.994.

I then added datahound.com to OMR-Bypass. My only dns server on my pc is openmptcprouter.

[image]

Flushed local dns cache.

[image]

Ran tracert to ip104-207-129-31.datahound.com.

Still going over VPS:

[image]

Here are others to try:
ip173-244-217-45.datahound.com
ip74-201-228-198.datahound.com
ip104-153-87-19.datahound.com

Not sure what else to do.. I've tried everything I can think of. I thought the next thing I might try is writing the rule myself and adding it to the firewall but I didn't want to mess up any of your rules in the process. Any insight would be greatly appreciated. It would be great if anyone else can replicate this.

If you need any other information let me know.

Ysurac, 2019-05-27:
Check that the only DNS used is the DNS from OpenMPTCProuter.

jwthecolorist, 2019-05-28:
Yes! I have been doing that. Currently, it's the only DNS server assigned by DHCP.

Any other ideas?

jwthecolorist, 2019-05-30:
Quick update in case it helps someone.

I've turned on Dnsmasq logging requests to the system log. This has been very useful. It seems specific to this domain OR this type of request, not sure. Here is what the request and reply normally looks like:

```
dnsmasq[26577]: 5 192.168.***.212/55626 query[A] ip104-153-87-12.datahound.com from 192.168.150.212

dnsmasq[26577]: 5 192.168.***.212/55626 forwarded ip104-153-87-12.datahound.com to 127.0.0.1

dnsmasq[26577]: 5 192.168.***.212/55626 ipset add omr-cs6 104.153.87.12 ip104-153-87-12.datahound.com

dnsmasq[26577]: 5 192.168.***.212/55626 reply ip104-153-87-12.datahound.com is 104.153.87.12
```
As you can see there is no `ipset add omr_dst_bypass_wan2` line in spite of datahound.com being in OMR Bypass. Very strange.

This is what it should look like:

```
dnsmasq[17033]: 16 192.168.***.212/65110 query[A] ip104-153-87-12.datahound.com from 192.168.150.212

dnsmasq[17033]: 16 192.168.***.212/65110 forwarded ip104-153-87-12.datahound.com to 127.0.0.1

dnsmasq[17033]: 16 192.168.***.212/65110 ipset add omr_dst_bypass_wan2 104.153.87.12 ip104-153-87-12.datahound.com

dnsmasq[17033]: 16 192.168.***.212/65110 ipset add omr6_dst_bypass_wan2 104.153.87.12 ip104-153-87-12.datahound.com

dnsmasq[17033]: 16 192.168.150.212/65110 reply ip104-153-87-12.datahound.com is 104.153.87.12
```

However, this rarely happens. Almost never at the moment.

Bypass routing for sub-domains of other domains I've tried seems to work fine. So I'm not sure where the issue is here.

Any ideas on how to solve this would be hugely appreciated.

jwthecolorist, 2019-06-06:
Update: Working much better now starting with version 0.50.1. Seems to accurately bind the resolved ip to the appropriate wan. The DNS query logs confirm this also. Thanks @Ysurac!

jwthecolorist, 2019-06-13:
Update+Question: On release 0.50.2, I notice Dnsmasq seems to reboot every few minutes. This appears to be something new, I hadn't noticed it on previous releases. It seems to be every time an uplink (I use 4) times out for a few seconds but I can't be certain. The largest issue I notice is that the OMR-Bypass mappings seem to be lost every time Dnsmasq reboots. 

Is this normal (the mappings lost when Dnsmasq reboots)?

Thanks.

Ysurac, 2019-06-14:
It's omr-bypass reboot that remove ipset data and reboot dnsmasq.
I wll check what I can do about that.

jwthecolorist, 2019-06-15:
Ok thanks! It would be great if the ipset data can remain even after an omr-bypass reboot.

Ysurac, 2019-06-15:
True, I make some test to keep them after a reload. This will be available in next release.

jwthecolorist, 2019-06-17:
Perfect! Thanks!