Ysurac/openmptcprouter #463: OMRVPN does not work since the last update

KoRnEr007, 2019-05-07:
Hello,
my vpn is down ... when i see the interface OMRVPN nothing is set ...
COuld you please help me to configure this ?
[image]
[image]
[image]
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.48 r0+9745-b0395cfc56
  - OpenMPTCProuter VPS version:  v0.992
  - OpenMPTCProuter platform: (/RPI3 x64)

Ysurac, 2019-05-07:
You should check system log

KoRnEr007, 2019-05-07:
Tue May  7 19:08:20 2019 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Tue May  7 19:08:20 2019 daemon.notice netifd: Network device 'tun0' link is up
Tue May  7 19:08:20 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Tue May  7 19:08:20 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Tue May  7 19:08:20 2019 daemon.notice netifd: Interface 'omrvpn' is now up
Tue May  7 19:08:20 2019 daemon.err glorytun-udp[23500]: running on device tun0 as pid 23500
Tue May  7 19:08:20 2019 kern.info kernel: [185170.267695] IPv6: ADDRCONF(NETDEV_CHANGE): tun0: link becomes ready
Tue May  7 19:08:20 2019 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Tue May  7 19:08:21 2019 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Tue May  7 19:08:21 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Tue May  7 19:08:23 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue May  7 19:08:49 2019 daemon.info dnsmasq[10524]: read /etc/hosts - 4 addresses
Tue May  7 19:08:49 2019 daemon.info dnsmasq[10524]: read /tmp/hosts/dhcp.cfg01411c - 5 addresses
Tue May  7 19:08:49 2019 daemon.info dnsmasq-dhcp[10524]: read /etc/ethers - 0 addresses
Tue May  7 19:08:49 2019 user.notice omr-bypass: OMR-ByPass is running

jwthecolorist, 2019-05-07:
Am I mistaken or is the Common Configuration for OMR-Bypass is set to unmanaged rather than DHCP Client?

Ysurac, 2019-05-07:
It's set as unmanaged in latest release. You can try to set DHCP.

KoRnEr007, 2019-05-07:
Tue May  7 19:32:14 2019 daemon.notice netifd: Network device 'tun0' link is down
Tue May  7 19:32:14 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Tue May  7 19:32:14 2019 daemon.notice netifd: Interface 'omrvpn' is now down
Tue May  7 19:32:14 2019 daemon.notice netifd: Interface 'omrvpn' is disabled
Tue May  7 19:32:14 2019 daemon.notice netifd: Interface 'omrvpn' is enabled
Tue May  7 19:32:14 2019 kern.info kernel: [186604.026288] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
Tue May  7 19:32:14 2019 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Tue May  7 19:32:14 2019 daemon.notice netifd: Network device 'tun0' link is up
Tue May  7 19:32:14 2019 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Tue May  7 19:32:14 2019 daemon.notice netifd: Interface 'omrvpn' is setting up now
Tue May  7 19:32:14 2019 daemon.err glorytun-udp[22088]: running on device tun0 as pid 22088
Tue May  7 19:32:14 2019 daemon.notice netifd: Interface 'omrvpn' is now up
Tue May  7 19:32:14 2019 kern.info kernel: [186604.111285] IPv6: ADDRCONF(NETDEV_CHANGE): tun0: link becomes ready
Tue May  7 19:32:14 2019 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Tue May  7 19:32:14 2019 daemon.info omr-tracker-ss: Shadowsocks is up (can contact 1.1.1.1)
Tue May  7 19:32:16 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue May  7 19:32:19 2019 daemon.info dnsmasq[10524]: read /etc/hosts - 4 addresses
Tue May  7 19:32:19 2019 daemon.info dnsmasq[10524]: read /tmp/hosts/dhcp.cfg01411c - 5 addresses
Tue May  7 19:32:19 2019 daemon.info dnsmasq-dhcp[10524]: read /etc/ethers - 0 addresses
Tue May  7 19:32:19 2019 user.notice omr-bypass: OMR-ByPass is running

always down ...

KoRnEr007, 2019-05-07:
and can't be set DHCP....

Ysurac, 2019-05-07:
Try to use glorytun TCP instead of glorytun UDP

KoRnEr007, 2019-05-07:
Tue May  7 19:39:29 2019 daemon.info glorytun: starting glorytun vpn instance vpn
Tue May  7 19:39:29 2019 daemon.err glorytun[31483]: couldn't create tun device
Tue May  7 19:39:59 2019 daemon.err glorytun[32204]: couldn't create tun device
Tue May  7 19:40:29 2019 daemon.err glorytun[359]: couldn't create tun device

Ysurac, 2019-05-07:
reboot openmptcprouter, there is a bug in current release.

jwthecolorist, 2019-05-07:
> It's set as unmanaged in latest release. You can try to set DHCP.

Oh ok, I see.

KoRnEr007, 2019-05-07:
look good :) thanks for the support

closed 2019-05-11 by KoRnEr007