Ysurac/openmptcprouter #464: How to manually set the best MTU per interfaces?

xabix, 2019-05-07:
## Expected Behavior

I was expecting OMR to automatically discover and set the MTU per interfaces but my MTU is set to all my 4 WANs to 1500. I checked it with "omr-mtu wan1"

## Actual Behavior

What is the best way to discover the MTU per interface?
Is it recommended to set the same lower MTU of all 4 interfaces for performance reasons or it's ok to have different MTUs per interfaces?

## Steps to Reproduce the Problem

1/ping -f -I wan1 -l **1472** google.com
PING google.com (216.58.198.206) from 192.168.2.3 wan1: 56(84) bytes of data.
268421 packets transmitted, 16340 received, **93% packet loss**, time 20641ms
rtt min/avg/max/mdev = 21.313/111.843/134.338/3.915 ms, pipe 1472, ipg/ewma 0.076/111.537 ms

2/ping -f -I wan1 -l **1452** google.com
PING google.com (216.58.198.206) from 192.168.2.3 wan1: 56(84) bytes of data.
53688 packets transmitted, 3296 received, **93% packet loss**, time 4159ms
rtt min/avg/max/mdev = 21.092/111.111/134.470/8.277 ms, pipe 1452, ipg/ewma 0.077/112.080 ms
3/ping -f -I wan1 -l **1352** google.com
PING google.com (216.58.198.206) from 192.168.2.3 wan1: 56(84) bytes of data.
48788 packets transmitted, 3217 received, **93% packet loss**, time 4047ms
rtt min/avg/max/mdev = 21.470/110.876/133.218/8.237 ms, pipe 1352, ipg/ewma 0.082/112.038 ms

## Specifications

  - OpenMPTCProuter version: v0.48 r0+9745-b0395cfc56
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: x86_64

[image]

Ysurac, 2019-05-07:
"omr-mtu" must be used with interface as first arg and OMR VPS as second arg.
What is the result ? 1500 for all interfaces ?
To test without using omr-mtu, you can use ```ping -w 1 -M do -c 1 -s 1472 -I wan1 149.91.xxxxxx```.
Some ISP/box can ban (or set to really low priority) ICMP request, this doesn't help to detect MTU.

xabix, 2019-05-07:
All omr-mtu give 1500.

> omr-mtu wan1 vps
1500
omr-mtu wan2 vps
1500
omr-mtu wan3 vps
1500
omr-mtu wan4 vps
1500

With your command it seems better to find the right MTU. Will you recommend me to find the MTU for each interface and configure it as the overlay MTU value?
root@OpenMPTCProuter:~# ping -w 1 -M do -c 1 -s **1472** -I wan1 149.91.xxx.xxx
PING 149.91..xx.xxx from 192.168.2.3 wan1: 1472(1500) bytes of data.
1480 bytes from 149.91.88.173: icmp_req=1 ttl=46 time=35.8 ms
1 packets transmitted, 1 received, **0% packet loss**, time 0ms
rtt min/avg/max/mdev = 35.887/35.887/35.887/0.000 ms

root@OpenMPTCProuter:~# ping -w 1 -M do -c 1 -s **1500** -I wan1 149.91..xxx.xxx
PING 149.91..xxx.xxx from 192.168.2.3 wan1: 1500(1528) bytes of data.
ping: local error: Message too long, mtu=1500
1 packets transmitted, 0 received, +1 errors, **100% packet loss**, time 0ms

thank you Ysurac

Ysurac, 2019-05-07:
1472+28 (20 byte (IP header) + 8 byte (ICMP header)) =1500, so it's really a 1500 MTU here.

xabix, 2019-05-07:
FYI, I did a manual discover per interface and I found:
WAN1: 1472
WAN2=WAN3=WAN4=1444

I did set them manually and rebooted but the GUI (status) doesn't show this manuel overlay values. 
I did retry the omr-mtu and I am getting 1500 for all 4 interfaces.

I am a little bit lost, I may have done something wrong or there may be a bug as the config in interface is still there so correctly saved.

xabix, 2019-05-07:
Just saw your comment. Ok so I will set 1500 for wan1 and 1444+28 for wan 2, 3 and 4 but still it should be set accordingly in the GUI no?

Ysurac, 2019-05-07:
Settings will be force set again after a while.
omr-mtu should give you same results as the command I give you. Don't forget to add 28 to value you get.

xabix, 2019-05-08:
Thank you, yes it was set after a while. MERCI
[image]

closed 2019-05-08 by xabix