Ysurac/openmptcprouter #465: udhcpc: Out of memory: Kill process netifd

xabix, 2019-05-07:
## Expected Behavior

The system seems to have issues with the memory

## Actual Behavior
I have attached the full log for more details.
[kernel.log](https://github.com/Ysurac/openmptcprouter/files/3154549/kernel.log)

`[21305.344546] Out of memory: Kill process 2978 (netifd) score 1 or sacrifice child
[21305.344800] Killed process 4420 (udhcpc) total-vm:1096kB, anon-rss:32kB, file-rss:0kB, shmem-rss:0kB`

> root@OpenMPTCProuter:~# free
              total        used        free      shared  buff/cache   available
Mem:         242840       66644      133444        1668       42752      131028

On the VPS:
> root@XabiX:~# free
              total        used        free      shared  buff/cache   available
Mem:        2099392      191312     1558144       21552      349936     1741572
Swap:             0           0           0

## Steps to Reproduce the Problem

Not sure how to reproduce it.

## Specifications

- OpenMPTCProuter version: v0.48 r0+9745-b0395cfc56
- OpenMPTCProuter VPS version: 0.992
- OpenMPTCProuter platform: x86_64

[image]

user747, 2019-05-08:
It looks like there is a mptcp issue. This possibly starts to consume memory, causing ss-redir to be killed first, since it uses the 2nd most memory. Or the mptcp error contributes some way to ss-redir consuming more memory as well, resulting in a crash. This crash seems to happen quite early in the boot process? This might make troubleshooting harder. 

` cgi-bin/luci/admin/system/openmptcprouter/debug ` has all uci settings. 

I would perform a reset of all settings on OMR. Then add 1 ADSL WAN only and see if you can reproduce crash.

Further troubleshooting if you have time or are curious: 
SSH to OMR and run htop to see where resources are at with a fresh start. Add a ADSL WAN, and test. Add a USB wan, and test. 

How much memory is allocated to the OpenWRT QEMU ? 

Some other information: 
Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.11.1-0-g0551a4be2c-prebuilt.qemu-project.org 04/01/2014
rootfstype=squashfs: version 4.0 
Kernel version: 4.14.108

You could try a newer build that has a newer Kernel, or try ext4 filesystem if things are not working. 

Also when 4.19 becomes more stable, consider `CONFIG_MPTCP_BLEST=y`  and `CONFIG_DEFAULT_MPTCP_SCHED="BLEST"`since you have a heterogeneous WAN configuration (copper/4G wireless) if you get things running and compare to default sched.

xabix, 2019-05-08:
OMR has

Attached the full config. 
[config.txt](https://github.com/Ysurac/openmptcprouter/files/3158919/config.txt)

I noted this line which includes 2 times WAN4:
firewall.@zone[1].network='wan wan1 wan2 wan3 WAN4 wan4'

OMR has 256Mb assigned and the VPS is supposed to have 2Gb

I am happy to either increase the debug level to understand the issue (meaning is it normal to have the process to crash), also to increase my RAM and if need to reset the OMR. Unless the new version with 4.19 is coming soon.

Merci

Ysurac, 2019-05-08:
256Mo is really too low for OMR. Current x86_64 image is using f2fs, so for me it's quite normal.

xabix, 2019-05-08:
What would be the recommended RAM then? Is there also a recommended installation to not use f2fs for better results? Sorry if I missed the info

Ysurac, 2019-05-08:
At least 512Mo, 1Go is better.
For now only f2fs for x86_64, ext4 will be available for next release.

xabix, 2019-05-08:
Thank you Ysurac.
I will set 1Go then and will reinstall when the next release is available.

BTW I assume I will have to use another image to avoid noise on vmware-vmsvc.log  like:

> [2019-05-08T19:51:00.910Z] [ message] [vmsvc] Log caching is enabled with maxCacheEntries=4096.
[2019-05-08T19:51:00.911Z] [ message] [vmsvc] Core dump limit set to -1
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Tools Version: 10.3.5.7752 (build-10430147)
[2019-05-08T19:51:00.911Z] [ message] [vmsvc] Cannot load message catalog for domain 'vix', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Plugin 'vix' initialized.
[2019-05-08T19:51:00.911Z] [ message] [vmsvc] Cannot load message catalog for domain 'powerops', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Plugin 'powerops' initialized.
[2019-05-08T19:51:00.911Z] [ message] [vmsvc] Cannot load message catalog for domain 'timeSync', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Plugin 'timeSync' initialized.
[2019-05-08T19:51:00.911Z] [ message] [vmsvc] Cannot load message catalog for domain 'vmbackup', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Plugin 'vmbackup' initialized.
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Unloading plugin 'vmbackup'.
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Unloading plugin 'timeSync'.
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Unloading plugin 'powerops'.
[2019-05-08T19:51:00.911Z] [ message] [vmtoolsd] Unloading plugin 'vix'.
[2019-05-08T19:51:05.913Z] [ message] [vmsvc] Log caching is enabled with maxCacheEntries=4096.
[2019-05-08T19:51:05.913Z] [ message] [vmsvc] Core dump limit set to -1
[2019-05-08T19:51:05.913Z] [ message] [vmtoolsd] Tools Version: 10.3.5.7752 (build-10430147)
[2019-05-08T19:51:05.914Z] [ message] [vmsvc] Cannot load message catalog for domain 'vix', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2019-05-08T19:51:05.914Z] [ message] [vmtoolsd] Plugin 'vix' initialized.
[2019-05-08T19:51:05.914Z] [ message] [vmsvc] Cannot load message catalog for domain 'powerops', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2019-05-08T19:51:05.914Z] [ message] [vmtoolsd] Plugin 'powerops' initialized.
[2019-05-08T19:51:05.914Z] [ message] [vmsvc] Cannot load message catalog for domain 'timeSync', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2019-05-08T19:51:05.914Z] [ message] [vmtoolsd] Plugin 'timeSync' initialized.
[2019-05-08T19:51:05.914Z] [ message] [vmsvc] Cannot load message catalog for domain 'vmbackup', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2019-05-08T19:51:05.914Z] [ message] [vmtoolsd] Plugin 'vmbackup' initialized.
[2019-05-08T19:51:05.914Z] [ message] [vmtoolsd] Unloading plugin 'vmbackup'.
[2019-05-08T19:51:05.914Z] [ message] [vmtoolsd] Unloading plugin 'timeSync'.
[2019-05-08T19:51:05.914Z] [ message] [vmtoolsd] Unloading plugin 'powerops'.
[2019-05-08T19:51:05.914Z] [ message] [vmtoolsd] Unloading plugin 'vix'.

Ysurac, 2019-05-08:
openvmtools is installed on all x86_64 and x86 images.
I don't have the issue on virtualbox.

closed 2019-05-08 by xabix

xabix, 2019-05-08:
FYI I was looking for the cpu and mem usage over a week and day and the memory usage stays very low. I increased it to 1Go and then 2Go to see if there was more memory being used but not. I general memory doesn't got higher than 120Mb thus why I set the limit to 256Mb.
[image]

Anyway I am not an expert but I will fix it to 1Go for now and monitor if the issue appears again.