Ysurac/openmptcprouter #468: Device not found afert reboot

Florent4223, 2019-05-09:
## Expected Behavior

Hello,

I wish to aggregate 2 Huawei E3342 4G keys in modem.
I configure the 4G key as a modem, the connection is established normally. On the other hand, after a reboot the interface systematically indicates that the device is not found. Usually after a Stop / start of the interface everything returns to normal

In addition each key goes on the interfce wwan0. Is it possible to put the 2nd key on another interface typically wwan1?

Thanks

## Steps to Reproduce the Problem

  1. Config E3372-153 modem
  2. connect to internet OK
  3. reboot 

## Specifications

  - OpenMPTCProuter version: 0.48
  - OpenMPTCProuter VPS version: 0.9
  - OpenMPTCProuter platform: RPI3b+

Florent4223, 2019-05-10:
[image]

dj0uleS, 2019-05-13:
Hello Florent,

I've got a small question about your Huawei E3372 configuration. I would to know how do you change the IP range of the dongle.

For the moment I use one of them with Hi-Link mode (use as DHCP in OpenMPTCP) and I will modify the dongle for use them with stick mode and NCM protocol, such as your configuration.

Your help is the welcome and I hope you solve quickly your problem !

Florent4223, 2019-05-14:
I have tried witch last version et same pb

Y a t il une solution?

Il suffit d'arreter l'interface et de la redemarrer mais y a t il une autre solution pour eviter la manipulation?

Florent4223, 2019-05-15:
> Hello Florent,
>
> I've got a small question about your Huawei E3372 configuration. I would to know how do you change the IP range of the dongle.
>
> For the moment I use one of them with Hi-Link mode [...]

Hello
The Hilink mode is for me the best compared to OMR at least until the problem to restart the Raspberry is not solved.
However to switch to Surfstick mode (NCM protocol so) you have to change the firmware of your key.
You have to have a 22.xx version, and you have to pass it in 21.xx. Personally I tried to pass one of my keys that is 21.318.xxxx 22.xxx without success. The methods on the internet to modify for me do not work.

Ysurac, 2019-05-15:
I would need OpenMPTCProuter kernel log after a reboot.

Florent4223, 2019-05-15:
`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.115 (circleci@default-a60549f1-c64a-4989-81a4-b540a65a2209) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r0+9945-bc85640cdc)) #0 SMP Mon May 6 10:43:01 2019
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] cma: Reserved 16 MiB at 0x000000003a400000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x80/0x390 with crng_init=0
[    0.000000] percpu: Embedded 19 pages/cpu s40344 r8192 d29288 u77824
[    0.000000] pcpu-alloc: s40344 r8192 d29288 u77824 alloc=19*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Memory: 928040K/970752K available (5118K kernel code, 342K rwdata, 1480K rodata, 320K init, 496K bss, 26328K reserved, 16384K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff8008080000 - 0xffffff8008580000   (  5120 KB)
[    0.000000]     .rodata : 0xffffff8008580000 - 0xffffff8008700000   (  1536 KB)
[    0.000000]       .init : 0xffffff8008700000 - 0xffffff8008750000   (   320 KB)
[    0.000000]       .data : 0xffffff8008750000 - 0xffffff80087a5808   (   343 KB)
[    0.000000]        .bss : 0xffffff80087a5808 - 0xffffff8008821850   (   497 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf00000000 - 0xffffffbf00ed0000   (    14 MB actual)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc03b400000   (   948 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	CONFIG_RCU_FANOUT set to non-default value of 32
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000194] Console: colour dummy device 80x25
[    0.000679] console [tty1] enabled
[    0.000711] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000742] pid_max: default: 32768 minimum: 301
[    0.000894] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000930] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.002014] ASID allocator initialised with 65536 entries
[    0.002129] Hierarchical SRCU implementation.
[    0.002698] smp: Bringing up secondary CPUs ...
[    0.003109] Detected VIPT I-cache on CPU1
[    0.003162] CPU1: Booted secondary processor [410fd034]
[    0.003717] Detected VIPT I-cache on CPU2
[    0.003753] CPU2: Booted secondary processor [410fd034]
[    0.004242] Detected VIPT I-cache on CPU3
[    0.004276] CPU3: Booted secondary processor [410fd034]
[    0.004394] smp: Brought up 1 node, 4 CPUs
[    0.004476] SMP: Total of 4 processors activated.
[    0.004495] CPU features: detected feature: 32-bit EL0 Support
[    0.004517] CPU: All CPU(s) started at EL2
[    0.004550] alternatives: patching kernel code
[    0.005331] devtmpfs: initialized
[    0.014166] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.014216] futex hash table entries: 1024 (order: 5, 131072 bytes)
[    0.014960] pinctrl core: initialized pinctrl subsystem
[    0.015266] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.015815] NET: Registered protocol family 16
[    0.019671] cpuidle: using governor ladder
[    0.019710] cpuidle: using governor menu
[    0.020004] vdso: 2 pages (1 code @ ffffff8008584000, 1 data @ ffffff8008754000)
[    0.021425] DMA: preallocated 256 KiB pool for atomic allocations
[... 701 lines cut ...]
[   14.371840] huawei_cdc_ncm 1-1.2:1.2: resetting NTB format to 16-bit
[   14.381474] huawei_cdc_ncm 1-1.2:1.2: MAC-Address: 00:1e:10:1f:00:00
[   14.388825] huawei_cdc_ncm 1-1.2:1.2: setting rx_max = 16384
[   14.403350] huawei_cdc_ncm 1-1.2:1.2: NDP will be placed at end of frame for this device.
[   14.413664] huawei_cdc_ncm 1-1.2:1.2: cdc-wdm0: USB WDM device
[... cut ...]
[   14.649186] 8021q: adding VLAN 0 to HW filter on device wan2
[   14.658851] cdc_ether 1-1.1.3:1.0 eth1: kevent 12 may have been dropped
[   14.676900] IPv6: ADDRCONF(NETDEV_CHANGE): wan1: link becomes ready
[   14.684637] IPv6: ADDRCONF(NETDEV_CHANGE): wan2: link becomes ready
[   15.529248] scsi 1:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
[   15.540280] sd 1:0:0:0: no of_node; not parsing pinctrl DT
[   15.558888] sd 1:0:0:0: [sdb] Attached SCSI removable disk
`

Thanks

Florent4223, 2019-05-15:
On Usb0 => ttyUSB0 = E3372 mode NCM (modem)

on ttyUSB1 : nothing

on eth1 : E3372 en mode hilink = mode router

Florent4223, 2019-05-15:
After restarting the 4g connection via the modem. The sgragment does not work

I noticed compared my other issue on the non-functional aggregation, that as long as I did not go into the wizard and save without modification, everything goes through my eth1 and nothing by usb0

Florent4223, 2019-05-16:
Is there a solution to restart the interface in SSH (as if I had stopped / connect via the web interface)
Ifdown and ifup is not enough

Also always in ssh, can we launch the validation of the wizard with shadowsock off?

Ysurac, 2019-05-16:
I would need system log too when you start 4g connection. The ifup should run again mptcp script that should display a line "Reloading mptcp config due to... of ..."

Florent4223, 2019-05-16:
I installed version 0.49.2
And after rebooting the RPI3, usually I got the message that the device is not found. I stop it and try to restart it but it does not work

Only solution: remove the interface and recreate it, b ut as he was not connected I have again stop / restart the connection and and he has connected well

I have the line that it reloads the config of the mptcp but when I am forced to remove the interface I do not see if it is really something or not since I am obliged to reconfigure the interface in the wizard and the mptc

note that my USB modem interface is now called LTE0001

log system for each step :

`Thu May 16 10:12:09 2019 daemon.notice netifd: LTE0001 (14864): Stopping network LTE0001
Thu May 16 10:12:10 2019 daemon.notice netifd: LTE0001 (14864): sending ->
Thu May 16 10:12:13 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Thu May 16 10:12:14 2019 daemon.notice netifd: Interface 'LTE0001' is now down
Thu May 16 10:12:14 2019 daemon.notice netifd: Interface 'LTE0001' is setting up now
Thu May 16 10:12:17 2019 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 16 10:12:18 2019 daemon.notice netifd: LTE0001 (15355): SIM ready
Thu May 16 10:12:18 2019 daemon.notice netifd: LTE0001 (15355): PIN set successfully
Thu May 16 10:12:18 2019 daemon.notice netifd: LTE0001 (15355): Configuring modem
Thu May 16 10:12:18 2019 daemon.notice netifd: LTE0001 (15355): WARNING: Variable 'modes' does not exist or is not an array/object
Thu May 16 10:12:18 2019 daemon.notice netifd: LTE0001 (15355): Setting mode
Thu May 16 10:12:18 2019 daemon.notice netifd: LTE0001 (15355): sending ->
Thu May 16 10:12:19 2019 daemon.info dnsmasq[10977]: read /etc/hosts - 4 addresses
Thu May 16 10:12:19 2019 daemon.info dnsmasq[10977]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu May 16 10:12:19 2019 daemon.info dnsmasq-dhcp[10977]: read /etc/ethers - 0 addresses
Thu May 16 10:12:19 2019 user.notice omr-bypass: OMR-ByPass is running
Thu May 16 10:12:20 2019 user.notice Shadowsocks: Rules UP
Thu May 16 10:12:20 2019 user.notice Shadowsocks: Reload omr-bypass rules
Thu May 16 10:12:20 2019 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 16 10:12:22 2019 daemon.info dnsmasq[10977]: read /etc/hosts - 4 addresses
Thu May 16 10:12:22 2019 daemon.info dnsmasq[10977]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu May 16 10:12:22 2019 daemon.info dnsmasq-dhcp[10977]: read /etc/ethers - 0 addresses
Thu May 16 10:12:22 2019 user.notice omr-bypass: OMR-ByPass is running
Thu May 16 10:12:43 2019 daemon.notice netifd: LTE0001 (15355): Timeout running AT-command
Thu May 16 10:12:43 2019 daemon.notice netifd: LTE0001 (15355): Failed to set operating mode
Thu May 16 10:12:43 2019 daemon.notice netifd: LTE0001 (17157): Stopping network LTE0001
Thu May 16 10:12:44 2019 daemon.notice netifd: LTE0001 (17157): sending ->
Thu May 16 10:12:48 2019 daemon.notice netifd: Interface 'LTE0001' is now down
Thu May 16 10:12:48 2019 daemon.notice netifd: Interface 'LTE0001' is setting up now
Thu May 16 10:12:51 2019 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 16 10:12:52 2019 daemon.notice netifd: LTE0001 (17281): SIM ready
Thu May 16 10:12:52 2019 daemon.notice netifd: LTE0001 (17281): PIN set successfully
Thu May 16 10:12:52 2019 daemon.notice netifd: LTE0001 (17281): Configuring modem
Thu May 16 10:12:52 2019 daemon.notice netifd: LTE0001 (17281): WARNING: Variable 'modes' does not exist or is not an array/object
Thu May 16 10:12:52 2019 daemon.notice netifd: LTE0001 (17281): Setting mode
Thu May 16 10:12:52 2019 daemon.notice netifd: LTE0001 (17281): sending ->
Thu May 16 10:12:53 2019 daemon.info dnsmasq[10977]: read /etc/hosts - 4 addresses
Thu May 16 10:12:53 2019 daemon.info dnsmasq[10977]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu May 16 10:12:53 2019 daemon.info dnsmasq-dhcp[10977]: read /etc/ethers - 0 addresses
Thu May 16 10:12:53 2019 user.notice omr-bypass: OMR-ByPass is running
Thu May 16 10:12:54 2019 user.notice Shadowsocks: Rules UP
Thu May 16 10:12:54 2019 user.notice Shadowsocks: Reload omr-bypass rules
Thu May 16 10:12:54 2019 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 16 10:12:54 2019 daemon.err omr-tracker-ss[14556]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Thu May 16 10:12:56 2019 daemon.info dnsmasq[10977]: read /etc/hosts - 4 addresses
Thu May 16 10:12:56 2019 daemon.info dnsmasq[10977]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu May 16 10:12:56 2019 daemon.info dnsmasq-dhcp[10977]: read /etc/ethers - 0 addresses
Thu May 16 10:12:56 2019 user.notice omr-bypass: OMR-ByPass is running
Thu May 16 10:13:18 2019 daemon.notice netifd: LTE0001 (17281): Timeout running AT-command
Thu May 16 10:13:18 2019 daemon.notice netifd: LTE0001 (17281): Failed to set operating mode
Thu May 16 10:13:18 2019 daemon.notice netifd: LTE0001 (19004): Stopping network LTE0001
Thu May 16 10:13:18 2019 daemon.notice netifd: LTE0001 (19004): sending ->
Thu May 16 10:13:23 2019 daemon.notice netifd: Interface 'LTE0001' is now down
Thu May 16 10:13:23 2019 daemon.notice netifd: Interface 'LTE0001' is setting up now
Thu May 16 10:13:25 2019 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 16 10:13:26 2019 daemon.notice netifd: LTE0001 (19143): SIM ready
Thu May 16 10:13:26 2019 daemon.notice netifd: LTE0001 (19143): PIN set successfully
[... cut ...]
Thu May 16 10:22:01 2019 user.notice Shadowsocks: Rules UP
Thu May 16 10:22:02 2019 user.notice Shadowsocks: Reload omr-bypass rules
Thu May 16 10:22:02 2019 user.notice omr-bypass: Starting OMR-ByPass...
Thu May 16 10:22:06 2019 daemon.info dnsmasq[10977]: read /etc/hosts - 4 addresses
Thu May 16 10:22:06 2019 daemon.info dnsmasq[10977]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Thu May 16 10:22:06 2019 daemon.info dnsmasq-dhcp[10977]: read /etc/ethers - 0 addresses
Thu May 16 10:22:06 2019 user.notice omr-bypass: OMR-ByPass is running`

Ysurac, 2019-05-17:
I never tested NCM. When reading OpenWrt doc https://openwrt.org/docs/guide-user/network/wan/wwan/ethernetoverusb_ncm I can see a delay can be set, this may solve some issue.
The issue can also be that the device take too much time to get an IP, so MPTCP script is executed before an IP is available. For that I may add a delay.

Florent4223, 2019-05-21:
For me there are two problems:
one I think is directly related to raspberry. What do you think is possible that the USB ports will power the USB key AFTER the connection request is made?
If it does, it can not see the device, knowing that the E3372 is long itself to start
Is it possible to add a time to see if the device is present?

On the other hand yes I could see that the connection of the modem can take several seconds (or almost 1 minute) to connect

closed 2019-07-04 by Ysurac