Ysurac/openmptcprouter #469: Instability with OMR on my virtual machine Synology

advandced, 2019-05-09:
## Observation

When i tried to load a web page, i have this error occured randomly (like picture down below)

[image]

This error append when i use OMR with Synology Virtual Machine Manager, but with my laptop, everything run fine.

## Actual Behavior
In system logs i have theses lines :
Thu May  9 19:21:32 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:32 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:34 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:34 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:34 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:34 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:37 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:38 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:38 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:38 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:38 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:38 2019 daemon.err /usr/bin/ss-local[16497]: remote_recv_cb_recv: Connection reset by peer
Thu May  9 19:21:40 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:40 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:40 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:40 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:40 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:40 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:41 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:41 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:42 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:45 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:46 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:46 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:46 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:47 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:47 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:52 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:52 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:55 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:21:56 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:57 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:57 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:57 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:57 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:57 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:57 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:57 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:21:57 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:58 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:58 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:21:58 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:21:59 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:22:00 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:22:01 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:22:03 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:22:03 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:22:04 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:22:04 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:22:05 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer
Thu May  9 19:22:05 2019 daemon.err /usr/bin/ss-redir[16500]: remote recv: Connection reset by peer
Thu May  9 19:22:05 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:22:05 2019 daemon.err /usr/bin/ss-redir[16501]: remote recv: Connection reset by peer
Thu May  9 19:22:05 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:22:06 2019 daemon.err /usr/bin/ss-redir[16499]: remote recv: Connection reset by peer
Thu May  9 19:22:06 2019 daemon.err /usr/bin/ss-redir[16498]: remote recv: Connection reset by peer

## Steps to Reproduce the Problem

Just load anything in web browser

## Specifications

  - OpenMPTCProuter version: 0.48
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: Synology Virtual Machine Manager

Ysurac, 2019-05-09:
There is a connection problem. Interface need to allow promiscuous mode, I have no idea if there is a setting for that on Synology.

advandced, 2019-05-15:
This instability appeared later on this VM.
Now i update to the version  0.49 & 0.993, and now it's stable.

There is a command, to know if the machine have this promiscuous mode allowed ?

closed 2019-05-25 by Ysurac