Ysurac/openmptcprouter #470: OpenVPN site-to-site UDP only uses master WAN

uryupinsk, 2019-05-09:
## Expected Behavior

Hello,

I have an OpenVPN UDP site-to-site tunnel. It needs to connect while aggregating all WAN available.

## Actual Behavior

While it connects successfully via Glorytun TCP, the aggregation doesn't work. It only uses the master WAN. When I switch to OpenVPN TCP, it works correctly via all WAN available. 

Maybe the problem is not isolated to OpenVPN but to all UDP connections, but I would need some instructions to verify this.

I am not able to switch to Glorytun UDP or MLVPN because the tunnel won't start between OpenMPTCProuter and the VPS.

## Steps to Reproduce the Problem

  1. Open an OpenVPN UDP tunnel via OpenMPTCProuter using Glorytun TCP
  2. Download or upload a file through the VPN

## Specifications

  - OpenMPTCProuter version: 0.48 r0+9745-b0395cfc56
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-05-09:
You can use ```omr-test-speed tun0``` to test Glorytun TCP speed. Even if slower than Shadowsocks, this aggregate speed. You may have to check MSS settings of OpenVPN.

uryupinsk, 2019-05-09:
After some research, I found the problem.

I reinstalled OpenMPTCProuter yesterday to restart config from scratch, and forgot to set time to UTC+2, as I am located in France. Thus, the clock between the router and the VPS was not in sync, and Glorytun refused to start the tunnel to prevent replay attacks.

The tunnel is now working successfully.

Sorry about that!

closed 2019-05-09 by uryupinsk