Ysurac/openmptcprouter #471: Glorytun TCP Speed & HTTPS content timeouts over Shadowsocks

jwthecolorist, 2019-05-10:
## Expected Behavior

 This is a combination of 2 issues actually. When loading an https page, Chrome tends to say "Establishing Secure Connection" for quite a while and many times parts of the page are very slow to load or do not load. For some content (usually video content) I have to reload the page a few times and it will finally connect. Once connected, the video plays fine unless the internet drops.

HTTP content tends to load very fast though.

## Actual Behavior

I've tried disabling Fast Open and it makes no difference. I found that if I set ss-redir for TCP to unset, the issue goes away as I'm running completely on Glorytun TCP at that point. Pages load as expected and fast as well. The downside is that Glorytun TCP seems to top out at 2Mbps both down and up. My main connections vary speed wise but generally are:

ADSL: 10 Mbps down, 0.75 Mbps up
4G LTE: Average 10-25 MBps down, 10-25Mbps up
3G: Average 8-10 Mbps down, 4Mbps up

When using shadowsocks and I start a download, I can usually see all 3 connections almost completely saturated. WIth Glorytun TCP, it tops out 1-2Mbps max. I can see it is definitely using MPTCP but just not saturating the connections like Shadowsocks does.

I tried adding the average speeds on the wizard page but it seems to make no difference. Glorytun UDP is out as UDP appears to be highly throttled.

## Steps to Reproduce the Problem

  1. Set ss-redir for TCP to unset.
  2. Verify traffic running over Glorytun.
  3. Run a speedtest of some form (Fast.com etc).

## Specifications

  - OpenMPTCProuter version: v0.48 r0+9745-b0395cfc56
  - OpenMPTCProuter VPS version: 0.992 (I'm currently testing 0.993 on one VPS)
  - OpenMPTCProuter platform: x86_64

suyuan168, 2019-05-10:
me too
This problem has been bothering me for a long time.

suyuan168, 2019-05-10:
Me too
This problem has been bothering me for a long time.
All https pages are not open very slowly, and must be refreshed many times. Sometimes CSS does not load, sometimes the image does not load.

user747, 2019-05-10:
Do you notice any speed increase if you restart networking on VPS ? 
`sudo systemctl restart systemd-networkd.service`

With `ss-redir for TCP <unset>` my aggregated speed cut in half, and latency increased by 10ms. After restarting networking on VPS, speed test was faster, but still latency and jitter. Probably best to use iperf3 for more accurate bench marking. 

I think this is known that Glorytun TCP is a little bit slower aggregating TCP traffic compared to SS. Last commit for Glorytun TCP branch was  [2 years ago ](https://github.com/angt/glorytun/commit/0a797b4f5fbbe5030abfec89d678236016146cb4)

suyuan168, 2019-05-10:
The speed has not increased. But glorytunvpn does not open the website abnormally.

jwthecolorist, 2019-05-11:
I just tested again now and tried restarting networkd after setting unset (Glorytun TCP only) but sadly it didn't make a difference. The HTTPS timeout appears to be gone but the throughput is extremely low, around 1-2Mbps max.

Any other ideas? Anything I might be doing wrong or that I can change with ShadowSocks?

If you need any further information just let me know.

Thanks!

user747, 2019-05-12:
Could try a newer generated build of OMR. Maybe even rebuild VPS. 

Don't add all your WAN's at once though. Try one, test HTTPS. Try a different build of Chrome or Incognito , or Firefox. Try a wired client and a wireless client and see if any difference.

Try disabling Unbound in OMR and set DNS server to 8.8.8.8. Run tcpdump -i any port not 22 and see what the traffic is doing. Or capture to .pcap and look at in Wireshark.

jwthecolorist, 2019-05-12:
Thanks for the feedback! I actually just upgraded to the latest version of OMR and VPS this evening:

VPS: 0.993 4.14.110-mptcp
OMR: v0.49 r0+9945-bc85640cdc

I have been testing across multiple WAN's either one at a time or together etc over the past few weeks. Different computers as well. That being said they probably all have the same version of Chrome so I could test another browser. Most of the time if the page won't load, the error returned is:

ERR_SSL_VERSION_INTERFERENCE

It just happened a few moments ago so I can see it's still happening even after upgrading to the latest versions. I've even tried multiple VPS across 2 different providers (Vultr & HOSTFLYTE).

I do recall having tested in Firefox with the same result but that has been many weeks so I could try again.

I did try disabling Unbound in OMR and set DNS server to 8.8.8.8 on Dnsmasq awhile back as I was having DNS issues. I also tried using 8.8.8.8 on the local machine (bypassing Dnsmasq/Unbound) and both tests helped DNS but the https timeouts remained. I've since restored the configuration to using Unbound and DNS is operating fine at the moment.

You're totally right about it possibly being a configuration issue. That's why I was curious if anyone else had the same problem and had found the fix. I'm going to keep testing to see if I can fix it locally.

user747, 2019-05-12:
"data corruption in your TCP packets, leading to failures when SSL handshaking. It maybe caused by some broken NAT devices in your environment."

https://github.com/shadowsocks/shadowsocks-libev/issues/2067#issuecomment-393389797

suyuan168, 2019-05-12:
I have tested with @Ysurac. I canceled fastopen and tried to cancel the default gateway and DNS on each WAN interface. I also tested the VPS script version and the orm version. But there is no way to open the page normally. I have tested this problem on RPI2 3 or bpi-r2 or x86. I open the ORM to grab network packets and many TCP retransmissions are lost with TCP. TCP out-of-order, I think the network is blocked, causing the sequence packet arrival time to be unreachable, the delay is too long or the packet is lost, and the data unit needs to be reassembled. Many errors include tcp retransmission or segment lost. But I think the main reason is the multi-network path between client and server. It's like a framework like load balance, so if two packets go in different paths, the packets sent later will arrive one step earlier than the packets sent earlier, and out-of-order will occur.

suyuan168, 2019-05-12:
https://github.com/shadowsocks/shadowsocks/issues/1446

suyuan168, 2019-05-12:
I don't know if I am right or not, I hope to help everyone.
Our current situation is that the previous vps, behind multiple wan, this is actually NAT, when the request reaches vps, it will be forwarded to the backend server after modifying the address data, but will not modify the timestamp data, for the backend For the server, the source address of the request is the address of the vps, and the port will be reused. Therefore, from the perspective of the back-end server, the original request of different clients may be considered as the same connection after being forwarded by vps. The time of different clients may be inconsistent, so there will be a timestamp disorder, so the following data packets are discarded. The specific performance is usually the SYN sent by the client, but the server does not respond to the ACK.

user747, 2019-05-12:
[Shadowsocks is designed to avoid any kind of handshake. Given shadowsocks works as a TCP proxy, RST to close a connection is just okay.](https://github.com/shadowsocks/shadowsocks-libev/issues/2364#issuecomment-491115497)

user747, 2019-05-13:
TLS and SSL issues are sometimes linked MTU. I know Glorytun has some sort of auto MTU mode, where Shadowsocks might be set to 1500.

Wonder if changing your WAN's MTU will fix things.

suyuan168, 2019-05-13:
I tested that it is not feasible to manually modify the mtu to 1350 or 1462 in each wan. There will still be page loads that are slowly refreshed many times. I also modified the configuration file in ORM and vps /etc/sysctl.d/ to close tcp open fast. Closing the results is not feasible. Still not solving my current problem. But it feels like shutting down the ss agent. All the problems with Glorytun are solved. I am also wondering if the MTU should be set up correctly.

user747, 2019-05-13:
> I tested that it is not feasible to manually modify the mtu to 1350 or 1462 in each wan. 

Yeah you would have to `Override MTU` for each WAN interface. Which is really just a virtual macvlan to forward traffic from your real WAN NIC. 

My real WAN's used DSL PPPoE with LLC, which adds additional overhead of 8 bytes = 2 (PPP) + 6 (PPPoE) 

[Since Ethernet has a maximum payload size of 1500 octets, the PPPoE header is 6 octets and the PPP Protocol ID is 2 octets, the PPP MTU MUST NOT be greater than 1492.](https://tools.ietf.org/html/rfc2516)

I need to learn more about what happens when real WAN 1492 comes into a 1500 macvlan.

closed 2019-07-04 by Ysurac