Ysurac/openmptcprouter #472: Agregation on glorytun

Florent4223, 2019-05-10:
## Expected Behavior

Agregation on VPN Glorytun

## Actual Behavior

Use on a single connection. **The connection switches from one to the other**
I have two 4G connection: one by 4G usb key and the other via wifi on a smartphone
[image]

wwan0 (USB0)
root@OpenMPTCProuter:~# omr-test-speed wwan0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 23.1M    0     0  2509k      0  1:09:39  0:00:09  1:09:30 2495k

wlan0:
root@OpenMPTCProuter:~# omr-test-speed wlan0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 8600k    0     0  1021k      0  2:51:00  0:00:08  2:50:52  885k

filezilal client : 
[image]

wwan0
[image]

wlan0
[image]

## Steps to Reproduce the Problem

  1. Serveur FTP install on VPS server
  2. transfer file to ftp on address 10.25.255.1
  3. rate recorded on the transfer

## Specifications

  - OpenMPTCProuter version: 0.48
  - OpenMPTCProuter VPS version: 0.994
  - OpenMPTCProuter platform: RPI3

Thanks for your help

Ysurac, 2019-05-13:
I'm not able to reproduce the issue, you should try latest release.

Florent4223, 2019-05-14:
I have tried with last version vps 0.993 et omr 0.49
same pb

I saw that we could disable shadowssock and go through glorytun

Is it possible to have my manipulation precise because when I disable shadowsock, I have more internet behind OMR.
However, the VPN is always OK

thank you

Florent4223, 2019-05-16:
Sorry I found what I did not do when I disabled shadowsock, I did not put on "unset" udp redirection
now everything goes well by vpn and everything works fine when i disable shadowsock

By cons I confirm what I suspected in my other issue, After a restart for the aggregation between 2 USB 4G, after having returned functional, I am obliged to return to the wizard and save without modification for it revive the set
Otherwise the connection uses only one link

Florent4223, 2019-05-21:
since last release no problem 
aggregate works fine after reboot