Ysurac/openmptcprouter #473: Coupures régulières

bosseb, 2019-05-10:
OpenMPTCProuter version: 0.48
OpenMPTCProuter VPS version: 0.992
OpenMPTCProuter platform: NUC INTEL/x86_64)

Bonsoir
pour faire suite à notre échange sur le forum la fibre, j'ai de nombreuses coupures, j'ai désactivé tous les DHCP des box et de tous les switch et points d'accès wifi, changé tous mes cables mais les coupures continuent
ci dessous capture de la page STATUS
[image]
ci dessous tous les settings
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].ipset='/iptv-pure.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/m6web.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/authserver.mojang.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/premium-visioniptv.myeffect.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/ev-fw.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/yggtorrent.gg/omr_dst_bypass_all,omr6_dst_bypass_all' '/install-pure.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/f.iptv-pure.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_default='0'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='22'
etherwake.setup=etherwake
etherwake.setup.pathes='/usr/bin/etherwake /usr/bin/ether-wake'
etherwake.setup.sudo='off'
etherwake.setup.broadcast='off'
etherwake.@target[0]=target
etherwake.@target[0].name='example'
etherwake.@target[0].mac='11:22:33:44:55:66'
etherwake.@target[0].password='AABBCCDDEEFF'
etherwake.@target[0].wakeonboot='off'
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='ACCEPT'
firewall.@defaults[0].output='ACCEPT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].disable_ipv6='1'
firewall.@zone[0]=zone
firewall.@zone[0].name='lan'
firewall.@zone[0].network='lan'
firewall.@zone[0].input='ACCEPT'
firewall.@zone[0].output='ACCEPT'
firewall.@zone[0].forward='ACCEPT'
firewall.@zone[1]=zone
firewall.@zone[1].name='wan'
[... 1051 lines cut ...]
upnpd.config.enabled='1'
upnpd.config.uuid='e249ee19-d10d-4f18-a3e0-417415f3cf36'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[0].ext_ports='442-65535'
upnpd.@perm_rule[0].int_ports='442-65535'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2' 'wan3' 'wan1' 'wan2' 'wan3' 'wan1' 'wan2' 'wan3' 'wan1' 'wan2' 'wan3' 'wan3' 'wan3'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='pci0000:00/0000:00:1c.5/0000:3a:00.0'
wireless.radio0.channel='auto'
wireless.radio0.htmode='HT40'
wireless.radio0.country='00'
wireless.radio0.legacy_rates='1'
wireless.radio0.hwmode='11g'
wireless.radio0.__toggle='Désactive'

Ysurac, 2019-05-10:
La configuration semble bonne. Il faut voir dans le journal système ce qu'il y a quand tu as une coupure.

bosseb, 2019-05-10:
il n' y a rien à paramétrer par rapport au switch où j'ai rassemblé mes arrivées des 3 box + la RJ45 qui va au nuc et celle qui va au switch répartiteur de la maison  ?

Ysurac, 2019-05-10:
non

bosseb, 2019-05-10:
Je vais donc surveiller le journal système quand il y a coupure, merci à toi :)

bosseb, 2019-05-10:
Une coupure à nouveau voici le journal avant et pendant
Fri May 10 23:07:09 2019 daemon.info dnsmasq-dhcp[5517]: DHCPREQUEST(eth0) 192.168.100.125 90:dd:5d:a2:23:67
Fri May 10 23:07:09 2019 daemon.info dnsmasq-dhcp[5517]: DHCPACK(eth0) 192.168.100.125 90:dd:5d:a2:23:67 Chambre
Fri May 10 23:12:35 2019 daemon.info dnsmasq-dhcp[5517]: DHCPREQUEST(eth0) 192.168.100.125 90:dd:5d:a2:23:67
Fri May 10 23:12:35 2019 daemon.info dnsmasq-dhcp[5517]: DHCPACK(eth0) 192.168.100.125 90:dd:5d:a2:23:67 Chambre
Fri May 10 23:13:25 2019 daemon.info dnsmasq-dhcp[5517]: DHCPREQUEST(eth0) 192.168.100.125 90:dd:5d:a2:23:67
Fri May 10 23:13:25 2019 daemon.info dnsmasq-dhcp[5517]: DHCPACK(eth0) 192.168.100.125 90:dd:5d:a2:23:67 Chambre
Fri May 10 23:14:46 2019 daemon.info dnsmasq-dhcp[5517]: DHCPDISCOVER(eth0) 90:3a:e6:15:be:12
Fri May 10 23:14:46 2019 daemon.info dnsmasq-dhcp[5517]: DHCPOFFER(eth0) 192.168.100.157 90:3a:e6:15:be:12
Fri May 10 23:14:46 2019 daemon.info dnsmasq-dhcp[5517]: DHCPREQUEST(eth0) 192.168.100.157 90:3a:e6:15:be:12
Fri May 10 23:14:46 2019 daemon.info dnsmasq-dhcp[5517]: DHCPACK(eth0) 192.168.100.157 90:3a:e6:15:be:12

Ysurac, 2019-05-10:
Si ton ordi est "chambre" essaye de le mettre en IP fixe, sinon il n'y a aucune coupure détectée au niveau d'OpenMPTCProuter.

bosseb, 2019-05-10:
ok ça marche merci

bosseb, 2019-05-11:
une autre coupure sur apple tv
Sat May 11 08:06:35 2019 daemon.notice netifd: omrvpn (4201): udhcpc: sending renew to 10.255.255.1
Sat May 11 08:08:10 2019 daemon.err uhttpd[4443]: luci: accepted login on / for root from 192.168.100.240

Ysurac, 2019-05-11:
C'est que l'AppleTV qui coupe ?
Sinon aucune coupure détectée dans le log non plus la.

bosseb, 2019-05-11:
et une dernière pour la route
Sat May 11 08:29:38 2019 daemon.info dnsmasq-dhcp[5517]: DHCPDISCOVER(eth0) 90:3a:e6:15:be:12
Sat May 11 08:29:38 2019 daemon.info dnsmasq-dhcp[5517]: DHCPOFFER(eth0) 192.168.100.157 90:3a:e6:15:be:12
Sat May 11 08:29:38 2019 daemon.info dnsmasq-dhcp[5517]: DHCPREQUEST(eth0) 192.168.100.157 90:3a:e6:15:be:12
Sat May 11 08:29:38 2019 daemon.info dnsmasq-dhcp[5517]: DHCPACK(eth0) 192.168.100.157 90:3a:e6:15:be:12

bosseb, 2019-05-11:
oui coupure en cours de visionage sur mycanal sur apple tv

Ysurac, 2019-05-11:
Passe l'Apple TV en IP fixe. Sinon toujours pas de coupure, juste des requêtes DHCP. Après si il y a des tonnes de requête DHCP c'est qu'un matériel provoque un DHCP flood qui sature le routeur.

bosseb, 2019-05-11:
mes 2 apple tv sont déjà en ip fixe, je vais chercher le matos qui sature, je le trouve je suppose par son ip dans le journal ?

Ysurac, 2019-05-11:
oui IP ou son adresse MAC. Si tu as d'autres matos Apple, commence par la.

bosseb, 2019-05-11:
Mon iphone, j'ai son ip mais je vois pas comment lui imposer une ip fixe

Ysurac, 2019-05-11:
Tu peux déjà essayer de le déconnecter du réseau pour voir si ça change quelque chose ou pas.

bosseb, 2019-05-11:
le tel est déconnecté du réseau et j'ai tjrs des coupures sur l'apple tv

Ysurac, 2019-05-11:
Il faut voir au niveau du log du VPS peut être via ```dmsg``` ou dans /var/log

bosseb, 2019-05-11:
dans /var/log j'ai syslog c'est cela que je dois regarder ?

Ysurac, 2019-05-11:
oui et daemon.log si il est présent.

bosseb, 2019-05-11:
apparemment y a un décalage de 2 heures sur les log, pour la coupure de ce matin vers 8h29 ça donne ça sur le daemon 
May  5 06:27:36 xxxxxxxxx[1]: Starting Daily apt upgrade and clean activities...
May  5 06:27:40 xxxxxxxxx[1]: Started Daily apt upgrade and clean activities.
May  5 06:27:40 xxxxxxxxx[1]: apt-daily-upgrade.timer: Adding 27min 15.760769s random time.
May  5 06:27:40 xxxxxxxxx[1]: apt-daily-upgrade.timer: Adding 14min 46.483157s random time.

bosseb, 2019-05-11:
et pour syslog à la même heure
ay 11 06:25:26 xxxxxx[613]:  2019-05-11 06:25:26 ERROR: getpeername: Transport endpoint is not connected
May 11 06:25:26 sxxx[613]:  2019-05-11 06:25:26 ERROR: getpeername: Transport endpoint is not connected
May 11 06:25:34 xxxx[613]:  2019-05-11 06:25:34 ERROR: getpeername: Transport endpoint is not connected
May 11 06:25:36 xxxxx[640]:  2019-05-11 06:25:36 ERROR: getpeername: Transport endpoint is not connected
May 11 06:33:27 xxxxxx[613]:  2019-05-11 06:33:27 ERROR: getpeername: Transport endpoint is not connected
May 11 06:33:27 xxxxxx[640]:  2019-05-11 06:33:27 ERROR: getpeername: Transport endpoint is not connected

Ysurac, 2019-05-11:
L'erreur signifie qu'un site n'est pas accessible par le serveur.

Ysurac, 2019-05-11:
Le VPS est sur Ubuntu ou sur Debian ? Je m'étonne pour les "Reached target Shutdown.", "Stopping User Manager",... Je vois pas ce qui pourrait provoquer ça surtout sans prévenir.

bosseb, 2019-05-11:
debian sur Scaleway

bosseb, 2019-05-11:
et re à l'instant

May 11 15:56:11 nom-du-serveur[1]: Stopped User Manager for UID 0.
May 11 15:56:11 nom-du-serveur[1]: Removed slice User Slice of root.
May 11 15:59:32 scw-nervous-clarke ss-server[640]:  2019-05-11 15:59:32 ERROR: server recv: Connection reset by peer
May 11 16:01:02 scw-nervous-clarke ss-server[613]:  2019-05-11 16:01:02 ERROR: getpeername: Transport endpoint is not connected
May 11 16:04:00 scw-nervous-clarke ss-server[640]:  2019-05-11 16:04:00 ERROR: getpeername: Transport endpoint is not connected
May 11 16:06:12 nom-du-serveur[1]: Created slice User Slice of root.
May 11 16:06:12 nom-du-serveur[1]: Started Session 336 of user root.
May 11 16:06:12 nom-du-serveur[1]: Starting User Manager for UID 0...
May 11 16:06:12 nom-du-serveur[15152]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
May 11 16:06:12 nom-du-serveur[15152]: Listening on GnuPG cryptographic agent and passphrase cache.
May 11 16:06:12 nom-du-serveur[15152]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
May 11 16:06:12 nom-du-serveur[15152]: Reached target Paths.
May 11 16:06:12 nom-du-serveur[15152]: Listening on GnuPG cryptographic agent (access for web browsers).
May 11 16:06:12 nom-du-serveur[15152]: Listening on GnuPG network certificate management daemon.
May 11 16:06:12 nom-du-serveur[15152]: Reached target Sockets.
May 11 16:06:12 nom-du-serveur[15152]: Reached target Timers.
May 11 16:06:12 nom-du-serveur[15152]: Reached target Basic System.
May 11 16:06:12 nom-du-serveur[15152]: Reached target Default.
May 11 16:06:12 nom-du-serveur[15152]: Startup finished in 11ms.
May 11 16:06:12 nom-du-serveur[1]: Started User Manager for UID 0.
May 11 16:06:53 nom-du-serveur[1]: Started Session 338 of user root.

bosseb, 2019-05-11:
May 11 15:56:14 nom-du-serveur kernel: [163982.208928] IPv4: martian source 10.255.255.1 from 17.130.74.5, on dev gt-tun0
May 11 15:56:29 nom-du-serveur kernel: [163997.064090] IPv4: martian source 10.255.255.1 from 17.253.113.15, on dev gt-tun0
May 11 15:59:32 nom-du-serveur ss-server[640]:  2019-05-11 15:59:32 ERROR: server recv: Connection reset by peer
May 11 16:01:02 nom-du-serveur ss-server[613]:  2019-05-11 16:01:02 ERROR: getpeername: Transport endpoint is not connected
May 11 16:04:00 nom-du-serveur ss-server[640]:  2019-05-11 16:04:00 ERROR: getpeername: Transport endpoint is not connected
May 11 16:05:01 nom-du-serveur CRON[14854]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)
May 11 16:06:12 nom-du-serveur systemd[1]: Created slice User Slice of root.
May 11 16:06:12 nom-du-serveur systemd[1]: Started Session 336 of user root.
May 11 16:06:12 nom-du-serveur systemd[1]: Starting User Manager for UID 0...
May 11 16:06:12 nom-du-serveur systemd[15152]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
May 11 16:06:12 nom-du-serveur systemd[15152]: Listening on GnuPG cryptographic agent and passphrase cache.
May 11 16:06:12 nom-du-serveur systemd[15152]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
May 11 16:06:12 nom-du-serveur systemd[15152]: Reached target Paths.
May 11 16:06:12 nom-du-serveur systemd[15152]: Listening on GnuPG cryptographic agent (access for web browsers).
May 11 16:06:12 nom-du-serveur systemd[15152]: Listening on GnuPG network certificate management daemon.
May 11 16:06:12 nom-du-serveur systemd[15152]: Reached target Sockets.
May 11 16:06:12 nom-du-serveur systemd[15152]: Reached target Timers.
May 11 16:06:12 nom-du-serveur systemd[15152]: Reached target Basic System.
May 11 16:06:12 nom-du-serveur systemd[15152]: Reached target Default.
May 11 16:06:12 nom-du-serveur systemd[15152]: Startup finished in 11ms.
May 11 16:06:12 nom-du-serveur systemd[1]: Started User Manager for UID 0.
May 11 16:06:53 nom-du-serveur systemd[1]: Started Session 338 of user root.
May 11 16:08:38 nom-du-serveur ss-server[640]:  2019-05-11 16:08:38 ERROR: server recv: Connection reset by peer
May 11 16:08:41 nom-du-serveur ss-server[613]:  2019-05-11 16:08:41 ERROR: remote recv: Connection reset by peer
May 11 16:09:50 nom-du-serveur kernel: [164798.436260] IPv4: martian source 10.255.255.1 from 17.253.109.17, on dev gt-tun0
May 11 16:09:51 nom-du-serveur kernel: [164799.431761] IPv4: martian source 10.255.255.1 from 17.253.109.17, on dev gt-tun0

bosseb, 2019-05-11:
je vais essayé un autre vps

bosseb, 2019-05-12:
pas mieux avec un autre vps, j'ai finalement tout réinstaller, vps et routeur, et sur la partie routeur je suis passé sur la version 32 bits ....jusqu'ici tout bien ....

bosseb, 2019-05-12:
les coupures repartent de plus belles,

Ysurac, 2019-05-12:
A part en regardant la TV, tu vois les coupures comment ?

bosseb, 2019-05-12:
alors pour les coupures oui cela se coupe sur mycanal, et je partage mon plex et l'on me dit que la lecture à distance  se coupe, de plus les enfants qui jouent en réseau sur pc me signalent des pings de malades .... et des coupures

Ysurac, 2019-05-12:
Essaye de mettre à jour le VPS en 0.993 et le routeur en 0.49, on sait jamais.

bosseb, 2019-05-12:
c'est déjà fait depuis quelques minutes ....

Ysurac, 2019-05-12:
Tu peux tenter de changer dans réseau->MPTCP le congestion crontrol en olia puis cubic.

bosseb, 2019-05-12:
pas mieux, j'ai tjrs des messages comme quoi il n'arrive pas à joindre le serveur script admin du vps ou que MPTCP n'est pas activé

Ysurac, 2019-05-12:
Oui enfin ces messages n'indiquent pas une coupure, juste un temps de réponse un peu lent.

bosseb, 2019-05-12:
Je vais essayer à nouveau avec olia puis cubic

bosseb, 2019-05-14:
depuis hier soir je suis passé sur cubic car des coupures sur mycanal avec olia ....

bosseb, 2019-05-17:
cela semble stable depuis plusieurs jour, j'ai faites toutes les MAJ que tu as lancé ... merci pour l'aide

bosseb, 2019-06-16:
les coupures reviennent, cette fois j'ai remarqué que c'est que lorsqu'un appareil est allumé que cela génère une coupure sur MYCANAL
exemple je regarde mycanal au salon sur appletv (dhcp désactivé et ip fixe) ma femme allume son tel : coupure; je regarde mycanal au salont sur appletv ma femme allume l'appletv de la chambre et du coup coupure sur my canal ....

Ysurac, 2019-06-16:
Surement car ils utilisent la même IP ;)

bosseb, 2019-06-16:
non bien evidemment je fais tjrs gaffe à cela 
HUAWEI_Mate_20_Pro-4a17ea	192.168.100.174
APPLE-TV-CHAMBRE 192.168.100.223
APPLETV-SALON 192.168.100.124

Ysurac, 2019-06-16:
c'est que mycanal ou coupure réseau ?
Et sinon comme d'hab faut voir dans le log système si il y a quelque chose à ce moment la.

bosseb, 2019-06-16:
mycanal

bosseb, 2019-06-17:
donc je t'envoie un log système quand MYCANAL se coupe ?

Ysurac, 2019-06-17:
oui

closed 2019-07-04 by Ysurac