Ysurac/openmptcprouter #475: Upgrading to 0.49 doesn't boot: fs error

xabix, 2019-05-12:
## Expected Behavior

Hello,

I have downloaded the image openmptcprouter-v0.49-r09945-bc85640cdc-x86-64-combined-ext4.img.gz and the SHA256 is the same 10B1D97ADF8E779F302B26B0177F6F024EEAF2DB07058DA2AEC3075DF228C7E7 as the one published on the website but i am getting fs errors for more than 5 minutes when the VM is rebooting. Is this normal? I was on a squashfs or VMware image before. Could that be the issue?

## Actual Behavior
I was running openmptcprouter v0.48 r0+9745-b0395cfc56 x86

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.48 r0+9745-b0395cfc56 going to 0.49
  - OpenMPTCProuter VPS version: 0.992
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-05-12:
If you updated from the OpenMPTCProuter interface, yes this can be the issue.

xabix, 2019-05-12:
i am trying to reinstall but can't get to convert the img to an iso or into a Proxmox local disk file. With limited internet now it gets more complex :).

Can I follow this and create one file?

> gunzip openmptcprouter-*.img.gz
dd bs=4M if=openmptcprouter-*.img of=/dev/sdX conv=fsync

It seems we can maybe do it locally with
qm importdisk ${ID_VM} xenial.img local

I will try that and report if it works.

xabix, 2019-05-12:
Got it working after a

> qemu-img convert openmptcprouter-v0.49-r09945-bc85640cdc-x86-64-combined-ext4.img -O qcow2 OpenMTCP-disk1.qcow2

and to resize the disk afterwards.

Maybe I could have done a dd directly in a raw file.

closed 2019-05-12 by xabix