Ysurac/openmptcprouter #476: Issues switching from Glorytun TCP to UDP

uryupinsk, 2019-05-13:
## Expected Behavior

When switching from Glorytun TCP to Glorytun UDP, the system should:

1. Restart the tunnel automatically.
2. In OpenMPTCProuter, change the OMRVPN subnet from 10.255.255.0/24 to 10.255.254.0/24.
3. In VPS, in `/etc/shorewall/params.vpn`:

    1. Change `VPS_ADDR` and `OMR_ADDR` to reflect the change of subnet.
    2. Change `VPS_IFACE` to `gt-udp-tun0`.

## Actual Behavior

It should restart the tunnel automatically but:

1. I have to stop and start Glorytun manually via the web GUI for it to restart the tunnel.
2. After that, the tunnel is starting correctly and UDP packets are routed through it.

It should do port forwarding from VPS to OpenMPTCProuter, but it doesn't work:

1. In OpenMPTCProuter, in `System > Glorytun`, `localip` and `remoteip` are not changed by the system to reflect the change of subnet and cannot be set manually.
2. In OpenMPTCProuter, in `Network > Interfaces`, OMRVPN has an IP in the 10.255.254.0/24 subnet allocated by DHCP, instead of `Unmanaged` when it is working.
3. In VPS, in `/etc/shorewall/params.vpn`, nothing has changed. When manually set, then clear and restart `shorewall`, the file resets itself but I don't know how.
4. In VPS, when I set manually the port forwarding rule with the IP of OpenMPTCProuter in the 10.255.254.0/24 subnet, it doesn't work.

## Steps to Reproduce the Problem

Switch from Glorytun TCP to Glorytun UDP in `System > OpenMPTCProuter > Advanced settings cas > VPN settings > Default VPN`.

## How I got this solved

I was on OpenMPTCProuter 0.48 and VPS 0.992 when this happened. Yesterday, I updated to 0.49 and 0.993 respectively.

I restored my OpenMPTCProuter config to 0.49, with Glorytun UDP already configured, then installed the VPS scripts v0.993 on a clean install. On a clean VPS install, the scripts triggered the correct settings for Glorytun UDP and all is working correctly, including port forwarding with all the file previously mentioned modified accordingly.

I think there is a bug in your scripts when you change the tunnel type on an already configured and running VPS, but I wasn't competent enough to find where and why.

## Specifications

  - OpenMPTCProuter version: 0.49 r0+9945-bc85640cdc
  - OpenMPTCProuter VPS version: 0.993
  - OpenMPTCProuter platform: x86_64

user747, 2019-05-13:
Some other Glorytun-UDP observations (that might be isolated to my configuration / lack of understanding)

- After changing to default to Glorytun-UDP,  the 'mptcp' checkbox remains checked in luci (http://192.168.100.1/cgi-bin/luci/admin/services/glorytun/settings/vpn)  I think it gets turned off in init though https://github.com/Ysurac/openmptcprouter-feeds/blob/1fced47334fa9779ae025a07a7eb9fbc038e0c97/glorytun-udp/init#L94

- On the topic of Glorytun-UDP init, there is a small difference of `config_foreach start_instance mud`
I'm not sure if this is needed to use mud. Without this, are we using mud? 

https://github.com/Ysurac/openmptcprouter-feeds/blob/1fced47334fa9779ae025a07a7eb9fbc038e0c97/glorytun-udp/init#L102-L105

https://github.com/ovh/overthebox-feeds/blob/61f61ca3474e715713d9c94cbdc721c1d6ff4ddf/glorytun/init#L68-L70

- After setting up, if you do `tcpdump -i any | grep UDP` I would assume all UDP traffic would be going over 65001 at this point. During my brief test though, I did see some going through 65101. Again, could be my configuration is broken.

Ysurac, 2019-05-13:
I'm not able to reproduce the issue when changing from glorytun TCP to Glorytun UDP using the wizard. Did you reboot the VPS after update ?
VPS_IFACE is modified if glorytun UDP is working, there is a script on the VPS that check if the VPN answer on each interface that can be used.

MPTCP checkbox remains checked but this is an ignored setting for glorytun UDP.

I'm not using the same init and config as OVH, no problem here.

Shadowsocks UDP is only used when VPN is down, but some Shadowsocks internal traffic can use it.

uryupinsk, 2019-05-13:
I just saw that 6 days ago, you pushed a commit named [Fix Glorytun UDP (23177d9)](https://github.com/Ysurac/openmptcprouter-vps-admin/commit/23177d99d00a9a7e94cd3e1eb63f60e587f92e07) on `openmptcprouter-vps-admin` that changes the Glorytun UDP IPs to the correct subnet.

Maybe there are no problems now because it is this script that is used in the VPS v0.993, and the v0.992 used the script prior to this commit?

Ysurac, 2019-05-13:
True

uryupinsk, 2019-05-14:
I guess I can close this issue as you're not able to reproduce it since your change.

Thanks!

closed 2019-05-14 by uryupinsk