Ysurac/openmptcprouter #478: Problem with Lastest build

ThibautBascoulergue, 2019-05-14:
## Expected Behavior

I just installed version 0.49 with a vps in 0.993 and I router does not connect to the vps.

[image]

## Actual Behavior

## Steps to Reproduce the Problem

Install new vps with script
Install new router on Esxi

## Specifications

  - OpenMPTCProuter version: (0.49)
  - OpenMPTCProuter VPS version: (0.993)
  - OpenMPTCProuter platform: (x86_64)

ThibautBascoulergue, 2019-05-14:
With old mptcprouter in version 0.47.6, it work with the latest vps verrsion 0.993. So i think the problem is in the openmptcprouter.
[image]

Ysurac, 2019-05-14:
WAN1 is in same network  as OpenMPTCProuter and lan. It's often not a good idea.

ThibautBascoulergue, 2019-05-14:
ok i check

ThibautBascoulergue, 2019-05-14:
Ysurac,  pi-hole script will be installed after the vps script or before ?

Ysurac, 2019-05-14:
after

ThibautBascoulergue, 2019-05-14:
ok thx

ThibautBascoulergue, 2019-05-14:
The problem appears as soon as I install pi-hole.
Once the script is launched, the router no longer dialogs with the vps: /
After running the script I have some error msg
[image]

Ysurac, 2019-05-14:
Pi-hole seems to be not installed. How do you run the script ?

ThibautBascoulergue, 2019-05-14:
I have launch vps script and reboot.
After i have launch the script : wget -O - http://www.openmptcprouter.com/server/omr-pihole.sh | sh

Ysurac, 2019-05-14:
Strange, on Ubuntu or Debian ?
You can relaunch the script.

ThibautBascoulergue, 2019-05-14:
Debian

ThibautBascoulergue, 2019-05-14:
same problem after réinstalle vps
[image]

Ysurac, 2019-05-15:
What is the provider ? I'm not able to reproduce the issue that seems to be a crash of Pi-hole official install script...

ThibautBascoulergue, 2019-05-15:
vps is on Scaleway.

ThibautBascoulergue, 2019-05-15:
Some news :)

With a new VPS, not Scaleway provider, pi-hole is working good :)

So the problem comes from the sclaeway vps.

closed 2019-05-16 by ThibautBascoulergue