Ysurac/openmptcprouter #479: Problem with 2 USB-Ethernet 4G

Florent4223, 2019-05-15:
## Expected Behavior

2x interface USB / ethernet with dongle 4G Huawei E3372h on eth1 and eth2

## Actual Behavior

the first dongle 4G is OK
The 2nd key inserted eth2 is well created but no dialogue with the key 4G

## Steps to Reproduce the Problem

  1. insert first dongle usb 4g, eth1 OK
  2. insert 2eme dongle usb 4g, eth2 created
  3. no dialog eth2 et dongle 4G

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.993
  - OpenMPTCProuter platform: RPI3B+

Log Core:

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.115 (circleci@default-a60549f1-c64a-4989-81a4-b540a65a2209) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r0+9945-bc85640cdc)) #0 SMP Mon May 6 10:43:01 2019
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] cma: Reserved 16 MiB at 0x000000003a400000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x80/0x390 with crng_init=0
[    0.000000] percpu: Embedded 19 pages/cpu s40344 r8192 d29288 u77824
[    0.000000] pcpu-alloc: s40344 r8192 d29288 u77824 alloc=19*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
[    0.000000] Memory: 928040K/970752K available (5118K kernel code, 342K rwdata, 1480K rodata, 320K init, 496K bss, 26328K reserved, 16384K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff8008080000 - 0xffffff8008580000   (  5120 KB)
[    0.000000]     .rodata : 0xffffff8008580000 - 0xffffff8008700000   (  1536 KB)
[    0.000000]       .init : 0xffffff8008700000 - 0xffffff8008750000   (   320 KB)
[    0.000000]       .data : 0xffffff8008750000 - 0xffffff80087a5808   (   343 KB)
[    0.000000]        .bss : 0xffffff80087a5808 - 0xffffff8008821850   (   497 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf00000000 - 0xffffffbf00ed0000   (    14 MB actual)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc03b400000   (   948 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	CONFIG_RCU_FANOUT set to non-default value of 32
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000193] Console: colour dummy device 80x25
[    0.000678] console [tty1] enabled
[    0.000710] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000741] pid_max: default: 32768 minimum: 301
[    0.000893] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.000930] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
[    0.002011] ASID allocator initialised with 65536 entries
[... 782 lines cut ...]
[  253.645960] be60: 0000000121759ccd 0000000000000000 0000000000000002 ffffff800882bef0
[  253.655648] be80: 0000000000000700 0000000000000000 000000002e8680c7 00000000b7ff08d5
[  253.665340] bea0: 000000005d96f680 fd84bc9182e94352 ffffff800816d1c8 0000007fbab674ac
[  253.675012] bec0: 9f59220ddfb253b9 ffffff8008758828 ffffff8008737008 ffffff8008758000
[  253.684702] bee0: 0000000000000000 0000000000000000 ffffffc038ca8000 b22218080452ae90
[  253.694425] bf00: d443f404cbc559e0 da95fc00dd9960c9 f20603003d01bbf0 ffffff800882bf70
[  253.704261] bf20: ffffff800808475c ffffff800882bf70 ffffff8008084760 0000000000000145
[  253.714204] bf40: b22218080452ae90 d443f404cbc559e0 ffffffffffffffff ffffff80080e16fc
[  253.724205] bf60: ffffff800882bf70 ffffff8008084760
[  253.730185] [<ffffff8008082870>] el1_irq+0xb0/0x140
[  253.736163] [<ffffff8008084760>] arch_cpu_idle+0x10/0x18
[  253.742556] [<ffffff80080c7fbc>] do_idle+0xec/0x1b8
[  253.748476] [<ffffff80080c81c8>] cpu_startup_entry+0x20/0x28
[  253.755164] [<ffffff800808c23c>] secondary_start_kernel+0x17c/0x188
[  253.762455] ---[ end trace c015cac59ac6bc86 ]---
[  253.768193] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  253.768193] 
[  253.784228] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  253.784228] 
[  253.800186] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  253.800186] 
[  253.816039] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  253.816039] 
[  253.845472] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  253.845472]

dnwk, 2019-05-15:
I have similar issue with 2 LTE modules via USB. One of the interface will be unstable and keeps hang up.

ThibautBascoulergue, 2019-05-17:
Maybe power problem on usb port.

Have you tested whether you have connected the 4G USB-Ethernet port to a USB-powered hub?

Ysurac, 2019-05-17:
Should work in OpenMPTCProuter v0.49.3

Florent4223, 2019-05-21:
very thanks

It's fine

closed 2019-05-21 by Florent4223