Ysurac/openmptcprouter #480: ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

suyuan168, 2019-05-16:
[   91.590109] usb 1-1.1.2.2: Product: HUAWEI_MOBILE
[   91.595586] usb 1-1.1.2.2: Manufacturer: HUAWEI_MOBILE
[   91.601759] usb 1-1.1.2.2: no of_node; not parsing pinctrl DT
[   91.606034] cdc_ether 1-1.1.2.2:1.0: no of_node; not parsing pinctrl DT
[   91.611702] cdc_ether 1-1.1.2.2:1.0 eth1: register 'cdc_ether' at usb-3f980000.usb-1.1.2.2, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[   91.731149] usb 1-1.1.2.3: new high-speed USB device number 8 using dwc_otg
[   91.883209] usb 1-1.1.2.3: New USB device found, idVendor=12d1, idProduct=14db
[   91.891439] usb 1-1.1.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   91.900819] usb 1-1.1.2.3: Product: HUAWEI_MOBILE
[   91.906599] usb 1-1.1.2.3: Manufacturer: HUAWEI_MOBILE
[   91.913206] usb 1-1.1.2.3: no of_node; not parsing pinctrl DT
[   91.917568] cdc_ether 1-1.1.2.3:1.0: no of_node; not parsing pinctrl DT
[   91.923302] cdc_ether 1-1.1.2.3:1.0 eth2: register 'cdc_ether' at usb-3f980000.usb-1.1.2.3, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[   92.051155] usb 1-1.1.2.4: new high-speed USB device number 9 using dwc_otg
[   92.203297] usb 1-1.1.2.4: New USB device found, idVendor=12d1, idProduct=14db
[   92.211840] usb 1-1.1.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   92.221637] usb 1-1.1.2.4: Product: HUAWEI_MOBILE
[   92.227293] usb 1-1.1.2.4: Manufacturer: HUAWEI_MOBILE
[   92.233723] usb 1-1.1.2.4: no of_node; not parsing pinctrl DT
[   92.237966] cdc_ether 1-1.1.2.4:1.0: no of_node; not parsing pinctrl DT
[   92.243846] cdc_ether 1-1.1.2.4:1.0 eth3: register 'cdc_ether' at usb-3f980000.usb-1.1.2.4, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[  140.030499] 8021q: adding VLAN 0 to HW filter on device eth0
[  140.040033] 8021q: adding VLAN 0 to HW filter on device wan1
[  140.050599] 8021q: adding VLAN 0 to HW filter on device wan2
[  142.412508] 8021q: adding VLAN 0 to HW filter on device wan1
[  142.552691] 8021q: adding VLAN 0 to HW filter on device wan2
[  143.652823] 8021q: adding VLAN 0 to HW filter on device wan1
[  144.913082] 8021q: adding VLAN 0 to HW filter on device wan2
[  170.232772] cdc_ether 1-1.1.2.2:1.0 eth1: kevent 11 may have been dropped
[  170.240545] cdc_ether 1-1.1.2.2:1.0 eth1: kevent 12 may have been dropped
[  205.862102] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
[  206.083010] cdc_ether 1-1.1.2.3:1.0 eth2: kevent 12 may have been dropped
[  206.091014] cdc_ether 1-1.1.2.3:1.0 eth2: kevent 11 may have been dropped
[  206.301396] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  206.301396] 
[  207.991154] raspberrypi-firmware soc:firmware: Get Throttled mailbox call failed
[  210.635615] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  210.643435] cdc_ether 1-1.1.2.3:1.0 eth2: kevent 12 may have been dropped
[  210.821253] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  210.821253] 

I inserted Huawei E8372 three as wan and then it failed or even crashed.
Rpi3 does not display kernel information in the system state.

  - OpenMPTCProuter version:  v0.49.2 r0+9945-bc85640)
  - OpenMPTCProuter VPS version: (0.933)
  - OpenMPTCProuter platform: (RPI3)

suyuan168, 2019-05-16:
[    8.115261] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.147845] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt failed with error -2
[    8.161385] brcmfmac mmc1:0001:1: Falling back to user helper
[    9.862268] firmware brcm!brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt: firmware_loading_store: map pages failed
[    9.970937] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.990716] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
[   10.034625] brcmfmac mmc1:0001:3: no of_node; not parsing pinctrl DT
[   10.034770] usbcore: registered new interface driver brcmfmac
[   10.042523] usbcore: registered new interface driver ipw
[   10.048708] usbserial: USB Serial support registered for IPWireless converter
[   10.057827] usbcore: registered new interface driver option
[   10.064507] usbserial: USB Serial support registered for GSM modem (1-port)
[   10.073875] usbcore: registered new interface driver qcserial
[   10.080812] usbserial: USB Serial support registered for Qualcomm USB modem
[   10.105285] usbcore: registered new interface driver rt2800usb
[   10.128350] usbcore: registered new interface driver rtl8192cu
[   10.135612] kmodloader: done loading kernel modules from /etc/modules.d/*
[   13.721279] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   13.721308] 8021q: adding VLAN 0 to HW filter on device eth0
[   13.736338] br-lan: port 1(eth0) entered blocking state
[   13.742971] br-lan: port 1(eth0) entered disabled state
[   13.770203] br-lan: port 1(eth0) entered blocking state
[   13.776467] br-lan: port 1(eth0) entered forwarding state
[   15.239209] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   15.248136] br-lan: port 2(wlan0) entered blocking state
[   15.254582] br-lan: port 2(wlan0) entered disabled state
[   15.747144] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   15.754659] br-lan: port 2(wlan0) entered blocking state
[   15.760897] br-lan: port 2(wlan0) entered forwarding state
[   49.151247] usb 1-1.1.3: new high-speed USB device number 5 using dwc_otg
[   49.285759] usb 1-1.1.3: New USB device found, idVendor=2109, idProduct=2813
[   49.293731] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   49.302732] usb 1-1.1.3: Product: USB2.0 Hub
[   49.307815] usb 1-1.1.3: Manufacturer: VIA Labs, Inc.
[   49.314169] usb 1-1.1.3: no of_node; not parsing pinctrl DT
[   49.321140] hub 1-1.1.3:1.0: no of_node; not parsing pinctrl DT
[   49.328107] hub 1-1.1.3:1.0: USB hub found
[   49.333407] hub 1-1.1.3:1.0: 4 ports detected
[   49.701111] usb 1-1.1.3.1: new high-speed USB device number 6 using dwc_otg
[   49.852827] usb 1-1.1.3.1: New USB device found, idVendor=2109, idProduct=2813
[   49.860881] usb 1-1.1.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   49.870074] usb 1-1.1.3.1: Product: USB2.0 Hub
[   49.875316] usb 1-1.1.3.1: Manufacturer: VIA Labs, Inc.
[   49.881571] usb 1-1.1.3.1: no of_node; not parsing pinctrl DT
[   49.888575] hub 1-1.1.3.1:1.0: no of_node; not parsing pinctrl DT
[   49.895701] hub 1-1.1.3.1:1.0: USB hub found
[   49.901024] hub 1-1.1.3.1:1.0: 4 ports detected
[   50.021209] usb 1-1.1.3.2: new high-speed USB device number 7 using dwc_otg
[   50.173371] usb 1-1.1.3.2: New USB device found, idVendor=12d1, idProduct=14db
[   50.181438] usb 1-1.1.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   50.190510] usb 1-1.1.3.2: Product: HUAWEI_MOBILE
[   50.196050] usb 1-1.1.3.2: Manufacturer: HUAWEI_MOBILE
[   50.202564] usb 1-1.1.3.2: no of_node; not parsing pinctrl DT
[   50.213777] cdc_ether 1-1.1.3.2:1.0: no of_node; not parsing pinctrl DT
[   50.227168] cdc_ether 1-1.1.3.2:1.0 eth1: register 'cdc_ether' at usb-3f980000.usb-1.1.3.2, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[   50.228165] cdc_ether 1-1.1.3.2:1.0 eth1: kevent 12 may have been dropped
[   50.362026] usb 1-1.1.3.3: new high-speed USB device number 8 using dwc_otg
[   50.514988] usb 1-1.1.3.3: New USB device found, idVendor=12d1, idProduct=14db
[   50.523294] usb 1-1.1.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   50.532899] usb 1-1.1.3.3: Product: HUAWEI_MOBILE
[   50.538735] usb 1-1.1.3.3: Manufacturer: HUAWEI_MOBILE
[   50.547293] usb 1-1.1.3.3: no of_node; not parsing pinctrl DT
[   50.558750] cdc_ether 1-1.1.3.3:1.0: no of_node; not parsing pinctrl DT
[   50.581289] cdc_ether 1-1.1.3.3:1.0 eth2: register 'cdc_ether' at usb-3f980000.usb-1.1.3.3, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[   50.582450] cdc_ether 1-1.1.3.3:1.0 eth2: kevent 12 may have been dropped
[   50.603303] cdc_ether 1-1.1.3.3:1.0 eth2: kevent 11 may have been dropped
[   50.721188] usb 1-1.1.3.4: new high-speed USB device number 9 using dwc_otg
[   50.791420] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[   50.791420] 
[   50.874552] usb 1-1.1.3.4: New USB device found, idVendor=12d1, idProduct=14db
[   50.883168] usb 1-1.1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   50.892804] usb 1-1.1.3.4: Product: HUAWEI_MOBILE
[   50.898656] usb 1-1.1.3.4: Manufacturer: HUAWEI_MOBILE
[   50.905146] usb 1-1.1.3.4: no of_node; not parsing pinctrl DT
[   50.916366] cdc_ether 1-1.1.3.4:1.0: no of_node; not parsing pinctrl DT
[... cut ...]
[   50.968057] cdc_ether 1-1.1.3.4:1.0 eth3: kevent 11 may have been dropped
[   51.131379] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[   51.131379] 
[   51.991164] raspberrypi-firmware soc:firmware: Get Throttled mailbox call failed
[   79.322858] ERROR::handle_hc_chhltd_intr_dma:2214: handle_hc_chhltd_intr_dma: Channel 7, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06600001
[   79.322858]

Ysurac, 2019-05-16:
I found a workaround for this issue. I test it and will be available asap.

suyuan168, 2019-05-16:
Thank you very much, good night.

closed 2019-05-17 by suyuan168

alfs, 2020-01-27:
Hi @Ysurac, I have the same problem. Can you explain the workaround that you found?

Thanks in advance, Stefan