Ysurac/openmptcprouter #483: How to upgrade OMR from the command line?

xabix, 2019-05-17:
## Expected Behavior

Hello,

Since I have OMR 0,49 in ext4, I can't anymore upgrade to 0,49,2 or ,3 through the web interface has it takes so much time to upload that the web session times out.

## Actual Behavior

Therefore I am looking to easier way to upgrade through the cmd line with something like wget + a specific local command.

## Steps to Reproduce the Problem

  1. Reboot OMR in 0.49, 
  2. go to Flash operations and 
  3. click Flash Image

## Specifications

  - OpenMPTCProuter version:  v0.49 r0+9945-bc85640cdc
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-05-17:
This shouldn't take so much time.
But you can put the image on /tmp and ```sysupgrade -v imagename```

xabix, 2019-05-17:
Excellent, it works like a charm (quicker and easier than through the web).

Well job and support as always!

closed 2019-05-17 by xabix