Ysurac/openmptcprouter #484: Can't get public IP address from ShadowSocks Can't contact Server Admin Script (No answer) MPTCP is not enabled on the server Version 0.993 4.14.110-mptcp

suyuan168, 2019-05-17:
Can't get public IP address from ShadowSocks
Can't contact Server Admin Script (No answer)
MPTCP is not enabled on the server
Version 0.993 4.14.110-mptcp
 
My VPS script is 0.933
I am using rpi3b+ version is 0.943 latest version

Default ORM status display
Can't get public IP address from ShadowSocks
Can't contact Server Admin Script (No answer)
MPTCP is not enabled on the server
Version 0.993 4.14.110-mptcp
And extremely unstable, my wan used 5. The network is very good.
But I ping the external network will show the ping packet loss timeout. But I am sure my network is very good.
Is this version unstable?
Thank you very much for the administrator.

jayanta525, 2019-05-17:
Same issue on VPS with last VPS script (0.993). Ubuntu18.04.

suyuan168, 2019-05-17:
> Same issue on VPS with last VPS script (0.993). Ubuntu18.04.

I am the latest version of debian vps. Very unstable.

jayanta525, 2019-05-17:
The solution now is to use 0.992 VPS script.

suyuan168, 2019-05-17:
> The solution now is to use 0.992 VPS script.

Nǐ chóngxīn ānzhuāng xìtǒngle ma?
10/5000
Have you reinstalled the system?

ThibautBascoulergue, 2019-05-17:
Have you installed pi-hole on vps or not ?

jayanta525, 2019-05-17:
Installed VPS script 0.992. Aggregation works fine. clone the 0.992 tree from github and run the script. pi-hole not installed.

suyuan168, 2019-05-17:
> Installed VPS script 0.992. Aggregation works fine. clone the 0.992 tree from github and run the script. pi-hole not installed.

Hello there. How to download 0.992 script on github? Thank you

jayanta525, 2019-05-17:
UPDATE: Install VPS script 0.993 and install pi-hole instantly after that. That should solve the issue.

suyuan168, 2019-05-17:
> pi-hole

I don't need pi-hole, it doesn't make any sense to me. It would be nice if it could be installed selectively.

Thank you for your answer

Ysurac, 2019-05-17:
As always, I would need the configuration.

suyuan168, 2019-05-17:
> As always, I would need the configuration.

thank you very much. I hope I can correct it.
Greetings

suyuan168, 2019-05-17:
I have just tested it.
Install pi-pi-hole after installing VPS 0.993 and follow the instructions to not enable pi-hole firewall. Sometimes there is no problem. unstable. There are often dropped calls and no network. And the ORM state will prompt Can't get public IP address from ShadowSocks
Version 0.993 4.14.110-mptcp
Load: 0.00 0.00 0.00
Uptime: 0h 30m 26s
thank.

Ysurac, 2019-05-17:
As I ask in previous post, I need OpenMPTCProuter config in System->OpenMPTCProuter->Full settings
I will also need OpenMPTCProuter system log when there is problem.

suyuan168, 2019-05-17:
Yes, set it up exactly as the article says.
I have just tested it.
Install pi-pi-hole after installing VPS 0.993 and follow the instructions to not enable pi-hole firewall. Sometimes there is no problem. unstable. There are often dropped calls and no network. And the ORM state will prompt Can't get public IP address from ShadowSocks
Version 0.993 4.14.110-mptcp
Load: 0.00 0.00 0.00
Uptime: 0h 30m 26s
Thank you.
And modify the settings such as the proxy settings in the advanced settings. Or any other settings. Will cause SS to not link properly. Repeated start and stop. The system log is as follows:
Fri May 17 17:42:08 2019 user.info : using tcp fast open
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5166]: enable TCP no-delay
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5166]: initializing ciphers... chacha20
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5164]: listening at 0.0.0.0:1100
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5166]: listening at 0.0.0.0:1100
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5166]: tcp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5166]: UDP relay enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5164]: tcp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5164]: UDP relay enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5166]: udp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5166]: running from root user
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5164]: udp port reuse enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5164]: running from root user
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-local[5163]: using tcp fast open
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-local[5163]: enable TCP no-delay
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-local[5163]: initializing ciphers... chacha20
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-local[5163]: listening at 127.0.0.1:1111
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-local[5163]: tcp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-local[5163]: udprelay enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-local[5163]: udp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-local[5163]: running from root user
Fri May 17 17:42:08 2019 user.info : using tcp fast open
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5165]: enable TCP no-delay
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5165]: resolving hostname to IPv6 address first
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5165]: initializing ciphers... chacha20
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5165]: listening at 0.0.0.0:1100
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5165]: tcp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5165]: UDP relay enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5165]: udp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5165]: running from root user
Fri May 17 17:42:08 2019 user.info : using tcp fast open
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5167]: enable TCP no-delay
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5167]: initializing ciphers... chacha20
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5167]: listening at 0.0.0.0:1100
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5167]: tcp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5167]: UDP relay enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5167]: udp port alternate enabled
Fri May 17 17:42:08 2019 daemon.info /usr/bin/ss-redir[5167]: running from root user
Fri May 17 17:42:09 2019 user.notice Shadowsocks: Rules DOWN
Fri May 17 17:42:09 2019 user.notice Shadowsocks: Rules UP
Fri May 17 17:42:09 2019 user.notice Shadowsocks: Reload omr-bypass rules
Fri May 17 17:42:09 2019 user.notice omr-bypass: Starting OMR-ByPass...

suyuan168, 2019-05-17:
I noticed the system log about Shadowsocks
Always prompt:
User.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Is it wrong to judge where? I see that the state of server-Shadowsocks is always restart-stop-start. Interval between them for a few seconds.

suyuan168, 2019-05-17:
In the meantime, new problems seem to have appeared, I unplug the lan cable. Plug in for about ten seconds. RPI3B+ will restart.

Ysurac, 2019-05-17:
The "Can't find Shadowsocks restart it" happen when Shadowsocks stop tried to connect to server.
So you have to check if the key is good.
Do you use v2ray or obfs ?
You can also check kernel logs.

suyuan168, 2019-05-18:
> The "Can't find Shadowsocks restart it" happen when Shadowsocks stop tried to connect to server.
> So you have to check if the key is good.
> Do you use v2ray or obfs ?
> You can also check kernel logs.

The "Can't find Shadowsocks restart it" happen when Shadowsocks stop tried to connect to server.
So you have to check if the key is good.
Yes, of course everything is correct. After the reboot, there will be a link for a while.
Do you use v2ray or obfs ?
NO, I have not enabled Enable ShadowSocks Obfuscating
 
You can also check kernel logs.
The kernel log looks good.

suyuan168, 2019-05-18:
To repeat such a question, /luci/admin/system/openmptcprouter/settings
Check VPS settings to check Redirects all ports from server to OpenMPTCProuter
Save Settings. Then the SS will restart without limit.

jayanta525, 2019-05-19:
After installing pi-hole using the provided script. everything works fine until the VPS is rebooted.
 `pihole -g` shows that DNS service is not running and dns resolving fails and that creates the no-output problem in OMR status page. 

Solution: run the vps installation script followed by the pi-hole installation script. the keys will be preserved and never reboot the vps.

Ysurac, 2019-05-19:
pi-hole is not needed for OpenMPTCProuter to work.
I'm not able to reproduce the issue. So if you want it to be resolved, I need some log/info.

jayanta525, 2019-05-19:
> pi-hole is not needed for OpenMPTCProuter to work.
> I'm not able to reproduce the issue. So if you want it to be resolved, I need some log/info.

after a reboot, DNS service(dnsmasq) fails to start as reported by `pihole -g`. This isn't a OMR bug, VPS fails to resolve any IP. Also, `pihole -d` shows it fails to ping the gateway IP address. I'll recreate this issue and post both pi-hole and omr logs here.

suyuan168, 2019-05-20:
I have tried it many times and still can't solve this problem. Install VPS0.933 and install pi-hole. Follow the tutorial exactly. He can work. However, the ORM system state always prompts Can't get public IP address from ShadowSocks Can't contact Server Admin Script. It will be normal again after a few seconds. The network has become unstable.
But switching to VPS0.992 is all right. Where is the problem?
Thank you for your answer.

closed 2019-06-05 by suyuan168

rifaterdemsahin, 2020-07-26:
I also get something similar .
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: DNS service limited to local subnets
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq-dhcp[28109]: DHCP, IP range 192.168.16.113 -- 192.168.16.192, lease time 12h
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using only locally-known addresses for domain test
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using only locally-known addresses for domain onion
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using only locally-known addresses for domain localhost
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using only locally-known addresses for domain local
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using only locally-known addresses for domain invalid
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using only locally-known addresses for domain bind
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using only locally-known addresses for domain use-application-dns.net
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using nameserver 127.0.0.1#5353
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: using only locally-known addresses for domain lan
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: read /etc/hosts - 4 addresses
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq[28109]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sun Jul 26 16:53:12 2020 daemon.info dnsmasq-dhcp[28109]: read /etc/ethers - 0 addresses
Sun Jul 26 16:53:12 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Jul 26 16:53:36 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Jul 26 16:54:05 2020 daemon.info dnsmasq-dhcp[28109]: DHCPREQUEST(eth0) 192.168.16.130 00:0e:c6:51:13:1b
Sun Jul 26 16:54:05 2020 daemon.info dnsmasq-dhcp[28109]: DHCPACK(eth0) 192.168.16.130 00:0e:c6:51:13:1b DESKTOP-4KQ2JNR
Sun Jul 26 16:54:08 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 198.11.132.250, 77.88.55.77, 1.1.1.1)
Sun Jul 26 16:54:08 2020 user.notice Shadowsocks: Rules DOWN
Sun Jul 26 16:58:07 2020 daemon.info dnsmasq-dhcp[28109]: DHCPREQUEST(eth0) 192.168.16.130 00:0e:c6:51:13:1b
Sun Jul 26 16:58:07 2020 daemon.info dnsmasq-dhcp[28109]: DHCPACK(eth0) 192.168.16.130 00:0e:c6:51:13:1b DESKTOP-4KQ2JNR