Ysurac/openmptcprouter #485: MTU issue with Glorytun UDP

uryupinsk, 2019-05-17:
## Expected Behavior

Have a MTU of 1500 on the tun0 interface when using Glorytun UDP, as Glorytun TCP.

## Actual Behavior

I have 2 WAN connections with MTU 1500 each. 

When I use Glorytun TCP, both the `tun0 `interface on OpenMPTCProuter and `gt-udp-tun0` interface on the VPS have an MTU 1500, which is correct. Speeds are somewhat good on a site to site OpenVPN UDP with a friend that also have OpenMPTCProuter running.

When I use Glorytun UDP, both the `tun0 `interface on OpenMPTCProuter and `gt-udp-tun0` interface on the VPS have an MTU 1378 (same thing on my friend's install). Speeds are horrible with the OpenVPN UDP tunnel.

When I try to force the MTU on Glorytun UDP with the LuCI interface on OpenMPTCProuter, it doesn't work, and I don't know how to change the MTU settings of `gt-udp-tun0` on the VPS.

## Specifications

  - OpenMPTCProuter version: 0.49.3
  - OpenMPTCProuter VPS version: 0.993
  - OpenMPTCProuter platform: x86_64

Ysurac, 2019-05-18:
With Glorytun UDP MTU is set by glorytun UDP himself.
You can set MSS seeting on OpenVPN configuration.

uryupinsk, 2019-05-18:
I already tried to change the `link-mtu`, `tun-mtu` or `mssfix `options in OpenVPN. I had the best results by setting `link-mtu` to 1350, but it's nothing close to the stability and performance or reverting to OpenVPN TCP or OpenVPN UDP through Glorytun TCP.

Are you saying that it's strictly a Glorytun issue and not an OpenMPTCProuter one?

uryupinsk, 2019-05-18:
I remember it was working well when I was on OpenMPTCProuter 0.48 and VPS 0.992.

Ysurac, 2019-05-18:
Can you give me result of ```glorytun-udp show dev tun0```, ```glorytun-udp path dev tun0``` and ```uci show openmptcprouter``` (remove keys, I only need interfaces data).

uryupinsk, 2019-05-18:
```
# glorytun-udp show dev tun0
client tun0:
  pid:       5049
  bind:      0.0.0.0 port 5000
  peer:      *REDACTED* port 65001
  mtu:       1378
  cipher:    aes256gcm
```

```
# glorytun-udp path dev tun0
path UP
  status:   OK
  bind:     192.168.20.2 port 5000
  public:   *REDACTED* port 30591
  peer:     *REDACTED* port 65001
  mtu:      1400 bytes
  rtt:      37.781 ms
  rttvar:   7.680 ms
  rate tx:  12500000 bytes/sec
  rate rx:  12500000 bytes/sec
  total tx: 247 packets
  total rx: 156 packets
path UP
  status:   OK
  bind:     192.168.10.2 port 5000
  public:   *REDACTED* port 5000
  peer:     *REDACTED* port 65001
  mtu:      1472 bytes
  rtt:      25.823 ms
  rttvar:   0.436 ms
  rate tx:  1260 bytes/sec
  rate rx:  19590 bytes/sec
  total tx: 132 packets
  total rx: 89 packets
```

```
# uci show openmptcprouter
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.version='0.49.3'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.shadowsocks_disable='0'
openmptcprouter.settings.external_check='1'
openmptcprouter.settings.master='change'
openmptcprouter.settings.vpn='glorytun_udp'
openmptcprouter.omr=router
openmptcprouter.omr.detected_ss_ipv4=*REDACTED*
openmptcprouter.omr.shadowsocks='up'
openmptcprouter.omr.detected_public_ipv4=*REDACTED*
openmptcprouter.wan1=interface
openmptcprouter.wan1.publicip=*REDACTED*
openmptcprouter.wan1.mtu='1500'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.asn='PROXAD'
openmptcprouter.wan1.mptcp_status='MPTCP enabled'
openmptcprouter.wan1.lc='1558204928'
openmptcprouter.wan1.latency_previous='12'
openmptcprouter.wan1.latency='11'
openmptcprouter.wan2=interface
openmptcprouter.wan2.state='up'
openmptcprouter.wan2.mtu='1500'
openmptcprouter.wan2.asn='FREEM'
openmptcprouter.wan2.mptcp_status='MPTCP enabled'
openmptcprouter.wan2.publicip=*REDACTED*
openmptcprouter.wan2.lc='1558205160'
openmptcprouter.wan2.latency_previous='30'
openmptcprouter.wan2.latency='25'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.state='down'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.vps='0.993'
openmptcprouter.latest_versions.omr='0.49.4'
openmptcprouter.latest_versions.lc='1558205118'
openmptcprouter.vps=server
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.ip=*REDACTED*
openmptcprouter.vps.port='65500'
openmptcprouter.vps.kernel='4.14.110-mptcp'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.omr_version='0.993'
openmptcprouter.vps.password=*REDACTED*
openmptcprouter.vps.admin_error='0'
openmptcprouter.vps.redirect_ports='1'
openmptcprouter.vps.available_vpn='glorytun-tcp' 'glorytun-udp' 'openvpn' 'mlvpn'
openmptcprouter.vps.get_config='0'
openmptcprouter.vps.token=*REDACTED
```

I didn't know of the `glorytun-udp path` command. It detects a MTU 1400 on the LTE WAN (Free Mobile) but the interface itself on OMR is detected as 1500, so does `omr-mtu-test wan2 vps`.

uryupinsk, 2019-05-18:
I concluded different ping tests:

Glorytun UDP tunnel started, with the test script you wrote:

```
# omr-mtu wan2 vps
1500
```

Glorytun UDP tunnel stopped, ping to a random server, with DF set through the LTE interface:

```
# ping -c 1 -M do -I wan2 -s 1472 free.fr
PING free.fr (212.27.48.10) from 192.168.20.2 wan2: 1472(1500) bytes of data.
1480 bytes from www.free.fr (212.27.48.10): icmp_req=1 ttl=58 time=24.6 ms

--- free.fr ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 24.698/24.698/24.698/0.000 ms
```

With Glorytun UDP tunnel stopped, ping to a random server, with DF set through the VDSL interface, which is clearly known to have a MTU 1500, is perfectly working too.

Ysurac, 2019-05-18:
I have a MTU of 1448 on Free LTE interface with glorytun-udp path.
I have no idea for now...

uryupinsk, 2019-05-18:
Very strange...
I'll reboot the LTE router first to see if it changes, then ask a friend who deploy LTE network at Free the real MTU of the link. But if it wasn't 1500, the ping with DF would not have worked...

uryupinsk, 2019-05-18:
Same thing after rebooting the LTE router: MTU 1500 on wan2 interface but MTU 1400 with `glorytun-udp path`.

Ysurac, 2019-05-18:
What is the result of ```uci show network | grep mtu``` on OpenMPTCProuter ?

uryupinsk, 2019-05-18:
`uci show network | grep mtu` doesn't return anything. By reading the result of the command without `grep`, I can confirm that there are no information about the MTU of interfaces.

Only `uci show openmptcprouter | grep mtu` return something about MTU. The result is 1500 for both the interface.

Ysurac, 2019-05-18:
ok, so no MTU set anywhere in configuration.

uryupinsk, 2019-05-18:
Yes, I have not set any MTU settings in configuration. All is autodetected by your scripts and glorytun.

angt, 2019-05-19:
@uryupinsk: not related to your MTU issue, but your rate conf is pretty low:
```
  rate tx:  1260 bytes/sec
  rate rx:  19590 bytes/sec
```

Ysurac, 2019-05-19:
This is my fault, I forget to add *1000 for kb/s to b/s. This will be fixed in 0.49.5 (CircleCI compile it now).

uryupinsk, 2019-05-19:
Good! I have no rate problem with TCP. I saturate the LTE bandwidth quite easily by doing nperf tests and downloading proof OVH files.

OpenVPN UDP through Glorytun TCP is a lot more efficient, but I obtain the best results for now by switching to OpenVPN TCP. I can saturate maybe 50-60 % of the actual bandwidth (maybe due to the inefficiency of OpenVPN TCP, double encryption, overhead, etc.). I know when it will be functioning, OpenVPN UDP through Glorytun UDP will be a lot more efficient.

uryupinsk, 2019-05-19:
After reading that some users where having MTU related problems with OpenVPN UDP on LTE connections (and on Free Mobile in France particularly), I decided to run some more ping tests.

With my Macbook connected through Ethernet on my LTE router (Huawei B525s-23a), I saw that a `ping -D -s 1472 free.fr` would return `Request timed out` and not `Message too long`. A ping > 1472 return `Message too long` as expected. By playing with different value, I finally got a response with a packet size of 1448, resulting in an MTU of 1476. This is consistent with the fact LTE should have more overhead than a plain IPoA or IPoE, but the implementation of Free Mobile I guess break the PTMUD.

Back to OMR, I forced the MTU to 1476 in Network > Interface. Then look at this:

```
# glorytun-udp path dev tun0
path UP
  status:   OK
  bind:     192.168.10.2 port 5000
  public:   *REDACTED* port 5000
  peer:     *REDACTED* port 65001
  mtu:      1472 bytes
  rtt:      25.988 ms
  rttvar:   0.434 ms
  rate tx:  125000000 bytes/sec
  rate rx:  125000000 bytes/sec
  total tx: 195 packets
  total rx: 126 packets
path UP
  status:   OK
  bind:     192.168.20.2 port 5000
  public:   *REDACTED* port 35500
  peer:     *REDACTED* port 65001
  mtu:      1448 bytes
  rtt:      42.453 ms
  rttvar:   5.683 ms
  rate tx:  125000000 bytes/sec
  rate rx:  125000000 bytes/sec
  total tx: 173 packets
  total rx: 121 packets
```

```
# glorytun-udp show dev tun0
client tun0:
  pid:       5082
  bind:      0.0.0.0 port 5000
  peer:      *REDACTED* port 65001
  mtu:       1426
  cipher:    aes256gcm
```

I remember that when I first installed OMR on 0.48 and VPS 0.992 I had MTU 1426 for the Glorytun UDP tunnel, when it was working great.

Also, when I force the MTU of wan2 it's changing it correctly, but in the network overview it doesn't take into account the change and show 1500.

Ysurac, 2019-05-20:
On status page it's only a bad display, this will be fixed in next release.
I will also test MTU on each interface more often when not set manually.

uryupinsk, 2019-05-21:
Glad to hear!

I still have very bad performance on OpenVPN UDP through Glorytun UDP but I think it's more an OpenVPN configuration issue now.

closed 2019-07-04 by Ysurac