Ysurac/openmptcprouter #486: Unfortunately, the Raspberry Pi 3B+ network can't exceed 100M anyway. This is a major bug

suyuan168, 2019-05-19:
Whether I use a few wan or a wan
My wan1 speed is 200M
My wan2 speed is 500M
My VPS speed is 200M
Now RPI3B+ runs ORM and everything is fine. But the speed can never exceed 100M. This is not a bug.
Thank you very much for your answer.
  - OpenMPTCProuter version: (Last is not a version)v0.49.4
  - OpenMPTCProuter VPS version: (Last is not a version)0.933
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64)RPI3B+

Ysurac, 2019-05-19:
I use a RPI3B+, with default config I can get 150-200Mb/s if I only use download link (there is some limit of CPU power using chacha20 encryption). USB is also not used (USB speed and ethernet speed are shared).

suyuan168, 2019-05-19:
I'm using no encryption, and it's a little better for cpu consumption. Maybe I'm using usb, so it slows down. But how do I get to the Internet? Rpi3B+ I used usb3.0 to expand a network cable. There is no better recommendation, it seems that raspberry pie is not suitable to be a router.

Ysurac, 2019-05-19:
I use RPI3B+ as my main router, so it's suitable if you have less than 200Mb/s and you don't need USB.
USB on RPI is really bad (wifi is also bad).
If you find something using an ARM or x86 with enough CPU power and widely available (without strange support like banana pi), and less than 150/100€/$ I will be happy to buy it and support it.
It's really difficult to find good hardware at less than 200/300€.

jayanta525, 2019-05-19:
@Ysurac What are your opinions on the other SBC's like the [ESPRESSObin](https://openwrt.org/toh/hwdata/globalscale/globalscale_espressobin_v5_0_1)? They are quite powerful and have true-gigabit ports. Also less than 100$. Availability is bit tight, but amazon ships worldwide.

Ysurac, 2019-05-19:
Seems to be the same platform as WRT32X/WRT3200ACM. So no real problem to support it, not sure if it's really better than a RPI but more ethernet ports. Why not.

suyuan168, 2019-05-19:
Up2 x86 has two gigabit network cards and supports usb3.0. I will buy one tomorrow to try an x86 installation.

suyuan168, 2019-05-19:
Espressobin, this is amazing.

Ysurac, 2019-05-19:
UP2 is a bit expensive.

jayanta525, 2019-05-19:
Check out [EdgeRouter X](https://openwrt.org/toh/hwdata/ubiquiti/ubiquiti_edgerouter_x).

Ysurac, 2019-05-19:
And not enough RAM and CPU.

jayanta525, 2019-05-19:
@suyuan168 Up2 x86 is way too expensive for the specs provided. x86 is already supported by OMR. There are cheaper custom x86 boards with Intel NIC, with and without AES-NI.

suyuan168, 2019-05-20:
I am very happy to tell you that I have purchased ESPRESSObin. It will arrive in the next two days. AMADA385 hopes to be supported by openwrt. I hope to use the ORM system. I feel that this motherboard is very powerful.
In addition, I have an x86 up2 and I will also test it. And share the results.

suyuan168, 2019-05-20:
I just tried my soft routing x86, he has 6 Gigabit Ethernet ports and has usb3.0. I used 6 4G network cards plus 200M fiber. With the ORM test speed is only 110M, the speed is very low. But I removed the 4G And only used 200M fiber. At this time the speed can be full. The maximum speed is 200M and the speed is very fast. So I think MPTCP is very unfriendly for USB 4G. Not only does the speed not improve, but it also drops .
The conclusion is that they feel that they will average the network speed.
Thank you everyone.

suyuan168, 2019-05-20:
https://github.com/multipath-tcp/mptcp/issues/282

PhatHub, 2019-05-22:
Maybe check out Atomic Pi? (They unfortunately sold out quickly)
[Atomic Pi!](https://dlidirect.com/products/atomic-pi)

- Quad-core Intel Atom  (x5-Z8350)
- USB 3.0
- Gigabit Ethernet
- 2.4GHz and 5GHz b/g/n/ac Wifi
- BT 4.0
- 2GB of RAM
- ...and only $35 dollars! ($18 shipping in the USA not included)

karjaj, 2019-05-25:
How about Orange PI PC2?
I don't know, how is CPU, but there's real(tek) gigabit ethernet and all USB-ports aren't connected to same hub. 
It's also cheap.

jayanta525, 2019-05-26:
The availability of OrangePi boards is limited to AliExpress and Banggood only. If you already own one, you can compile your own OMR build from the sources.

PhatHub, 2019-06-24:
Update: 
I bought an AtomicPi board and got a USB hub and some USB network devices. Luci runs a lot faster compared to my old single-core 32-bit Celeron, but the bandwidth has not been too much better, but I will have to do more tests and change configurations. I will also test three 4g connections to see if there are bandwidth increases over two. I'll have to try BLEST as well.

My main concern is to at least get 720p on a movie stream, so I'm happy when it runs without buffering with two phones when they're both throttled (from 80-100Mbps down to 600Kbps) for exceeding the monthly 4G allotment.

The board is currently available with Prime shipping on amazon for $43.

suyuan168, 2019-06-25:
I used up2 and it has two Gigabit Ethernet ports. I achieved a speed of 200MBps

Bestle, 2019-07-01:
Any results yet for the Raspberry Pi 4?

closed 2019-07-04 by Ysurac