Ysurac/openmptcprouter #488: Three Ethernet usb 4G agreggate

Florent4223, 2019-05-21:
## Expected Behavior

Agreggate three usb ethernet 4g dongle

## Actual Behavior

I mounted my tests with 3 USB keys - 4G ethernet.
And I have a new weird case. I have the very strong impression that OMR freeze.
What is on it is that he reboot

## Steps to Reproduce the Problem

  1. connecte et configure on OMR
  2. glorytun tcp ok
  3. transfert file ftp between client and VPS 

## Specifications

  - OpenMPTCProuter version: 0.49.5
  - OpenMPTCProuter VPS version: 0.993
  - OpenMPTCProuter platform: (RPI2/RPI3/x86/x86_64) RPI3b+

`Tue May 21 14:13:11 2019 kern.info kernel: [  232.869661] usb 1-1.1.3: USB disconnect, device number 12
Tue May 21 14:13:11 2019 kern.info kernel: [  232.876140] cdc_ether 1-1.1.3:1.0 eth1: unregister 'cdc_ether' usb-3f980000.usb-1.1.3, CDC Ethernet Device
Tue May 21 14:13:11 2019 daemon.notice netifd: Network device 'eth1' link is down
Tue May 21 14:13:11 2019 daemon.notice netifd: Interface 'LTE0005' has link connectivity loss
Tue May 21 14:13:11 2019 daemon.notice netifd: LTE0005 (14589): udhcpc: SIOCGIFINDEX: No such device
Tue May 21 14:13:11 2019 daemon.notice netifd: LTE0005 (14589): udhcpc: received SIGTERM
Tue May 21 14:13:11 2019 daemon.notice netifd: Interface 'LTE0005' is now down
Tue May 21 14:13:11 2019 daemon.notice netifd: Interface 'LTE0005' is disabled
Tue May 21 14:13:11 2019 kern.info kernel: [  233.064760] usb 1-1.2: USB disconnect, device number 15
Tue May 21 14:13:11 2019 kern.info kernel: [  233.071059] cdc_ether 1-1.2:1.0 eth2: unregister 'cdc_ether' usb-3f980000.usb-1.2, CDC Ethernet Device
Tue May 21 14:13:11 2019 daemon.notice netifd: Network device 'eth2' link is down
Tue May 21 14:13:11 2019 daemon.notice netifd: Interface 'LTE0002' has link connectivity loss
Tue May 21 14:13:11 2019 daemon.notice netifd: LTE0002 (15068): udhcpc: SIOCGIFINDEX: No such device
Tue May 21 14:13:11 2019 daemon.notice netifd: LTE0002 (15068): udhcpc: received SIGTERM
Tue May 21 14:13:11 2019 daemon.notice netifd: Interface 'LTE0002' is now down
Tue May 21 14:13:11 2019 daemon.notice netifd: Interface 'LTE0002' is disabled
Tue May 21 14:13:12 2019 kern.info kernel: [  233.643131] usb 1-1.3: USB disconnect, device number 16
Tue May 21 14:13:12 2019 kern.info kernel: [  233.649534] cdc_ether 1-1.3:1.0 eth3: unregister 'cdc_ether' usb-3f980000.usb-1.3, CDC Ethernet Device
Tue May 21 14:13:12 2019 daemon.notice netifd: Network device 'eth3' link is down
Tue May 21 14:13:12 2019 daemon.notice netifd: Interface 'LTE0001' has link connectivity loss
Tue May 21 14:13:12 2019 daemon.notice netifd: LTE0001 (15059): udhcpc: SIOCGIFINDEX: No such device
Tue May 21 14:13:12 2019 daemon.notice netifd: LTE0001 (15059): udhcpc: received SIGTERM
Tue May 21 14:13:12 2019 daemon.notice netifd: Interface 'LTE0001' is now down
Tue May 21 14:13:12 2019 daemon.notice netifd: Interface 'LTE0001' is disabled
Tue May 21 14:13:12 2019 daemon.err omr-tracker[6548]: sendto: No such device
Tue May 21 14:13:12 2019 daemon.err omr-tracker[6548]: terminate called after throwing an instance of 'std::runtime_error'
Tue May 21 14:13:12 2019 daemon.err omr-tracker[6548]:   what():  Packet::SocketSendRecv() : Listening device eth1: No such device exists (SIOCGIFHWADDR: No such device)
Tue May 21 14:13:12 2019 daemon.err omr-tracker[6548]: Aborted
Tue May 21 14:13:12 2019 user.notice post-tracking-post-tracking: Get status and settings for LTE0005... Done
Tue May 21 14:13:15 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue May 21 14:13:17 2019 daemon.info dnsmasq[9294]: read /etc/hosts - 4 addresses
Tue May 21 14:13:17 2019 daemon.info dnsmasq[9294]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue May 21 14:13:17 2019 daemon.info dnsmasq-dhcp[9294]: read /etc/ethers - 0 addresses
Tue May 21 14:13:17 2019 user.notice omr-bypass: OMR-ByPass is running
Tue May 21 14:13:18 2019 kern.info kernel: [  240.331876] usb 1-1.1.3: new high-speed USB device number 17 using dwc_otg
Tue May 21 14:13:19 2019 kern.info kernel: [  240.463225] usb 1-1.1.3: New USB device found, idVendor=12d1, idProduct=1f01
Tue May 21 14:13:19 2019 kern.info kernel: [  240.471332] usb 1-1.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Tue May 21 14:13:19 2019 kern.info kernel: [  240.480842] usb 1-1.1.3: Product: HUAWEI_MOBILE
Tue May 21 14:13:19 2019 kern.info kernel: [  240.486389] usb 1-1.1.3: Manufacturer: HUAWEI_MOBILE
Tue May 21 14:13:19 2019 kern.info kernel: [  240.492375] usb 1-1.1.3: SerialNumber: 0123456789ABCDEF
Tue May 21 14:13:19 2019 kern.debug kernel: [  240.498881] usb 1-1.1.3: no of_node; not parsing pinctrl DT
Tue May 21 14:13:19 2019 kern.debug kernel: [  240.506275] uas 1-1.1.3:1.0: no of_node; not parsing pinctrl DT
Tue May 21 14:13:19 2019 kern.debug kernel: [  240.513351] usb-storage 1-1.1.3:1.0: no of_node; not parsing pinctrl DT
Tue May 21 14:13:19 2019 kern.info kernel: [  240.521060] usb-storage 1-1.1.3:1.0: USB Mass Storage device detected
Tue May 21 14:13:19 2019 kern.info kernel: [  240.528964] scsi host0: usb-storage 1-1.1.3:1.0
Tue May 21 14:13:19 2019 kern.info kernel: [  240.805683] usb 1-1.1.3: USB disconnect, device number 17
Tue May 21 14:13:19 2019 kern.info kernel: [  241.211892] usb 1-1.1.3: new high-speed USB device number 18 using dwc_otg
Tue May 21 14:13:19 2019 kern.info kernel: [  241.342942] usb 1-1.1.3: New USB device found, idVendor=12d1, idProduct=14dc
[... 253 lines cut ...]
Tue May 21 14:13:48 2019 kern.info kernel: [  270.191118] scsi host2: usb-storage 1-1.3:1.2
Tue May 21 14:13:49 2019 kern.notice kernel: [  270.701371] scsi 1:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
Tue May 21 14:13:49 2019 kern.debug kernel: [  270.711467] sd 1:0:0:0: no of_node; not parsing pinctrl DT
Tue May 21 14:13:49 2019 kern.notice kernel: [  270.728925] sd 1:0:0:0: [sdb] Attached SCSI removable disk
Tue May 21 14:13:49 2019 user.notice omr-tracker: Can't create route to 1.0.0.1 via 192.168.10.1 dev eth2 src 192.168.10.100. waiting...
Tue May 21 14:13:49 2019 kern.notice kernel: [  271.261266] scsi 2:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
Tue May 21 14:13:49 2019 kern.debug kernel: [  271.271334] sd 2:0:0:0: no of_node; not parsing pinctrl DT
Tue May 21 14:13:49 2019 kern.notice kernel: [  271.280066] sd 2:0:0:0: [sdc] Attached SCSI removable disk
Tue May 21 14:13:50 2019 user.notice omr-tracker: Can't create route to 80.67.169.12 via 192.168.10.1 dev eth2 src 192.168.10.100. waiting...
Tue May 21 14:13:51 2019 user.notice omr-bypass: Starting OMR-ByPass...
Tue May 21 14:13:51 2019 daemon.err omr-tracker-ss[6551]: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Tue May 21 14:13:51 2019 daemon.notice netifd: Network device 'eth1' link is up
Tue May 21 14:13:51 2019 daemon.notice netifd: Interface 'LTE0005' has link connectivity
Tue May 21 14:13:51 2019 daemon.notice netifd: Interface 'LTE0005' is setting up now
Tue May 21 14:13:51 2019 daemon.notice netifd: LTE0005 (2249): udhcpc: started, v1.30.1
Tue May 21 14:13:51 2019 daemon.notice netifd: LTE0005 (2249): udhcpc: sending discover
Tue May 21 14:13:51 2019 daemon.notice netifd: LTE0005 (2249): udhcpc: sending select for 192.168.8.100
Tue May 21 14:13:51 2019 daemon.notice netifd: LTE0005 (2249): udhcpc: lease of 192.168.8.100 obtained, lease time 86400
Tue May 21 14:13:51 2019 daemon.notice netifd: Interface 'LTE0005' is now up
Tue May 21 14:13:51 2019 user.notice omr-tracker: Can't create route to 80.67.169.40 via 192.168.10.1 dev eth2 src 192.168.10.100. waiting...
Tue May 21 14:13:51 2019 user.notice firewall: Reloading firewall due to ifup of LTE0005 (eth1)
Tue May 21 14:13:51 2019 user.notice mptcp: Reloading mptcp config due to ifup of LTE0005 (eth1)
Tue May 21 14:13:52 2019 user.notice omr-tracker: Can't create route to 4.2.2.1 via 192.168.10.1 dev eth2 src 192.168.10.100. waiting...
Tue May 21 14:13:53 2019 user.notice omr-tracker: Can't create route to 8.8.8.8 via 192.168.10.1 dev eth2 src 192.168.10.100. waiting...
`

suyuan168, 2019-05-21:
Please check your USB NIC for power. As far as I know you need a usb hub with a separate 5V power supply. Experience tells me to rely on rpi3 to supply power. It doesn't move so much USB. Will crash and restart.

Florent4223, 2019-05-21:
I think about it too
But as long as I do not use the VPN the rpi3 and USB stick does not move

suyuan168, 2019-05-21:
So, it's still a problem with the power supply. RPI does not have such a powerful power supply. If it is running at full load, it needs 5V 2.4A. Your USB 4G requires a 1A operating current. So still add a USB HUB with power.

closed 2019-07-04 by Ysurac