Ysurac/openmptcprouter #489: mptcp_detect_mapping Mappings do not match!

xabix, 2019-05-21:
Hello,

I have the same issue with the latest release: openmptcprouter v0.49.5 r0+9945-bc85640cdc

What can I info should I be looking to better troubleshoot the issue?
[image]

2Gb of RAM on the VPS:
              total        used        free      shared  buff/cache   available
Mem:        2099392      336080     1202556       21552      560756     1592056
Swap:             0           0           0

and CPU load very low (2 vCPU)

  1  [**@                                                            3.4%]   Tasks: 31, 4 thr; 1 running
  2  [*                                                              0.7%]   Load average: 0.01 0.03 0.00 
  Mem[||||||||||||###***************                           350M/2.00G]   Uptime: 13 days, 07:12:23
  Swp[                                                              0K/0K]

  PID USER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command                                                                                    
31500 mlvpn      20   0 51188 28452  1964 S  1.4  1.4 53:34.94 mlvpn: mlvpn !wan8 !wan7 !wan6 @wan5 @wan4 @wan3 @wan2 !wan1                               
31567 root       20   0  307M 34024 10080 S  0.7  1.6 12:57.77 python3 /usr/local/bin/omr-admin.py
32151 root       20   0 25164  5396  2812 S  0.7  0.3 18:59.64 /usr/bin/ss-server -c /etc/shadowsocks-libev/config2.json --no-delay
29128 root       20   0 24748  3924  3008 R  0.0  0.2  0:00.03 htop
    1 root       20   0  199M  6984  5292 S  0.0  0.3  0:30.34 /sbin/init
Merci
XabiX

_Originally posted by @xabix in https://github.com/Ysurac/openmptcprouter/issues/335#issuecomment-494375011_

xabix, 2019-05-24:
Hello, I still have a lot of these messages. What can I do to find the issue? unless this is not impacting and I should just disregard it.

Ysurac, 2019-05-25:
What are the connections used and RTT of each ?

xabix, 2019-05-27:
Hello Ysurac,

not sure how best to monitor/calculate the RTT. Would the latency below answer the question?

I have noticed that this issue start after some time (what I mean is that I don't have the issue when I just reboot and look at the console).

I have 2 * ADSL and 2 * 4G as described below:
[image]

Merci
XabiX

Ysurac, 2019-05-28:
You can try to update and reboot the VPS. If no change you can try to change the TCP congestion control. If still no change, you will have to wait for release based on 4.19 kernel with BLEST scheduler that may solve the issue.

xabix, 2019-06-05:
Hello, I still have the issue after reboot the VPS and installing 0.50.1. While I wait for the 4.19 kernel etc... can you be more specific when you say change the TCP congestion control? Is it linked to SQM? or the IPv4 TCP FIN timeout; SYN retries?

Ysurac, 2019-06-05:
try olia and cubic at least

closed 2019-07-04 by Ysurac

xabix, 2019-09-03:
I have configured Blest and Reno and seem good. I found this nice article but quite technical for me :)
http://www.satnac.org.za/proceedings/2012/papers/2.Core_Network_Technologies/15.pdf