Ysurac/openmptcprouter #490: VPS-Kernel lacks sit-module (for IPv6 oder IPv4-Tunnels) on Ubuntu

gstammw, 2019-05-23:
## Expected Behavior

  The kernel should include the "sit"-tunnel-module so that one can set up IPv6-over-IPv4-tunnels to bring IPV6-connectivity to machines that have native IPv4-only.

## Actual Behavior

The module is not present.
Module sit not found in directory /lib/modules/4.15.0

## Steps to Reproduce the Problem

  1. modprobe sit
  2. Check results:
"modprobe sit
modprobe: ERROR: ../libkmod/libkmod.c:514 lookup_builtin_file() could not open builtin file '/lib/modules/4.15.0/modules.builtin.bin'
modprobe: FATAL: Module sit not found in directory /lib/modules/4.15.0"

## Specifications

  - OpenMPTCProuter version: n/a
  - OpenMPTCProuter VPS version: 0.993
  - OpenMPTCProuter platform: x64, Ubuntu 18.04 LTS

Ysurac, 2019-05-23:
Wrong kernel.
Did you reboot after installing VPS script ?

gstammw, 2019-05-23:
You are right, the mptcp-kernel was not active. I was tricked by virtuozzo which didn't boot any of my installed kernels. With kvm things are working fine now :-)

closed 2019-05-23 by gstammw