Ysurac/openmptcprouter #491: Raspberry pi3+ image does not boot on USB only, require eMMC

ragefan, 2019-05-26:
## Expected Behavior

Take a fresh new Raspberry pi 3 B+ (which can boot from USB), flash openMptcpimage to an new USBkey using appropriate tool. Plug the key in the raspberry pi. The boot starts but the system is waiting for /root on mmc. 

## Actual Behavior
The system cannot boot with a usb key only. It requires a mmc on the sd slot of the raspberry pi.

## Steps to Reproduce the Problem

  1. flash image to usb
  2. remove mmc from raspberry pi3+
  3. boot raspberry pi3+ with usb only

## Specifications

  - OpenMPTCProuter version: 0.49.7
  - OpenMPTCProuter VPS version: 0.49.7
  - OpenMPTCProuter platform: RPI3 B+

Ysurac, 2019-05-26:
It's a really bad idea to boot on USB for a RPI.
USB is shared with ethernet, so for a router this can't work well.
This issue will not be solved.

ragefan, 2019-05-26:
Indeed, this may impact performance but sdcard lifetime is a real issue with raspberry

Ysurac, 2019-05-26:
There is not so much data writed on disk, so not really a problem.

closed 2019-05-26 by ragefan