Ysurac/openmptcprouter #492: Restoring configuration on another router keeps previous router MAC address

ragefan, 2019-05-26:
## Expected Behavior

When restoring the configuration on another omr, the MAC address of LAN interface shall be migrated with the new router NIC.

## Actual Behavior

The MAC address of the original omr is restored as a custom mac address bypassing the new omr one.
This cause mac address conflict when both old & new machines are on the same network.
The file "network" is containing :

config interface 'lan'
	option ifname 'eth0'
	option proto 'static'
	option ipaddr '192.168.100.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option delegate '0'
	option multipath 'off'
	option ip4table 'lan'
	option macaddr 'c0:3f:d5:68:b6:47' <= previous MAC address

## Steps to Reproduce the Problem

  1. take an omr, backup configuration
  2. restore configuration on another machine
  3. the newly restore machine LAN interface contains the mac address of the previous machine

## Specifications

  - OpenMPTCProuter version: 0.49.7
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-05-26:
It's expected. It's a full backup.

closed 2019-05-26 by Ysurac

ragefan, 2019-05-26:
Can i update documentation somewhere to precise it then ?