Ysurac/openmptcprouter #493: Enhancement: Add temp in home page

Dragonox77, 2019-05-27:
Hello,
Thank you for your solution.

I think add cpu temp in the home page is a good idea. It can be easily read from this file: /sys/class/thermal/thermal_zone0/temp for RPI.

Thank you.

jayanta525, 2019-05-27:
You can use luci-app-statistics to display temperature graphs.

[image]

Dragonox77, 2019-05-27:
Oh didn't see thank you for your answer.

jayanta525, 2019-05-27:
Don't forget to install collectd-mod-thermal and select the thermal zone in the configuration.

Dragonox77, 2019-05-27:
How to install luci-app-statistics. I got this error: Unknown package 'luci-app-statistics'.

jayanta525, 2019-05-27:
`opkg update` followed by `opkg install luci-app-statistics collectd-mod-thermal`

Dragonox77, 2019-05-27:
Collected errors:
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

And then i can't run opkg install luci-app-statistics collectd-mod-thermal

manueldev, 2019-05-27:
```
uci set network.lan.dns=8.8.8.8
uci commit && service network restart
```

Dragonox77, 2019-05-27:
> ```
> uci set network.lan.dns=8.8.8.8
> uci commit && service network restart
> ```

Not working either.

jayanta525, 2019-05-27:
wget is trying to download over IPv6 and I guess you don't have IPv6 connectivity. 
In that case, head over to `System > Software > Configuration`

Replace Custom Feeds with a mirror; for France:
```
src/gz openwrt_luci http://gra.mirror.cyberbits.eu/openwrt/snapshots/packages/aarch64_cortex-a53/luci
src/gz openwrt_packages http://gra.mirror.cyberbits.eu/openwrt/snapshots/packages/aarch64_cortex-a53/packages
src/gz openwrt_base http://gra.mirror.cyberbits.eu/openwrt/snapshots/packages/aarch64_cortex-a53/base
src/gz openwrt_routing http://gra.mirror.cyberbits.eu/openwrt/snapshots/packages/aarch64_cortex-a53/routing
src/gz openwrt_telephony http://gra.mirror.cyberbits.eu/openwrt/snapshots/packages/aarch64_cortex-a53/telephony

```
Save configuration and `opkg update`

You can find the list of mirrors here: https://openwrt.org/downloads

Dragonox77, 2019-05-28:
Nice, thank you, all is working like it should.

closed 2019-05-28 by Dragonox77