Ysurac/openmptcprouter #496: Aggregate two households

ChristianKniep, 2019-05-28:
## Expected Outcome
I would like to connect two houses in close proximity using OpenMPTCProuter.

```
LAN1  
   |
OpenMTPCProuter1 ------ ADSL1 
   |                      |
Nanostation1              |
   |(--NS3-R3-ADSL3-)    VPS
Nanostation2              |
   |                      |
OpenMTPCProuter2 ------ ADSL2
   |
LAN2
```

I assume as a first iteration I could allow Internet access from both sites via the interconnect so that both routers do not know about their peers on the other side.
Which would mean that they do not know about the other sites network, but just use the interconnect as a second internet provider.

## Wishful thinking
I am wondering whether this is a good approach?
In the future I might want to add another neighbour to the nanostation network (NS3=Nanostation3 and so on)...

Ysurac, 2019-05-28:
Should work but not sure about RTT that nanostation can give.

ChristianKniep, 2019-05-28:
@Ysurac Should work for my assumption or the wishful approach I assume...
Having all routers use the remote sites as an additional route to reach the VPS.

Ysurac, 2019-05-28:
Each nanostation will be like a modem to access internet for OpenMPTCProuter, so good approach but you don't really have other choices to do that.

ChristianKniep, 2019-05-28:
parfait... merci

closed 2019-05-28 by ChristianKniep