Ysurac/openmptcprouter #497: Netflix OMR-Bypass no longer works

rubaz100, 2019-05-31:
## Expected Behavior
Netflix Content filtered to wan2

## Actual Behavior
Netflix shows that I use a Proxy
[image]

## Steps to Reproduce the Problem

[image]

## Specifications
  - DNS changed to the OpenMPTCPRouter  (192.168.100.1)
  - OpenMPTCProuter version: 0.49.7
  - OpenMPTCProuter VPS version: 0.994
  - OpenMPTCProuter platform: WRT3200ACM (ARM)
[image]

Ysurac, 2019-05-31:
When a domain is in both DSCP and omr-bypass then only one is working. This will be solved in next release.

rubaz100, 2019-05-31:
Oh okay, so we need to wait for the next release? I don't mind... When do you want to publish it?
Do you know how to fix it? :=)

Thank you for everything you did for us.

<3

Ysurac, 2019-05-31:
It's already fixed in openmptcprouter-feeds tree. Only need to test and tag for release ;)

rubaz100, 2019-06-01:
I can't build it :(

[image]

Ysurac, 2019-06-01:
Strange.
But I'm building it, this will be OpenMPTCProuter 0.50.
Images will be available here https://www.openmptcprouter.com/release/targets/ if CircleCI can compile it ( https://circleci.com/gh/Ysurac/openmptcprouter-feeds )

rubaz100, 2019-06-01:
Okay, thank you. I will definitely donate some € for you. Incredible!!!

closed 2019-06-02 by rubaz100