Ysurac/openmptcprouter #498: VPN Tunel Down...

mycanaletto, 2019-06-02:
Bonjour,

Tout allait très bien, sauf que j'ai fait une mise à jour des deux cotés...
[image]

Ysurac, 2019-06-02:
Reboot, regarde les logs.

mycanaletto, 2019-06-02:
J'ai restauré le routeur .48 et c'est OK

Je vais refaire la MAJ

mycanaletto, 2019-06-02:
Et après la MAJ et reboot des deux cotés  j'ai ça...
[image]

Ysurac, 2019-06-02:
Il faut voir du côté du journal système sur la partie routeur.

mycanaletto, 2019-06-02:
Sun Jun  2 17:32:25 2019 daemon.info mlvpn[19422]: new password set
Sun Jun  2 17:32:25 2019 daemon.info mlvpn[19422]: 3 tunnel added
Sun Jun  2 17:32:25 2019 daemon.warn mlvpn[19422]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:25 2019 daemon.info mlvpn[19422]: 4 tunnel added
Sun Jun  2 17:32:25 2019 daemon.warn mlvpn[19422]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:25 2019 daemon.info mlvpn[19422]: created interface `mlvpn0'
Sun Jun  2 17:32:25 2019 daemon.info mlvpn[19422]: 3 bind to 192.168.210.214
Sun Jun  2 17:32:25 2019 kern.info kernel: [ 2222.912431] mlvpn[19422]: segfault at 19 ip 00007fa4f15ac805 sp 00007ffc311b50b0 error 6 in libc.so[7fa4f15aa000+49000]
Sun Jun  2 17:32:25 2019 daemon.err /usr/bin/ss-local[8476]: remote_recv_cb_recv: Connection reset by peer
Sun Jun  2 17:32:27 2019 daemon.err /usr/bin/ss-local[8476]: remote_recv_cb_recv: Connection reset by peer
Sun Jun  2 17:32:28 2019 daemon.err uhttpd[4935]: ping: bad address 'mlvpn0'
Sun Jun  2 17:32:30 2019 daemon.info mlvpn[19565]: new password set
Sun Jun  2 17:32:30 2019 daemon.info mlvpn[19565]: 3 tunnel added
Sun Jun  2 17:32:30 2019 daemon.warn mlvpn[19565]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:30 2019 daemon.info mlvpn[19565]: 4 tunnel added
Sun Jun  2 17:32:30 2019 daemon.warn mlvpn[19565]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:30 2019 daemon.info mlvpn[19565]: created interface `mlvpn0'
Sun Jun  2 17:32:30 2019 daemon.info mlvpn[19565]: 3 bind to 192.168.210.214
Sun Jun  2 17:32:30 2019 kern.info kernel: [ 2227.933405] mlvpn[19565]: segfault at 19 ip 00007f849624e805 sp 00007fffa1339890 error 6 in libc.so[7f849624c000+49000]
Sun Jun  2 17:32:31 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Jun  2 17:32:35 2019 daemon.info mlvpn[19646]: new password set
Sun Jun  2 17:32:35 2019 daemon.info mlvpn[19646]: 3 tunnel added
Sun Jun  2 17:32:35 2019 daemon.warn mlvpn[19646]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:35 2019 daemon.info mlvpn[19646]: 4 tunnel added
Sun Jun  2 17:32:35 2019 daemon.warn mlvpn[19646]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:35 2019 daemon.info mlvpn[19646]: created interface `mlvpn0'
Sun Jun  2 17:32:35 2019 daemon.info mlvpn[19646]: 3 bind to 192.168.210.214
Sun Jun  2 17:32:35 2019 kern.info kernel: [ 2232.959608] mlvpn[19646]: segfault at 19 ip 00007f331938e805 sp 00007ffc7afb2200 error 6 in libc.so[7f331938c000+49000]
Sun Jun  2 17:32:35 2019 daemon.err /usr/bin/ss-local[8476]: remote_recv_cb_recv: Connection reset by peer
Sun Jun  2 17:32:38 2019 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.210.1 dev wan2
Sun Jun  2 17:32:38 2019 user.notice post-tracking-post-tracking: Can't find MLVPN, restart it...
Sun Jun  2 17:32:38 2019 daemon.info mlvpn[19807]: new password set
Sun Jun  2 17:32:38 2019 daemon.info mlvpn[19807]: 3 tunnel added
Sun Jun  2 17:32:38 2019 daemon.warn mlvpn[19807]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:38 2019 daemon.info mlvpn[19807]: 4 tunnel added
Sun Jun  2 17:32:38 2019 daemon.warn mlvpn[19807]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:38 2019 daemon.info mlvpn[19807]: created interface `mlvpn0'
Sun Jun  2 17:32:38 2019 daemon.info mlvpn[19807]: 3 bind to 192.168.210.214
Sun Jun  2 17:32:38 2019 kern.info kernel: [ 2236.477320] mlvpn[19807]: segfault at 19 ip 00007f9417f21805 sp 00007ffe36ce5b00 error 6 in libc.so[7f9417f1f000+49000]
Sun Jun  2 17:32:43 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Jun  2 17:32:43 2019 daemon.info mlvpn[19872]: new password set
Sun Jun  2 17:32:43 2019 daemon.info mlvpn[19872]: 3 tunnel added
Sun Jun  2 17:32:43 2019 daemon.warn mlvpn[19872]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:43 2019 daemon.info mlvpn[19872]: 4 tunnel added
Sun Jun  2 17:32:43 2019 daemon.warn mlvpn[19872]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:43 2019 daemon.info mlvpn[19872]: created interface `mlvpn0'
Sun Jun  2 17:32:43 2019 daemon.info mlvpn[19872]: 3 bind to 192.168.210.214
Sun Jun  2 17:32:43 2019 kern.info kernel: [ 2241.502862] mlvpn[19872]: segfault at 19 ip 00007fc0cf70a805 sp 00007ffd78027940 error 6 in libc.so[7fc0cf708000+49000]
Sun Jun  2 17:32:44 2019 user.notice post-tracking-post-tracking: Can't find MLVPN, restart it...
Sun Jun  2 17:32:44 2019 daemon.info mlvpn[20003]: new password set
Sun Jun  2 17:32:44 2019 daemon.info mlvpn[20003]: 3 tunnel added
Sun Jun  2 17:32:44 2019 daemon.warn mlvpn[20003]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:44 2019 daemon.info mlvpn[20003]: 4 tunnel added
Sun Jun  2 17:32:44 2019 daemon.warn mlvpn[20003]: Enabling automatic bandwidth adjustment
Sun Jun  2 17:32:44 2019 daemon.info mlvpn[20003]: created interface `mlvpn0'
Sun Jun  2 17:32:44 2019 daemon.info mlvpn[20003]: 3 bind to 192.168.210.214
Sun Jun  2 17:32:44 2019 kern.info kernel: [ 2242.120776] mlvpn[20003]: segfault at 19 ip 00007fa06fba5805 sp 00007ffd14342290 error 6 in libc.so[7fa06fba3000+49000]
Sun Jun  2 17:32:45 2019 daemon.err /usr/bin/ss-local[8476]: remote_recv_cb_recv: Connection reset by peer
Sun Jun  2 17:32:47 2019 daemon.err /usr/bin/ss-local[8476]: remote_recv_cb_recv: Connection reset by peer
Sun Jun  2 17:32:48 2019 daemon.err uhttpd[4935]: ping: bad address 'mlvpn0'
[... cut ...]
Sun Jun  2 17:37:03 2019 kern.info kernel: [ 2501.285908] mlvpn[29144]: segfault at 19 ip 00007fa1ef60d805 sp 00007fff34a5fa30 error 6 in libc.so[7fa1ef60b000+49000]
Sun Jun  2 17:37:06 2019 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Sun Jun  2 17:37:07 2019 daemon.err /usr/bin/ss-local[8476]: remote_recv_cb_recv: Connection reset by peer
Sun Jun  2 17:37:07 2019 daemon.err /usr/bin/ss-local[8476]: remote_recv_cb_recv: Connection reset by peer

Ysurac, 2019-06-02:
Visiblement tu utilises mlvpn qui plante.

mycanaletto, 2019-06-02:
Je vais essayer de trouver ou changer ça et pourquoi il ne plantait pas avant cette mise a jour...

mycanaletto, 2019-06-02:
Avec ou sans MLVPN ça ne change rien. A tout hasard j'ai juste activé le brouillage et là tout est OK...
[image]

mycanaletto, 2019-06-03:
Et encore merci pour ton aide :-)

mycanaletto, 2019-06-15:
En fait c'est très instable, le débit baisse te l'agrégation ne semble pas se faire... Retour à la .48 qui ets OK

Ysurac, 2019-06-15:
Sans log je peux rien dire. Ensuite il faut que le VPS soit à jour aussi.

tukutt, 2019-06-17:
J'avais un soucis similaire depuis la mise à jour, j'ai réinstallé le RPI, le VPS sans succès également.

Finalement tout est rentré dans l'ordre lorsque j'ai réglé l'heure du RPI avec le bon fuseau horaire et la bonne heure (Europe/Paris) dans Système -> Système.
Mes VPS sont en UTC et le RPI en UTC+2.

mycanaletto, 2019-06-18:
Réinitialisation du routeur + reconfiguration + mise à l'heure ok

Tout semble fonctionner. 
Par contre les débits ne sont pas au rdv. 100 Mbps au lieu de 170 si je repasse en .48

On gagne un peu de débit en désactivant le brouillage pour ShadowSocks. Existe t'il autre chose qui peut ralentir ? (chacun doit pouvoir trouver l'équilibre entre ses exigences de sécurité et de performances).

tukutt, 2019-06-18:
Content de savoir que la mise à l'heure permet de résoudre le problème de stabilité.
Concernant le débit, je ne pourrais pas t'être utile avec mes 45mbits/s.

mycanaletto, 2019-06-19:
Je pense que c’est surtout la réinitialisation qui a résolu le problème. J'avais installé cette VM en important la config d'une .48 et le problème venait peut être de là.

closed 2019-07-04 by Ysurac