Ysurac/openmptcprouter #499: DHCP Problem - No Connection

rubaz100, 2019-06-02:
## Expected Behavior

  In English please.

## Actual Behavior
Sometimes the system fails randomly. I dont know why and how its happening but after a restart it will work perfectly.

dmesg log:
https://pastebin.com/YrZww2aW

syslog:
https://pastebin.com/E7UUWwAh

## Specifications
[image]

jayanta525, 2019-06-03:
This unstable system has been observed with 49.7 release. You can downgrade or use cron to periodically reboot the router. 
The same can be achieved with luci-app-watchcat

rubaz100, 2019-06-05:
I copied parts of the logs:
```
[20108.509109] ieee80211 phy1: staid 5 deleted
[20312.479543] ieee80211 phy0: staid 1 deleted
[20593.544113] ieee80211 phy0: staid 1 deleted
[21017.019765] ieee80211 phy0: staid 1 deleted
[21152.024954] ieee80211 phy0: staid 1 deleted
[21342.455944] ieee80211 phy0: staid 1 deleted
[21377.900080] ieee80211 phy0: staid 1 deleted
[21377.991528] ieee80211 phy0: staid 1 deleted
[21603.988011] ieee80211 phy1: staid 5 deleted
[21807.608234] ieee80211 phy1: staid 5 deleted
[22673.450430] ieee80211 phy1: staid 5 deleted
[22990.605809] ieee80211 phy1: staid 5 deleted
[23075.601693] ieee80211 phy1: staid 5 deleted
[23289.727350] ieee80211 phy1: staid 5 deleted
[23493.742336] ieee80211 phy1: staid 5 deleted

Wed Jun  5 15:21:07 2019 daemon.err /usr/bin/ss-redir[7620]: server recv: Host is unreachable
Wed Jun  5 15:43:45 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Wed Jun  5 15:48:42 2019 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Wed Jun  5 16:00:00 2019 cron.info crond[3541]: USER root pid 19943 cmd /etc/init.d/vnstat_backup backup

Wed Jun  5 16:26:06 2019 daemon.info unbound: [18927:0] info: generate keytag query _ta-4f66. NULL IN

Wed Jun  5 19:28:00 2019 daemon.err /usr/bin/ss-redir[7621]: remote recv: Operation timed out
Wed Jun  5 19:28:00 2019 daemon.err /usr/bin/ss-redir[7620]: remote recv: Operation timed out
Wed Jun  5 19:28:12 2019 daemon.err /usr/bin/ss-redir[7620]: remote recv: Operation timed out
Wed Jun  5 19:28:12 2019 daemon.err /usr/bin/ss-redir[7621]: remote recv: Operation timed out
Wed Jun  5 19:28:12 2019 daemon.err /usr/bin/ss-redir[7620]: remote recv: Operation timed out
Wed Jun  5 19:28:12 2019 daemon.err /usr/bin/ss-redir[7620]: remote recv: Operation timed out
Wed Jun  5 19:28:12 2019 daemon.err /usr/bin/ss-redir[7620]: remote recv: Operation timed out
Wed Jun  5 19:28:16 2019 daemon.err /usr/bin/ss-redir[7621]: remote recv: Operation timed out
Wed Jun  5 19:28:17 2019 daemon.err /usr/bin/ss-redir[7621]: remote recv: Operation timed out

Wed Jun  5 19:56:29 2019 daemon.err /usr/bin/ss-redir[7620]: remote recv: Operation timed out
Wed Jun  5 19:56:44 2019 daemon.err /usr/bin/ss-redir[7620]: remote recv: Operation timed out
Wed Jun  5 19:57:25 2019 daemon.err /usr/bin/ss-redir[7621]: remote recv: Operation timed out
```

Ysurac, 2019-06-05:
What do you call failure ?

rubaz100, 2019-06-05:
Router is not available for about 5-10min.. no Luci no ssh, no network connection. It refuses everything. Sometimes I need to hard reboot it.

Ysurac, 2019-06-05:
via ethernet or wifi link ?
Do you have any package installed ?

rubaz100, 2019-06-05:
Both, WiFi and Ethernet. No i dont have any new packages. Its had a clear installation

Ysurac, 2019-06-05:
Do you have the same problem with wifi disabled ?

rubaz100, 2019-06-05:
I don't know. Never tried it before. I will try that in 2h because my roommate would cry without an internet connection.

rubaz100, 2019-06-08:
Yeah same problem: I think that's the right log here:

[19990.123086] ieee80211 phy1: staid 4 deleted
[20056.296799] ieee80211 phy1: staid 4 deleted
[20123.784948] ieee80211 phy1: staid 5 deleted
[21746.953324] ieee80211 phy1: staid 4 deleted
[21814.223393] ieee80211 phy1: staid 5 deleted
[23609.963460] ieee80211 phy1: staid 4 deleted
[28492.235113] ieee80211 phy1: staid 4 deleted
[35377.022717] mptcp_init4_subsockets: token 0xae1b992 bind() to 10.255.255.28 index 29 failed, error -99
[35377.035199] mptcp_init4_subsockets: token 0x8648835f bind() to 10.255.255.28 index 29 failed, error -99
[35377.035203] mptcp_init4_subsockets: token 0x7cb62730 bind() to 10.255.255.28 index 29 failed, error -99
[35377.035314] mptcp_init4_subsockets: token 0x7cb62730 bind() to 10.255.255.28 index 29 failed, error -99
[35377.059763] mptcp_init4_subsockets: token 0x8648835f bind() to 10.255.255.28 index 29 failed, error -99
[35377.104913] mptcp_init4_subsockets: token 0xdd2f6c0d bind() to 10.255.255.28 index 29 failed, error -99
[35377.265810] mptcp_init4_subsockets: token 0xba3d4417 bind() to 10.255.255.28 index 29 failed, error -99
[35377.277813] mptcp_init4_subsockets: token 0xba3d4417 bind() to 10.255.255.28 index 29 failed, error -99
[35377.378290] mptcp_init4_subsockets: token 0x24b75361 bind() to 10.255.255.28 index 29 failed, error -99
[35377.391095] mptcp_init4_subsockets: token 0x24b75361 bind() to 10.255.255.28 index 29 failed, error -99
[35398.110970] IPv6: ADDRCONF(NETDEV_UP): tun0: link is not ready
[37363.081909] ieee80211 phy1: staid 4 deleted

Ysurac, 2019-06-09:
Do you use static IP or DHCP ? If DHCP try with a static IP.

rubaz100, 2019-06-09:
Both 13 devices have a static and everything else dhcp

rubaz100, 2019-06-24:
Problem is solved. ✌️

closed 2019-06-24 by rubaz100