Ysurac/openmptcprouter #500: Installation of 0,50 reboots into grub

xabix, 2019-06-03:
## Expected Behavior

  I have proceed with the update to ext4 image: openmptcprouter-v0.50-r10121-06e63aad57-x86-64-combined-ext4.img.gz (2019/06/03 - 47.24Mo - sha256sum) and the OMR doesn't reboot and I get a grub command line at boot.

Is there anything to be done instead of a re installation?

Merci

xabix, 2019-06-03:
I have tried to install the latest image and I have :
[image]

Could it be an issue with the image?

My config (which was working till now) in Proxmox:
[image]

Ysurac, 2019-06-03:
1) No choice, you have to reinstall.
2) Maybe later.

renaudadu, 2019-06-03:
hello,
i have the same issue

Specifications
OpenMPTCProuter version: (0.50)
OpenMPTCProuter VPS version: (0.995)
OpenMPTCProuter platform: x86-64 UEFI (64-bit)

Ysurac, 2019-06-03:
I'm able to reproduce the issue. I'm checking where the problem is.

Ysurac, 2019-06-03:
This will be fixed in next release that will be out soon

Randdrick, 2019-06-03:
Bonjour,
j'ai eu le même problème avec l'image openmptcprouter-v0.50-r10121-06e63aad57-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img.gz ( Donc RPI 3)

Cependant le problème a été plus grave, puisque perte total de configuration et impossibilité de recharger une sauvegarde malgré une réinstallation complète. J'ai dû faire un retour arrière sur la version 0.45 qui elle fonctionne sans problème.

Ysurac, 2019-06-03:
Le soucis ne peut pas être le même sur RPI, celui-ci n'utilisant pas Grub.

Ysurac, 2019-06-04:
Fixed in 0.50.1

closed 2019-06-04 by Ysurac

xabix, 2019-06-04:
Hello,

Is it possible to know what is being fixed in each release? to avoid to have to install them all if not really required? 
Now that I have 0.50 so just wondering if I need to really install 0.50.1 if I just resintalled it yesterday evening

xabix, 2019-06-04:
This issue is not closed. I had exactly the same issue upgrading to 0.50.1.

I will stop installing new release if there is nothing special that I need to avoid such issues. I couldn't find a 'release note', where should this be? I look at the tags etc.. but I could only see the 0.50 release and nothing about the 0.50.1 but with no details on what is supposed to be fixed or enhanced. WHere should I be looking for such details?

Ysurac, 2019-06-04:
Release 0.50.2 (that is compiling now) seem to fix the update problem.
For changes you can check commit in both https://github.com/Ysurac/openmptcprouter and https://github.com/Ysurac/openmptcprouter-feeds
I don't make release note for minor release.

xabix, 2019-06-04:
Got it thanks!

xabix, 2019-06-05:
BTW I had the exact same issue trying to upgrade to 0.50.2 from 0.50.1