Ysurac/openmptcprouter #501: Unstable Connection

CoSysopK, 2019-06-05:
## Actual Behavior

[image]

[image]

When I have this text that appears in my journal :
Wed Jun  5 15:28:48 2019 daemon.err omr-tracker[22328]: /bin/omr-tracker: line 221: can't create 1: Read-only file system
I have connection problems that become unstable.
This problem does not happen every day, just once or twice a week.
My VPS server is at Amazon.
At OVH, I never had this type of problem.

## Steps to Reproduce the Problem

  1. Install image for Raspberry PI 3/3B+ (64-bit)
  2. Install VPS Ubuntu Server 18.04 x86_64
  3. Configure the router with the wan1 static and, wan2 static
  4. Add the IP address VPS and, all the informations (shadowsocks key, glorytun key, MLVPN 
  password and OpenMPTCProuter VPS Admin key) retrived on the VPS
  5. 1 key 4G on usb PI 3B and, 1 key 4g wifi

## Specifications

  - OpenMPTCProuter version: Version 0.49.5
  - OpenMPTCProuter VPS version: Version 0.993
  - OpenMPTCProuter platform: RPI3B+ 64 bits

Ysurac, 2019-06-05:
Your filesystem is read only, this is a problem.

CoSysopK, 2019-06-06:
Hello,
when I restart my router, this sentence (can't create 1: Read-only file system) does not appear in my journal.

There is always this phrase :
Can't contact Server Admin Script (No token)
Version 0.993 4.14.110-mptcp

How should I repair these issues ?

CoSysopK, 2019-06-06:
I make this command : mount

sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=491936k,nr_inodes=122984,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=101700k,mode=755)
/dev/xvda1 on / type ext4 (rw,relatime,discard,data=ordered)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=26,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=2448)
mqueue on /dev/mqueue type mqueue (rw,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,pagesize=2M)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/var/lib/snapd/snaps/core_6818.snap on /snap/core/6818 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/core_6673.snap on /snap/core/6673 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/core_6964.snap on /snap/core/6964 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/amazon-ssm-agent_1068.snap on /snap/amazon-ssm-agent/1068 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/amazon-ssm-agent_930.snap on /snap/amazon-ssm-agent/930 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/amazon-ssm-agent_1335.snap on /snap/amazon-ssm-agent/1335 type squashfs (ro,nodev,relatime,x-gdu.hide)
lxcfs on /var/lib/lxcfs type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=101696k,mode=700,uid=1000,gid=1000)

Ysurac, 2019-06-06:
I would need mount result but on OpenMPTCProuter via SSH, not the VPS.

CoSysopK, 2019-06-07:
root@OpenMPTCProuter:~# mount
/dev/root on / type ext4 (ro,noatime,block_validity,delalloc,barrier,user_xattr)
devtmpfs on /dev type devtmpfs (rw,relatime,size=464020k,nr_inodes=116005,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mmcblk0p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)

Ysurac, 2019-06-07:
So ext4 is really readonly. This can happen after a hard reboot or with a faulty SD card.

CoSysopK, 2019-06-08:
Can I use an ubuntu Live CD to change the mount from ro to rw ?
or do I need to change my SD card ?

CoSysopK, 2019-06-08:
I just made an installation on another SD card. Here is the result :
root@OpenMPTCProuter:~# mount
/dev/root on / type ext4 (ro,noatime,block_validity,delalloc,barrier,user_xattr)
devtmpfs on /dev type devtmpfs (rw,relatime,size=464020k,nr_inodes=116005,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mmcblk0p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)

CoSysopK, 2019-06-08:
I opted for the second solution with another SD card which is the fastest method. And that allowed me to update the firmware.

Here is the result that must be satisfying ^^ :

root@OpenMPTCProuter:~# mount
/dev/root on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=464020k,nr_inodes=116005,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mmcblk0p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)

Have a good day.

closed 2019-07-04 by Ysurac