Ysurac/openmptcprouter #502: can't upgrade vps 0.94 to 0.995

mshost, 2019-06-05:
## Expected Behavior
vps on debian 9 
user :root
when try to run script 
wget -O - http://www.openmptcprouter.com/server/debian9-x86_64.sh | sh

it shows 

URL transformed to HTTPS due to an HSTS policy
--2019-06-05 22:14:02--  https://www.openmptcprouter.com/server/debian9-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... failed: Temporary failure in name resolution.
wget: unable to resolve host address ‘www.openmptcprouter.com’
  In English please.

## Actual Behavior

## Steps to Reproduce the Problem

  1. restart the vps but no luck
  2.
  3.

## Specifications

  - OpenMPTCProuter version: 0.49
  - OpenMPTCProuter VPS version: 0.94
  - OpenMPTCProuter platform: RPI3
[image]

Ysurac, 2019-06-05:
You seems to have a DNS problem on your VPS. Check DNS configuration.

mshost, 2019-06-24:
solved
thanks

closed 2019-06-24 by mshost