Ysurac/openmptcprouter #503: Request: Support "custom route script"

lars18th, 2019-06-07:
Hi @Ysurac ,

I feel you don't like supporting exotic functionalities (as example, you rejected #369). However, it may be quite simple to let the user do some advanced configurations by adding a few simple changes.

As I want to control my ROUTE TABLES manually, I request if you can support to execute an user script **after** you have done changes in the route tables.

You agree?
Thank you for this great project!

lars18th, 2019-06-11:
Hi @Ysurac ,

Would you please be good enough to state your opinion? I'm really interested on a simple solution (in the sense that with little effort) for using OMR with custom routes.

Thank you another time for your good work!

Ysurac, 2019-06-11:
You can try in Network->firewall->Custom rules

lars18th, 2019-06-11:
> You can try in Network->firewall->Custom rules

Thanks for the answer! I will try it out and will comment the results here. :+1:

lars18th, 2019-06-12:
Hi @Ysurac ,

It doesn't work!
Here is the list of commands that I need to execute:
```
ip route flush 0/0
ip route add default via 192.168.0.1 dev br-lan
ip route add 192.168.3.0/24 via 192.168.1.1 dev wan1 metric 1
ip route add 192.168.3.0/24 via 192.168.2.1 dev wan2 metric 2
```

At time I execute them _manually_. However, after every change in the network status (reboot, reconnect, etc.) I need to repeat them.
Any idea?

Ysurac, 2019-06-12:
And I think this is not working if you set them in menu network->route ?

lars18th, 2019-06-13:
> And I think this is not working if you set them in menu network->route ?

For sure not! That's the reason to request this. Your scripts are changing the route table without respecting any user configured values.

But in order to not make a complex change, I request to add support for a "custom route script". More or less like the "custom rules for firewall". Then you only need to call (`exec`) this script _after_ you do changes in the route table (not every time, but when you end to do changes). Futhermore it's not required to include a GUI, for me it's sufficient for something like "call to `/etc/config/user-route.sh`".

You agree?

lars18th, 2019-06-23:
Hi @Ysurac ,

Please, can you help in any sense to obtain the route table that I need?

Thank you for this amazing project!

lars18th, 2019-06-27:
Hi @Ysurac ,

I appreciate the good work you did with this impressive project!
Please, can you comment about this request?
Thank you!

Ysurac, 2019-06-27:
For now you can add your entry at the end of /etc/init.d/mptcp in start function.

lars18th, 2019-06-28:
> For now you can add your entry at the end of /etc/init.d/mptcp in start function.

Thank you! I'll try it. :smiley:

lars18th, 2019-06-28:
> For now you can add your entry at the end of /etc/init.d/mptcp in start function.

Awesome! It works like a charm! :+1: 

I've added the call to a new function `custom_route()` within the `start_service()` before the `uci commit network` and it does the trick:
https://github.com/Ysurac/openmptcprouter-feeds/blob/master/mptcp/files/etc/init.d/mptcp#L307

Thank you! :wink: 

Maybe you like to add then someting like this to the `/etc/init.d/mptcp` file:
```
#!/bin/sh /etc/rc.common
# Copyright (C) 2018 Ycarus (Yannick Chabanois) <ycarus@zugaina.org>
# Released under GPL 3. See LICENSE for the full terms.

START=90

USE_PROCD=1

. /usr/lib/unbound/iptools.sh
. /lib/functions/network.sh

custom_route() {
        if [ -f "/etc/config/user-custom-route.sh" ]
        then
            source /etc/config/user-custom-route.sh
        fi
}
[...]

start_service() {
[...]
	fi
        custom_route
	uci -q commit network
[...]
```

Regards.

lars18th, 2019-07-05:
Hi @Ysurac ,

This patch over the `/etc/init.d/mptcp` file works... but not everytime. If I don't touch the OMR router (just simple reboots, WAN connection/disconnections, etc.) all works. However, if I do some change in the UI, sometimes (when the change has some impact in the routing) it fails. So I suspect, in some cases your scripts are changing the routing table outside of this file.

Please, can you put in the roadmap (to consider it in the future) to add support for the file `/etc/config/user-custom-route.sh` (or whatever name that you prefer)?

Thank you!

closed 2019-12-20 by Ysurac

lars18th, 2019-12-20:
Hi @Ysurac ,

> Please, can you put in the roadmap (to consider it in the future) to add support for the file `/etc/config/user-custom-route.sh` (or whatever name that you prefer)?

So you reject this suggestion?

Ysurac, 2019-12-20:
Yes. Too much possible conflict with MPTCP conf and it's a corner use case...
But you a are free to submit a pull request.