Ysurac/openmptcprouter #505: how to set v2ray-plugin

fmitk, 2019-06-07:
## Expected Behavior

Allow devices from local network to use proxy function

## Actual Behavior
It could not implement proxy 

## Steps to Reproduce the Problem

  1.I install this openwrt on my rpi-3b
  2.i have a remote vps has installed a v2ray server
  3.i want to implement proxy for my rpi's local network

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.50.2
  - OpenMPTCProuter VPS version: debian 9
  - OpenMPTCProuter platform: RPI3

Ysurac, 2019-06-07:
You need to install OpenMPTCProuter VPS script on the VPS.

fmitk, 2019-06-08:
as you said , i have got a vps admin key ,and ssh port, but I still confuse that how to them inmy pi that has installed openwrt

Ysurac, 2019-06-08:
In System->OpenMPTCProuter, you set the key and WAN configuration in wizard tab. In advanced settings tab you can set v2ray settings.

closed 2019-07-04 by Ysurac