Ysurac/openmptcprouter #506: lan (No gateway defined) & omr6in4 (No IP defined)

CoSysopK, 2019-06-08:
## Actual Behavior

[image]

[image]

## Steps to Reproduce the Problem

  1. Install image for Raspberry PI 3/3B+ (64-bit)
  2. Install VPS Ubuntu Server 18.04 x86_64
  3. Configure the router with the wan1 static and wan2 static
  4. Add the IP address VPS and, all the informations (shadowsocks key, glorytun key, MLVPN
password and OpenMPTCProuter VPS Admin key) retrived on the VPS
  5. 1 key 4G on usb PI 3B and, 1 key 4g wifi

## Specifications

  - OpenMPTCProuter version: Version 0.50.2
  - OpenMPTCProuter VPS version: Version 0.993
  - OpenMPTCProuter platform: RPI3B+ 64 bits

Ysurac, 2019-06-08:
Lan and omr6in4 should not have multipath enabled.

CoSysopK, 2019-06-08:
I do not understand ! 
I just do a reinstall today

CoSysopK, 2019-06-08:
multipath is well on disable

[image]

CoSysopK, 2019-06-08:
The problem is here :

[image]

interface parameter, there are LAN and OMR6IN4 !!!
I deleted these 2 interfaces but it caused an error.
So I made an "update" on the router with the same version and it put everything back in the right order.

[image]

It Work !

closed 2019-06-10 by Ysurac