Ysurac/openmptcprouter #507: shadowsock is not running and can't get public ip address from shadowsocks

renaudadu, 2019-06-08:
hello,
since the version 0.50, i have this problem.
[image]

i have reinstall the router, the vps and i have the same.

where can i search ?

for the parameter on router
i put the key vps
ip of the modem
some omr-bypass
active snmp V1

## Specifications

  - OpenMPTCProuter version: (0.50.2)
  - OpenMPTCProuter VPS version: (0.995 with pihole) firstheberg and sapinet (also)
  - OpenMPTCProuter platform: (x86_64) i try on raspberry same think

Ysurac, 2019-06-08:
Check OpenMPTCProuter system log and VPS /var/log/daemon.log if you use Debian.

Ysurac, 2019-06-08:
You can also try to go on wizard and save again if you use a previous backup.

renaudadu, 2019-06-09:
where can i change the shadows key on the vps ? in /etc/shadowsocks/config.json ?
i have a trouble on vps i have something like that xXZ-h_M and on router i have xXZ+h/M

Ysurac, 2019-06-09:
It's not a problem. config.json on both side will have the same key, + and / are replaced by - and _

renaudadu, 2019-06-09:
this is my router conf file. would you like something else ?
i have reinstall the vps this morning. but nothings workings
wan2 down is normal 
wan3 backup normal too

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].server='10.255.255.1#53'
dhcp.@dnsmasq[0].rebind_protection='0'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/6cloud.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/6play.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/www.6play.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/geo.6play.fr/omr_dst_bypass_all,omr6_dst_bypass_all' '/netflix.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/netflix.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxext.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dst_bypass_all,omr6_dst_bypass_all'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.ignore='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@domains[0]=domains
dscp.@domains[0].name='googlevideo.com'
dscp.@domains[0].class='cs4'
dscp.@domains[0].comment='Youtube'
dscp.@domains[1]=domains
dscp.@domains[1].name='nflxvideo.net'
dscp.@domains[1].class='cs4'
dscp.@domains[1].comment='NetFlix'
dscp.@domains[2]=domains
dscp.@domains[2].name='s3.ll.dash.row.aiv-cdn.net'
dscp.@domains[2].class='cs4'
dscp.@domains[2].comment='AmazonVideo'
dscp.@domains[3]=domains
[... 1220 lines cut ...]
upnpd.config.enabled='0'
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3'

Ysurac, 2019-06-09:
Even if this should be the case, check on the VPS that shadowsocks use chacha20 too.
And check the logs...

renaudadu, 2019-06-28:
i have resinstall all and all is good during 2 week.
yesterdays i want to active ofuscation and now i am in the same situation.
ofuscation is now disable but it's not working good.
i noticed this on the vps : 
systemctl status shadowsocks-libev
● shadowsocks-libev.service - Shadowsocks-libev Default Server Service
   Loaded: loaded (/lib/systemd/system/shadowsocks-libev.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2019-06-28 06:36:15 CEST; 17min ago
     Docs: man:shadowsocks-libev(8)
  Process: 6099 ExecStart=/usr/bin/ss-server -c $CONFFILE $DAEMON_ARGS (code=exited, status=255)
 Main PID: 6099 (code=exited, status=255)

Jun 28 06:36:15 test2 systemd[1]: Started Shadowsocks-libev Default Server Service.
Jun 28 06:36:15 test2 ss-server[6099]:  2019-06-28 06:36:15 INFO: resolving hostname to IPv6 address first
Jun 28 06:36:15 test2 ss-server[6099]:  2019-06-28 06:36:15 INFO: using tcp fast open
Jun 28 06:36:15 test2 ss-server[6099]:  2019-06-28 06:36:15 INFO: plugin "/usr/local/bin/obfs-server" enabled
Jun 28 06:36:15 test2 systemd[1]: shadowsocks-libev.service: Main process exited, code=exited, status=255/n/a
Jun 28 06:36:15 test2 systemd[1]: shadowsocks-libev.service: Unit entered failed state.
Jun 28 06:36:15 test2 systemd[1]: shadowsocks-libev.service: Failed with result 'exit-code'.

the config.xml : cat /etc/shadowsocks-libev/config.json
{
    "ipv6_first": true,
    "server": [
        "[::0]",
        "0.0.0.0"
    ],
    "mode": "tcp_and_udp",
    "key": "XXXXXXXXXXXXXXXXXXXXX",
    "server_port": 65101,
    "fast_open": true,
    "plugin_opts": "obfs=tls;mptcp;fast-open;t=400;host=ip36.ip-37-59-231.eu",
    "plugin": "/usr/local/bin/obfs-server",
    "no_delay": true,
    "reuse_port": true,
    "ebpf": false,
    "method": "chacha20",
    "verbose": 1,
    "mptcp": true,
    "local_port": 1081,
    "timeout": 600,
    "prefer_ipv6": false
do you see something wrong

Ysurac, 2019-06-28:
obfuscation is still enabled (the 2 plugins lines).
If you are not in china, no need to activate this.

renaudadu, 2019-06-28:
on the router it's disabled
i don't understand why it's activate on vps, i have delete the line et restart the service.

i verify on the router and i have restart it
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: started, version 2.80 cachesize 150
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: DNS service limited to local subnets
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: using local addresses only for domain test
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: using local addresses only for domain onion
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: using local addresses only for domain localhost
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: using local addresses only for domain local
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: using local addresses only for domain invalid
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: using local addresses only for domain bind
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: using nameserver 10.255.255.1#53
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: using local addresses only for domain lan
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: read /etc/hosts - 4 addresses
Fri Jun 28 07:48:02 2019 daemon.info dnsmasq[2555]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Fri Jun 28 07:48:02 2019 authpriv.info dropbear[2680]: Not backgrounding
Fri Jun 28 07:48:02 2019 user.notice Shadowsocks: Rules UP
Fri Jun 28 07:48:02 2019 user.notice Shadowsocks: Reload omr-bypass rules
Fri Jun 28 07:48:03 2019 user.notice : Added device handler type: 8021ad
Fri Jun 28 07:48:03 2019 user.notice : Added device handler type: 8021q
Fri Jun 28 07:48:03 2019 user.notice : Added device handler type: macvlan
Fri Jun 28 07:48:03 2019 user.notice : Added device handler type: veth
Fri Jun 28 07:48:03 2019 user.notice : Added device handler type: bridge
Fri Jun 28 07:48:03 2019 user.notice : Added device handler type: Network device
Fri Jun 28 07:48:03 2019 user.notice : Added device handler type: tunnel
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]:
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]: Initalizing available sources
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]:
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]: Initalizing entropy source hwrng
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]:
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]: Enabling RDRAND rng support
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]:
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]: Initalizing entropy source rdrand
Fri Jun 28 07:48:03 2019 daemon.err rngd[3025]:
Fri Jun 28 07:48:03 2019 kern.notice kernel: [    9.553718] random: crng init done
Fri Jun 28 07:48:03 2019 kern.notice kernel: [    9.553731] random: 1 urandom warning(s) missed due to ratelimiting
Fri Jun 28 07:48:03 2019 cron.info crond[3200]: crond (busybox 1.30.1) started, log level 8
Fri Jun 28 07:48:03 2019 daemon.notice netifd: Interface 'lan' is enabled
Fri Jun 28 07:48:03 2019 daemon.notice netifd: Interface 'lan' is setting up now
Fri Jun 28 07:48:03 2019 daemon.notice netifd: Interface 'lan' is now up
Fri Jun 28 07:48:03 2019 daemon.notice netifd: Interface 'loopback' is enabled
Fri Jun 28 07:48:03 2019 daemon.notice netifd: Interface 'loopback' is setting up now
Fri Jun 28 07:48:03 2019 daemon.notice netifd: Interface 'loopback' is now up
Fri Jun 28 07:48:03 2019 daemon.notice netifd: Interface 'omr6in4' is setting up now
Fri Jun 28 07:48:03 2019 kern.info kernel: [   10.260909] 8021q: adding VLAN 0 to HW filter on device eth0
Fri Jun 28 07:48:03 2019 kern.info kernel: [   10.266378] e1000e 0000:02:00.0 eth1: changing MTU from 1500 to 1499
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Interface 'wan1' is enabled
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Interface 'wan1' is setting up now
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Interface 'wan1' is now up
Fri Jun 28 07:48:04 2019 kern.info kernel: [   10.541038] 8021q: adding VLAN 0 to HW filter on device eth1
Fri Jun 28 07:48:04 2019 kern.info kernel: [   10.544436] e1000e 0000:03:00.0 eth2: changing MTU from 1500 to 1475
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Interface 'wan2' is enabled
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Interface 'wan2' is setting up now
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Interface 'wan2' is now up
Fri Jun 28 07:48:04 2019 kern.info kernel: [   10.821420] 8021q: adding VLAN 0 to HW filter on device eth2
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Interface 'omr6in4' is now up
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Network device 'lo' link is up
Fri Jun 28 07:48:04 2019 daemon.notice netifd: Interface 'loopback' has link connectivity
Fri Jun 28 07:48:04 2019 daemon.notice netifd: tunnel '6in4-omr6in4' link is up
Fri Jun 28 07:48:04 2019 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/dhcp
Fri Jun 28 07:48:04 2019 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/radvd
Fri Jun 28 07:48:04 2019 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/macvlan
Fri Jun 28 07:48:04 2019 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/omr6in4
[... cut ...]
Fri Jun 28 07:48:16 2019 user.notice post-tracking-post-tracking: Set server vps (X.X.X.X) route via 192.168.1.1 metric 3
Fri Jun 28 07:48:16 2019 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Fri Jun 28 07:48:20 2019 user.notice post-tracking-post-tracking: Set server vps (X.X.X.X) route via 192.168.9.1 metric 4
Fri Jun 28 07:48:20 2019 user.notice post-tracking-post-tracking: **_Can't find Shadowsocks, restart it._**..
Fri Jun 28 07:48:27 2019 daemon.info procd: Instance vmtoolsd::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

renaudadu, 2019-06-28:
i find the solution more instable since the 0.49.5
i have the impression that paramater was  not apply some time.

closed 2019-07-04 by Ysurac